lubeda/EspHoMaTriXv2 #3: ehmtx.weekday.color and ehmtx.today.color actions fail

skgsergio, 2023-05-04:
I tried doing the following in the `esphome->on_boot->then`:

```yaml
esphome:
  # [...]
  on_boot:
    then:
      # [...]

      - ehmtx.weekday.color:
          id: rgb8x32
          red: !lambda return 100;
          green: !lambda return 100;
          blue: !lambda return 100;

      - ehmtx.today.color:
          id: rgb8x32
          red: !lambda return 199;
          green: !lambda return 90;
          blue: !lambda return 0;
```

Which looks like it should be valid according to this:

https://github.com/lubeda/EspHoMaTriXv2/blob/main/components/ehmtxv2/__init__.py#L211
https://github.com/lubeda/EspHoMaTriXv2/blob/main/components/ehmtxv2/__init__.py#L194

But I get the following error:

```
Compiling /data/pixel-clock/.pioenvs/pixel-clock/src/main.cpp.o
src/main.cpp:86:10: error: 'SetWeekdayColor' in namespace 'esphome' does not name a template type
 esphome::SetWeekdayColor<> *esphome_setweekdaycolor;
          ^~~~~~~~~~~~~~~
src/main.cpp:87:10: error: 'SetTodayColor' in namespace 'esphome' does not name a template type
 esphome::SetTodayColor<> *esphome_settodaycolor;
          ^~~~~~~~~~~~~
/config/esphome/pixel-clock.yaml: In function 'void setup()':
/config/esphome/pixel-clock.yaml:379:3: error: 'esphome_setweekdaycolor' was not declared in this scope
/config/esphome/pixel-clock.yaml:379:42: error: 'SetWeekdayColor' in namespace 'esphome' does not name a template type
/config/esphome/pixel-clock.yaml:379:58: error: expected primary-expression before '>' token
/config/esphome/pixel-clock.yaml:51:3: error: 'esphome_settodaycolor' was not declared in this scope
       - ehmtx.today.color:
   ^   ~~~~~~~~~~~~~~~~~
/config/esphome/pixel-clock.yaml:51:40: error: 'SetTodayColor' in namespace 'esphome' does not name a template type
       - ehmtx.today.color:
                                        ^            
/config/esphome/pixel-clock.yaml:51:54: error: expected primary-expression before '>' token
       - ehmtx.today.color:
                                                      ^
/config/esphome/pixel-clock.yaml:59:112: error: no matching function for call to 'esphome::Automation<>::add_actions(<brace-enclosed initializer list>)'
 
                                                                                                                ^
In file included from src/esphome/components/display/display_buffer.h:5,
                 from src/esphome/components/addressable_light/addressable_light_display.h:5,
                 from src/esphome.h:4,
                 from src/main.cpp:3:
src/esphome/core/automation.h:237:8: note: candidate: 'void esphome::Automation<Ts>::add_actions(const std::vector<esphome::Action<Ts ...>*>&) [with Ts = {}]'
   void add_actions(const std::vector<Action<Ts...> *> &actions) { this->actions_.add_actions(actions); }
        ^~~~~~~~~~~
src/esphome/core/automation.h:237:8: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'const std::vector<esphome::Action<>*, std::allocator<esphome::Action<>*> >&'
*** [/data/pixel-clock/.pioenvs/pixel-clock/src/main.cpp.o] Error 1
```

skgsergio, 2023-05-04:
If that's the case then I'll do it with the lambda, no problem. Just need to remove the action as it is misleading.

Thanks for the quick response!

skgsergio, 2023-05-04:
Done, feel free to close this issue or keep it as a reminder for removing the actions :)

```yaml
esphome:
  # ...
  on_boot:
    then:
      # ...

      - lambda: |-
          id(rgb8x32)->set_weekday_color(100, 100, 100);
          id(rgb8x32)->set_today_color(199, 90, 0);
```

lubeda, 2023-05-05:
the actions are already removed in the "development" branch

closed 2023-05-05 by lubeda