lubeda/EspHoMaTriXv2 #4: rainbow_icon_screen service shows wrong icon

skgsergio, 2023-05-04:
I'm calling the service as follows and it is showing the first icon that I have configured (`ha`) and not the `calendar` one:

```yaml
service: esphome.pixel_clock_rainbow_icon_screen
data:
  default_font: false
  icon_name: calendar
  text: test
  lifetime: 1
  screen_time: 1
```

However `icon_screen` service works correctly:

```yaml
service: esphome.pixel_clock_icon_screen
data:
  default_font: false
  icon_name: calendar
  text: test
  lifetime: 1
  screen_time: 1
  r: 200
  g: 200
  b: 200
```

My icons:

```yaml
  icons:
    - id: ha
      lameid: 7956
    - id: tempc
      lameid: 2422
    - id: plug
      lameid: 403
    - id: humidity
      lameid: 51764
    - id: co2
      lameid: 30662
    - id: calendar
      lameid: 12345
```

lubeda, 2023-05-05:
Hi, thanx for this perfect issue-entry!! Most people don't provide enough information for me to investigate.

I take a look at this issue!

lubeda, 2023-05-05:
fixed in "develop"

closed 2023-05-05 by lubeda