lubeda/EspHoMaTriXv2 #7: Show date

idodov, 2023-05-05:
Please return back the function that support not showing the date card as defualt (show_date: false).

lubeda, 2023-05-05:
Hi, 

there is another way to remove the date display:

The date display is a screen in the queue, you can remove it in the YAML:
```
esphome:
  name: $devicename 
  on_boot:
    then: 
      - delay: 30s
      - lambda: |-
          id(rgb8x32)->del_screen("*",3);
```
or via service call:
```
service: esphome.ulanzi_del_screen
data:
  icon_name: "*"
  mode: 3
```

If you want a blank screen use:

```
service: esphome.esp32mic_blank_screen
data:
  lifetime: 1440
  screen_time: 10
```

closed 2023-05-05 by lubeda

idodov, 2023-05-06:
I have devised another method. I added automation to home assistant that delete the screen when I power on the device.

Thanks!!!

idodov, 2023-05-06:
Thank you so much!

lubeda, 2023-05-08:
Hi,
what's your aim? You will only use the display as a clock and nothing else? 

The behavior now is: On device start when there is nothing else in the queue, it will display: 
```
clock_screen(24 * 60, 10)
date_screen(24 * 60, 5)
```
After one day (24*60) both screens are at their end of life and will be removed from the queue => empty queue => and again:
```
clock_screen(24 * 60, 10)
date_screen(24 * 60, 5)
```
And so on.

fbradyirl, 2023-05-08:
ok I see. My intention is to have it mostly as a clock and push other screens upon trigger as temperature changes, etc. 

Just the default of showing the date every few seconds is not something I want.

lubeda, 2023-05-08:
OK, I will think about a solution.

lubeda, 2023-05-31:
Hi, the function is there, but I forgot to write it in the docs. Give it a try.

ofirsnb, 2023-05-31:
Awesome!  Works like a charm!! 👏  
Thanks a lot:)

idodov, 2023-05-31:
After adding 
```
show_date: false
``` 
The black screen appears every 10 seconds (for 5 seconds).

Adding 
```
  on_boot:
    then:
      - ds1307.read_time:
      - delay: 30s        
      - lambda: |-
          id(rgb8x32)->del_screen("*",1);

``` 
Didn't solved the problem.

ofirsnb, 2023-05-31:
Indeed.  I didn't pay attention cause I had the "delete" lambda on boot as well,  forgot to remove it.
(BTW,  You need to delete mode "3" for date: `id(rgb8x32)->del_screen("*",3);`,   it works )

Let's check in the next "tick" in one day (24*60) if it's getting reset or not.

ofirsnb, 2023-06-02:
A day and a half later,  the Date screen didn't popped back.
So for now we have to keep the:
```
on_boot:
    then:
      - delay: 30s 
      - lambda: |-
          id(rgb8x32)->del_screen("*", 3);
```

Regardless,  @lubeda do you think it's an easy fix - to remove the blank date screen in the first place?   So we won't have to delete it on boot.