lubeda/EspHoMaTriXv2 #8: [BUG] Blueprints not working

fbradyirl, 2023-05-07:
# Bug report

Thanks for this repo but the Blueprints I’ve tried all show me this error.

[image]

[image]

## Additional information

- used Hardware:
  - ESP32
  - EspHoMaTriXv2 version: latest master code from 7 May 2023

## To Reproduce

Steps to reproduce the behavior:

## Expected behavior

A clear and concise description of what you expected to happen.

## Configuration

```text
(optional) The YAML you used in epshome without any password
```

## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Services calls

```text
(optional) The YAML of your service calls
```

lubeda, 2023-05-08:
I will investigate

lubeda, 2023-05-08:
Hi, please try this version: https://github.com/lubeda/EspHoMaTriXv2/blob/develop/copy2esphome/EHMTX_easy_state.yaml

fbradyirl, 2023-05-08:
Yes this works thank you. I think the issue is with all the other blueprints also.

lubeda, 2023-05-08:
I will check, but not today.

fbradyirl, 2023-05-08:
No problem thanks for sorting this.

closed 2023-05-11 by lubeda