lubeda/EspHoMaTriXv2 #10: Cannot install [BUG]

gurudc, 2023-05-10:
# Bug report

## Describe the bug

Attempted to install on new Ulanzi clock today.

## Additional information

Ulanzi TC001 

## To Reproduce

Followed documentation attempting to install ulanzi-easy.yaml onto TC001.

## Expected behavior

A clear and concise description of what you expected to happen. Could not get it to compile with following error.

Compiling /data/ulanzi/.pioenvs/ulanzi/src/main.cpp.o
/config/esphome/ulanzi-easy.yaml: In lambda function:
/config/esphome/ulanzi-easy.yaml:256:14: error: 'x' was not declared in this scope
           iconname: !lambda "return x.c_str();"
              ^
/config/esphome/ulanzi-easy.yaml: In lambda function:
/config/esphome/ulanzi-easy.yaml:257:14: error: 'y' was not declared in this scope
           text: !lambda "return y.c_str();"
[logs_ulanzi-easy_compile.txt](https://github.com/lubeda/EspHoMaTriXv2/files/11439643/logs_ulanzi-easy_compile.txt)

              ^
/config/esphome/ulanzi-easy.yaml:257:14: note: suggested alternative: 'yn'
           text: !lambda "return y.c_str();"

Log file attached

lubeda, 2023-05-10:
Hi, I just pushed the fixed version. Please try the new version https://raw.githubusercontent.com/lubeda/EspHoMaTriXv2/main/copy2esphome/ulanzi-easy.yaml

gurudc, 2023-05-10:
Compiles and installs.

thanks

closed 2023-05-10 by gurudc