lubeda/EspHoMaTriXv2 #12: [BUG] Icon Screens, Screen Time... not honored?

darkpoet78, 2023-05-14:
# Bug report

Calling various icon screens from Home Assistant, it would appear that the screen times are not being honored.  I'd like to keep my icon screens short but I can see in the code that the default is 5 seconds and shortening the time in Home Assistant doesn't change that...

```
      - service: esphome.pixelclock_yellow_icon_screen
        data:
          default_font: true
          icon_name: >-
            weather_{{ states('sensor.openweathermap_condition')|replace("-",
            "_") }}
          text: "{{ states('sensor.openweathermap_temperature')|round(1) }}°"
          lifetime: 5
          screen_time: 3
          r: 192
          g: 192
          b: 192
```

I also tried this in my YAML and it doesn't lengthen the default clock screen:

```
esphome:
  comment: "EHMTXv2 from LuBeDa"
  name: $devicename 
  on_boot:
    then:      
      - ds1307.read_time:
      - delay: 30s
      - lambda: |-
          id(rgb8x32)->set_clock_time(30);
```

So far, I fixed this by simply doing this in my Home Assistant (deleting the default clock screen and inserting my own):
```
  - service: esphome.pixelclock_yellow_del_screen
    data:
      icon_name: "*"
      mode: 2
  - service: esphome.pixelclock_yellow_clock_screen
    data:
      default_font: false
      lifetime: 5
      screen_time: 20
      r: 64
      g: 128
      b: 255
```

Sidenote that I miss one of the old features of clock intervals from v1... that the default clock screen would display a minimum of every x seconds (that was clock intervals)... this made it easier for me to force every 2nd screen to be the clock (ie. Clock, Weather, Clock, Inside Temperature, Clock, Humidity, etc.)  You know what I mean?  It easy enough just to change the clock interval to 5 seconds, info screens 5 seconds, forcing my preferred behavior... I can ignore losing that if I can speed through all my info screens...

lubeda, 2023-05-14:
I will investigate the bug and think about a solution for the lost feature!

lubeda, 2023-05-14:
OK, I found the error.
What would be the best way, thinking of two possible text length?
For short text, that fits without scrolling:
- **use the `screen time`?**
For long text, that has to be scrolled:
- use `screen time`, regardless if the text is scrolled at least once?
- use the maximum of `screen_time` or the time it takes to scroll the text `scroll_count` times?
- **use the time it takes to scroll the text `scroll_count` times?**

The **bold** lines are my favorite options.

lubeda, 2023-05-15:
Please check again, the clock_intervall will follow next weekend

closed 2023-05-16 by lubeda

lubeda, 2023-05-16:
Hi, you can try the develop branch for testing clock_interval. By default, it is disabled!

darkpoet78, 2023-05-16:
Switched to the develop branch, added this to my YAML:

```
  on_boot:
    then:      
      - ds1307.read_time:
      - delay: 30s
      - lambda: |-
          id(rgb8x32)->set_clock_time(8);
          id(rgb8x32)->set_clock_interval(2);
```

Then set my icon screens to 3 seconds... It seems to behave a little different than the old interval.  Icon screen will finish out whatever its time was set to before returning to the clock.

So, ie I set the interval to 5 and Icon screens set to 4 each, then 2 icon screens will go by before returning to the clock.

In other words, it works perfectly!

BTW, I'm guessing that this clock is actually the default clock and adding any clock screens may interfere with this expected behavior.  I haven't had time to test that out.  I just removed the clock screens from my Home Assistant automation before playing around.

Awesome job!