lubeda/EspHoMaTriXv2 #14: [BUG] can't update esphome to firmware 4.5

idodov, 2023-05-17:
Error OTA updating & Error compiling with no more info at the log

lubeda, 2023-05-17:
If OTA isn't working try update via USB!
Compile errors do always have a log entry!
What do you mean with firmware 4.5?

ehmtx does work on my ulanzi after update of esphome to 2023.5.0.

closed 2023-05-17 by lubeda

darkpoet78, 2023-05-17:
Copy and paste your ESPHome log output if you want some help.  Also, do you have any experience with ESPHome?  This is tough project for a first run with ESPHome...  What's your hardware?  Is it an Ulanzi?

If this is all true, might I recommend downloading a legacy bin to your computer, plugging the Ulanzi in, then flashing with Tasmotizer instead?

Ulanzis appear to have some variation BTW.  I have two and one is much more finicky than the other about being flashed.  Both make a different squealing sound when in flash mode, too.  They're wonderful but weird.

idodov, 2023-05-17:
Sorry. I was talking about version 5.0 from today.

Here is the log. 
Any help will be helpful

Updating /config/esphome/pixel.yaml
------------------------------------------------------------

INFO Reading configuration /config/esphome/pixel.yaml...
INFO Updating https://github.com/lubeda/EspHoMaTriXv2@main
WARNING 
WARNING This is a beta version of https://github.com/lubeda/EspHoMaTriXv2
WARNING 
WARNING GPIO15 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'Asia/Jerusalem'
INFO Detected timezone 'Asia/Jerusalem'
INFO Generating C++ source...
INFO Preparing icons, this may take some seconds.
INFO EsphoMaTrix: wrote html-file with icon preview: /config/esphome/pixel.html
INFO List of icons for e.g. blueprint:

["error","music","tv","home_assistant","temperature","lightbulb","music2","phone","car","sleep8x32",]

INFO Compiling app...
Processing ulanzi (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.3.0)
--------------------------------------------------------------------------------
Library Manager: Installing SPI
INFO Installing SPI
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- ESPmDNS @ 2.0.0
|-- Wire @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- NeoPixelBus @ 2.7.3
|   |-- SPI @ 2.0.0
Compiling /data/ulanzi/.pioenvs/ulanzi/src/main.cpp.o
/config/esphome/pixel.yaml: In lambda function:
/config/esphome/pixel.yaml:299:14: error: 'x' was not declared in this scope
           iconname: !lambda "return x.c_str();"
              ^
/config/esphome/pixel.yaml: In lambda function:
/config/esphome/pixel.yaml:300:14: error: 'y' was not declared in this scope
           text: !lambda "return y.c_str();"
              ^
/config/esphome/pixel.yaml:300:14: note: suggested alternative: 'yn'
           text: !lambda "return y.c_str();"
              ^
              yn
In file included from src/esphome.h:12,
                 from src/main.cpp:3:
src/esphome/components/api/homeassistant_service.h: In instantiation of 'esphome::api::TemplatableStringValue<X>::TemplatableStringValue(F) [with F = setup()::<lambda(std::__cxx11::string, std::__cxx11::string)>; typename std::enable_if<esphome::is_invocable<F, X ...>::value, int>::type <anonymous> = 0; X = {std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >}]':
src/esphome/components/api/homeassistant_service.h:27:108:   required from 'esphome::api::TemplatableKeyValuePair<Ts>::TemplatableKeyValuePair(std::__cxx11::string, T) [with T = setup()::<lambda(std::__cxx11::string, std::__cxx11::string)>; Ts = {std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >}; std::__cxx11::string = std::__cxx11::basic_string<char>]'
src/esphome/components/api/homeassistant_service.h:42:36:   required from 'void esphome::api::HomeAssistantServiceCallAction<Ts>::add_data_template(std::__cxx11::string, T) [with T = setup()::<lambda(std::__cxx11::string, std::__cxx11::string)>; Ts = {std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >}; std::__cxx11::string = std::__cxx11::basic_string<char>]'
/config/esphome/pixel.yaml:300:4:   required from here
src/esphome/components/api/homeassistant_service.h:22:90: error: invalid use of void expression
       : TemplatableValue<std::string, X...>([f](X... x) -> std::string { return to_string(f(x...)); }) {}
                                                                                 ~~~~~~~~~^~~~~~~~~
src/esphome/components/api/homeassistant_service.h: In instantiation of 'esphome::api::TemplatableStringValue<X>::TemplatableStringValue(F) [with F = setup()::<lambda(std::__cxx11::string, std::__cxx11::string)>; typename std::enable_if<esphome::is_invocable<F, X ...>::value, int>::type <anonymous> = 0; X = {std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >}]':
src/esphome/components/api/homeassistant_service.h:27:108:   required from 'esphome::api::TemplatableKeyValuePair<Ts>::TemplatableKeyValuePair(std::__cxx11::string, T) [with T = setup()::<lambda(std::__cxx11::string, std::__cxx11::string)>; Ts = {std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >}; std::__cxx11::string = std::__cxx11::basic_string<char>]'
[... 304 lines cut ...]
      lameid: 40530
    - id: music
      lameid: 5865
    - id: tv
      lameid: 10972
    - id: home_assistant
      lameid: 47693
    - id: temperature
      lameid: 2056
    - id: lightbulb
      lameid: 1762
    - id: music2
      lameid: 45625
    - id: phone
      lameid: 1232
    - id: car
      lameid: 2819
    - id: sleep8x32
      url: https://user-images.githubusercontent.com/16407309/224850723-634c9b2d-55d9-44f2-9f93-765c0485b090.gif 
  on_next_screen:
    - homeassistant.event:
        event: esphome.new_screen
        data_template:
          iconname: !lambda "return x.c_str();"
          text: !lambda "return y.c_str();"

lubeda, 2023-05-17:
ALSO remove the part in the yaml

starting with

`on_next_screen` to the end.

There where some changes!

idodov, 2023-05-17:
Seems that I **solved** the problem.
My yaml code and fonts are working perfect after removing this lines. 
`on_next_screen:
    - homeassistant.event:
        event: esphome.new_screen
        data_template:
          iconname: !lambda "return x.c_str();"
          text: !lambda "return y.c_str();"`

thank for your help and looking forward to the RTL direction support!

lubeda, 2023-05-17:
For the RTL support: can you sketch what it should look like?

Like this for the word "text"
[image]
[image]
[image]

idodov, 2023-05-17:
This is the order for RTL:
First:
[image]
Middle:
[image]
Last:
[image]

The string start from Right to Left.

lubeda, 2023-05-18:
Hi, is this the right  way for rtl?
https://youtu.be/mHEXAHn6b0A

idodov, 2023-05-20:
After having support for the direction of the given text string, If possible, could you consider adding support for the Bi-directional Text (BiDi) algorithm to your code? 

Currently, ESPHome does not have native support for languages that are written in a right-to-left direction, such as Hebrew, Arabic, and others. The BiDi algorithm is a standard method for handling such languages. 

You can find more information about the algorithm here: 
https://unicode-org.github.io/icu/userguide/transforms/bidi.html

There is an open-source solution available that addresses this issue, called the Bidi Reference C++ project. You can access it here:

https://unicode.org/Public/PROGRAMS/BidiReferenceCpp/v26/

Integrating this solution into your custom ESPHome code would greatly enhance its compatibility and usability for right-to-left languages.

lubeda, 2023-05-20:
Hi, after investigating some background information about bidi, I will not implement it in my software. Because it does not belong to driving the display. It is more about overall display handling. Especially the reversing of numbers and {[()]} is beyond my scope.

I see the problem, but I can't help.

idodov, 2023-05-20:
It's ok. 
We still have some workaround method to handle the bidi problem