lubeda/EspHoMaTriXv2 #18: [BUG]

thewolf-oz, 2023-05-22:
# Bug report

## Describe the bug

When adding the light component, the provided code/lambdas to turn on/off are returning the 'set_enabled' function is not available. Incidentally, the sample code also has inconsistent indentations

Without this, the light flashes briefly and then turns off at next screen refresh

## Additional information

- used Hardware:
  - Ulanzi
  - EspHoMaTriXv2 version (current - just compiled on 22 May 2023)
  - ESPHome 2023.5.1

## To Reproduce

Steps to reproduce the behavior:
Added sample code to YAML and corrected the indentations

## Expected behavior

A clear and concise description of what you expected to happen.

## Configuration

```text
light:
  - platform: neopixelbus
    id: ehmtx_light
    type: GRB
#    internal: true
    variant: WS2812
    pin: $matrix_pin
    num_leds: 256
    color_correct: [30%, 30%, 30%]
    gamma_correct: 2.0
    name: "$device_friendly_name Light"
    restore_mode: ALWAYS_OFF
    on_turn_on:
      lambda: |-
        id(rgb8x32)->set_enabled(false);
    on_turn_off:
      lambda: |-
        id(rgb8x32)->set_enabled(true);
```

## Logs

```text
/config/esphome/aw-clock.yaml: In lambda function:
/config/esphome/aw-clock.yaml:213:16: error: 'class esphome::EHMTX' has no member named 'set_enabled'
         id(rgb8x32)->set_enabled(false);
                ^~~~~~~~~~~
/config/esphome/aw-clock.yaml: In lambda function:
/config/esphome/aw-clock.yaml:216:16: error: 'class esphome::EHMTX' has no member named 'set_enabled'
         id(rgb8x32)->set_enabled(true);
                ^~~~~~~~~~~
```

lubeda, 2023-05-22:
Hi,
the documentation was wrong. This is part of the display component.

Please try:
```
light:
  - platform: neopixelbus
    id: ehmtx_light
    type: GRB
#    internal: true
    variant: WS2812
    pin: $matrix_pin
    num_leds: 256
    color_correct: [30%, 30%, 30%]
    gamma_correct: 2.0
    name: "$device_friendly_name Light"
    restore_mode: ALWAYS_OFF
    on_turn_on:
      lambda: |-
        id(ehmtx_light)->set_enabled(false);
    on_turn_off:
      lambda: |-
        id(ehmtx_light)->set_enabled(true);
```

thewolf-oz, 2023-05-23:
Thanks, this compiles now, will check functionality when I get home from work tonight :)

thewolf-oz, 2023-05-23:
Thanks… confirmed this is working as expected now

closed 2023-05-23 by thewolf-oz