lubeda/EspHoMaTriXv2 #20: [QUESTION] What are the required dependencies and versions of those dependencies

mmoerdijk, 2023-05-22:
# Question
Hey fist of all thank you very much for this project!

Could you add some documentation on what versions are required for this to run? 

I have the feeling that I'm running with the wrong version of one of the dependencies.

## Additional information

- used Hardware:
  - ESP8266

  - EspHoMaTriXv2 version: main / develop (both versions have this problem)
  
  - esphome(dev)

## Logs

```text
 Reading configuration /config/esphome/led-display.yaml...
WARNING 
WARNING This is a beta version of https://github.com/lubeda/EspHoMaTriXv2
WARNING 
INFO Detected timezone 'Europe/Amsterdam'
INFO Generating C++ source...
INFO Preparing icons, this may take some seconds.
INFO EsphoMaTrix: wrote html-file with icon preview: /config/esphome/led-display.html
INFO List of icons for e.g. blueprint:

["xani","xsta","pipo","d180","d080","fullscreen","samplegif",]

INFO Compiling app...
Processing led-display (board: d1_mini; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- NeoPixelBus @ 2.7.3
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/ehmtxv2/EHMTX.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/core/component.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/core/component_iterator.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/core/controller.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/core/entity_base.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/core/helpers.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/core/log.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/core/scheduler.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/core/string_ref.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/core/util.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/main.cpp.o
Generating LD script /data/led-display/.pioenvs/led-display/ld/local.eagle.app.v6.common.ld
Compiling /data/led-display/.pioenvs/led-display/lib67b/ESP8266WiFi/BearSSLHelpers.cpp.o
src/esphome/components/ehmtxv2/EHMTX.cpp: In member function 'void esphome::EHMTX::bitmap_screen(std::string, int, int)':
src/esphome/components/ehmtxv2/EHMTX.cpp:109:29: error: 'JSON_ARRAY_SIZE' was not declared in this scope; did you mean 'LWIP_ARRAYSIZE'?
  109 |     const size_t CAPACITY = JSON_ARRAY_SIZE(256);
      |                             ^~~~~~~~~~~~~~~
      |                             LWIP_ARRAYSIZE
src/esphome/components/ehmtxv2/EHMTX.cpp:110:5: error: 'StaticJsonDocument' was not declared in this scope
  110 |     StaticJsonDocument<CAPACITY> doc;
      |     ^~~~~~~~~~~~~~~~~~
src/esphome/components/ehmtxv2/EHMTX.cpp:110:34: error: 'doc' was not declared in this scope
  110 |     StaticJsonDocument<CAPACITY> doc;
      |                                  ^~~
src/esphome/components/ehmtxv2/EHMTX.cpp:111:5: error: 'deserializeJson' was not declared in this scope
  111 |     deserializeJson(doc, text);
      |     ^~~~~~~~~~~~~~~
src/esphome/components/ehmtxv2/EHMTX.cpp:112:5: error: 'JsonArray' was not declared in this scope
  112 |     JsonArray array = doc.as<JsonArray>();
      |     ^~~~~~~~~
src/esphome/components/ehmtxv2/EHMTX.cpp:115:10: error: 'JsonVariant' was not declared in this scope
  115 |     for (JsonVariant v : array)
      |          ^~~~~~~~~~~
src/esphome/components/ehmtxv2/EHMTX.cpp:129:22: error: expected ')' before ';' token
  129 |     screen->text = "";
      |                      ^
      |                      )
src/esphome/components/ehmtxv2/EHMTX.cpp:115:9: note: to match this '('
  115 |     for (JsonVariant v : array)
      |         ^
src/esphome/components/ehmtxv2/EHMTX.cpp:130:5: error: 'screen' was not declared in this scope; did you mean 'strlen'?
  130 |     screen->endtime = this->clock->now().timestamp + lifetime * 60;
      |     ^~~~~~
      |     strlen
Compiling /data/led-display/.pioenvs/led-display/lib67b/ESP8266WiFi/CertStoreBearSSL.cpp.o
Compiling /data/led-display/.pioenvs/led-display/lib67b/ESP8266WiFi/ESP8266WiFi.cpp.o
Compiling /data/led-display/.pioenvs/led-display/lib67b/ESP8266WiFi/ESP8266WiFiAP.cpp.o
Compiling /data/led-display/.pioenvs/led-display/lib67b/ESP8266WiFi/ESP8266WiFiGeneric.cpp.o
*** [/data/led-display/.pioenvs/led-display/src/esphome/components/ehmtxv2/EHMTX.cpp.o] Error 1
```
This is my yaml config, i changed the time component and removed the `wifi` section 
```text
api:

# esphome config file with all features
substitutions:
  devicename: led-display
  ledpin: D3
  loglevel: DEBUG

external_components:
  - source:
      type: git
      url: https://github.com/lubeda/EspHoMaTriXv2
      ref: develop # optional select a special branch or tag

esphome:
  name: $devicename
  comment: 8x32 RGB Display in kitchen
      
binary_sensor:
  - platform: status
    name: "$devicename Status"

esp8266:
  board: d1_mini

font: 
  - file: _fonts/EHMTXv2.ttf
    id: default_font
    size: 16
    glyphs:  |
       !"%()+*=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz€@üöäÜÖÄß
  
logger:
  level: WARN

light:
  - platform: neopixelbus
    id: ehmtx_light
    type: GRB
    variant: WS2812
    pin: $ledpin
    num_leds: 256
    color_correct: [30%, 30%, 30%]
    name: "$devicename Light"
    restore_mode: ALWAYS_OFF
    effects:
      - e131:
          universe: 1
          channels: RGB
    on_turn_on:
      lambda: |-
         id(ehmtx_display)->set_enabled(false);
    on_turn_off:
       lambda: |-
         id(ehmtx_display)->set_enabled(true);

e131:
  method: multicast # default: register E1.31 to Multicast group

time:
  - platform: homeassistant
    id: ehmtx_time
      
display:
  - platform: addressable_light
    id: ehmtx_display
    addressable_light_id: ehmtx_light
    width: 32
    height: 8
    pixel_mapper: |-
      if (x % 2 == 0) {
        return (x * 8) + y;
      }
      return (x * 8) + (7 - y);
    rotation: 0°
    update_interval: 16ms
    auto_clear_enabled: true
    lambda: |-
      id(rgb8x32)->tick();
      id(rgb8x32)->draw();

ehmtxv2:
  id: rgb8x32
  show_seconds: true
  rainbow_interval: 32
  matrix_component: ehmtx_display
  time_component: ehmtx_time
  icons2html: true
  default_font_id: default_font
  special_font_id: default_font
  special_font_yoffset: 6
  default_font_yoffset: 6
  icons: 
    - id: xani
      lameid: 6075
    - id: xsta
      lameid: 11236
    - url: https://developer.lametric.com/content/apps/icon_thumbs/48720.gif
      pingpong: true
      id: pipo
      frame_duration: 300
    - lameid: 5965
      frame_duration: 180
      id: d180
    - lameid: 5965
      frame_duration: 80
      id: d080
    - id: fullscreen
      file: icons/sample8x32.gif
    - id: samplegif
      file: icons/sample8x8.gif
```

## Additional context

Add any other context about the problem here.

lubeda, 2023-05-22:
Which version of esphome do you use? It should compile on esphome 2023.5.0 or newer.

jblb, 2023-05-22:
same issue here with esphome 2023.5.0

mmoerdijk, 2023-05-23:
> Which version of esphome do you use? It should compile on esphome 2023.5.0 or newer.

I tried with the dev version an 2023.5.3 both have the same problem.

lubeda, 2023-05-23:
crazy! How are you using esphome? Is it installed as add-on in the home assistant os (like I use it) or do you use a separate esphome docker? An wich platform does it run (i.e. raspberry or x86...)

jpbede, 2023-05-23:
I think that it does not work with esphome version 2023.5.3. With esphome version 2023.5.2 I've not had any problems.

jpbede, 2023-05-23:
Now, after downgrading the esphome addon, I still have the same problem.

jblb, 2023-05-23:
here 
```
> esphome version
Version: 2023.5.0

```
esphome installed from command line in a virtual environment on ubuntu 20.04  x84_64

mmoerdijk, 2023-05-23:
> crazy! How are you using esphome? Is it installed as add-on in the home assistant os (like I use it) or do you use a separate esphome docker? An wich platform does it run (i.e. raspberry or x86...)

Yes as an addon in HomeAssistant.

> Now, after downgrading the esphome addon, I still have the same problem.
> 
> Edit: Ok, with the git tag `beta` everything works fine :)

git tag beta from esphome or EspHoMaTriXv2 ?

jpbede, 2023-05-23:
> git tag beta from esphome or EspHoMaTriXv2 ?

Sry, from EspHoMaTriXv2.

> crazy! How are you using esphome? Is it installed as add-on in the home assistant os (like I use it) or do you use a separate esphome docker? An wich platform does it run (i.e. raspberry or x86...)

Running esphome as addon (on HAOS) on x86

mmoerdijk, 2023-05-23:
> Sry, from EspHoMaTriXv2.
>
>
> Running esphome as addon (on HAOS) on x86

Thanks, can confirm it works with that!

lubeda, 2023-05-23:
So, I think this is more a problem with esphome than with my component.  But I will investigate, because this will be a new feature, that enables you to dynamically write to the display. The version tagged with "beta" does not have this feature.

My local esphome (hassos on x64) works fine, the GitHub CI/CD pipeline works fine. !?! 😱 !?!

I need more background information about your local system.

Can you provide a complete log file after the step shown in the screenshot?

[image]

Normally, this will reinstall all dependencies, including this: ` ArduinoJson @ 6.18.5`

mmoerdijk, 2023-05-23:
1. I cleaned the build files:
2. Build output,
with
```yaml
external_components:
  - source:
      type: git
      url: https://github.com/lubeda/EspHoMaTriXv2
      ref: develop # optional select a special branch or tag
```
and esphome beta

```text
INFO Reading configuration /config/esphome/led-display.yaml...
WARNING 
WARNING This is a beta version of https://github.com/lubeda/EspHoMaTriXv2
WARNING 
INFO Detected timezone 'Europe/Amsterdam'
INFO Generating C++ source...
INFO Preparing icons, this may take some seconds.
INFO EsphoMaTrix: wrote html-file with icon preview: /config/esphome/led-display.html
INFO List of icons for e.g. blueprint:

["xani","xsta","pipo","d180","d080","fullscreen","samplegif",]

INFO Compiling app...
Processing led-display (board: d1_mini; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
Library Manager: Installing makuna/NeoPixelBus @ 2.7.3
INFO Installing makuna/NeoPixelBus @ 2.7.3
Unpacking  [####################################]  100%
Library Manager: NeoPixelBus@2.7.3 has been installed!
INFO NeoPixelBus@2.7.3 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- NeoPixelBus @ 2.7.3
|   |-- SPI @ 1.0
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/addressable_light/addressable_light_display.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/api/api_connection.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/api/api_server.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/api/list_entities.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/api/proto.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/api/user_services.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/binary_sensor/automation.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/binary_sensor/binary_sensor.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/binary_sensor/filter.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/display/display_buffer.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/e131/e131.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/e131/e131_addressable_light_effect.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/e131/e131_packet.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/ehmtxv2/EHMTX.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/ehmtxv2/EHMTX_icons.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/ehmtxv2/EHMTX_queue.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/esp8266/core.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/esp8266/gpio.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/esp8266/preferences.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/homeassistant/time/homeassistant_time.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/light/addressable_light.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/light/automation.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/light/esp_color_correction.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/light/esp_hsv_color.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/light/esp_range_view.cpp.o
[... 16 lines cut ...]
      |     ^~~~~~~~~
src/esphome/components/ehmtxv2/EHMTX.cpp:115:10: error: 'JsonVariant' was not declared in this scope
  115 |     for (JsonVariant v : array)
      |          ^~~~~~~~~~~
src/esphome/components/ehmtxv2/EHMTX.cpp:129:22: error: expected ')' before ';' token
  129 |     screen->text = "";
      |                      ^
      |                      )
src/esphome/components/ehmtxv2/EHMTX.cpp:115:9: note: to match this '('
  115 |     for (JsonVariant v : array)
      |         ^
src/esphome/components/ehmtxv2/EHMTX.cpp:130:5: error: 'screen' was not declared in this scope; did you mean 'strlen'?
  130 |     screen->endtime = this->clock->now().timestamp + lifetime * 60;
      |     ^~~~~~
      |     strlen
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/light/light_call.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/light/light_json_schema.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/light/light_output.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/light/light_state.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/logger/logger.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/md5/md5.cpp.o
Compiling /data/led-display/.pioenvs/led-display/src/esphome/components/mdns/mdns_component.cpp.o
*** [/data/led-display/.pioenvs/led-display/src/esphome/components/ehmtxv2/EHMTX.cpp.o] Error 1
========================== [FAILED] Took 8.77 seconds ==========================
```

lubeda, 2023-05-23:
@jpbede: Thanx, I will take a look on how to include the JSON without the webserver component.

lubeda, 2023-05-23:
@jpbede:  Thank you very much, I didn't see the difference in the config!!!

If you use this section, you can check if it works.

```
external_components:
  - source:
        type: git
        url: https://github.com/lubeda/EsphoMaTrixV2
        ref: "develop"
    refresh: 60s 
    components: [ ehmtxv2 ]   
```

jpbede, 2023-05-23:
@lubeda: Works like a charm, thank you!

jpbede, 2023-05-23:
@lubeda Hmm tried it now on the device itself, but that doesn't work so well.
The code compiles but the ESP is stuck in a boot loop with the following error:

```

[I][logger:262]: Log initialized
[I][EHMTXv2:847]: set_brightness 80 => 31.37 %
[I][EHMTXv2:833]: weekstart: monday
[I][EHMTXv2:820]: show day of week
[I][EHMTXv2:785]: show date
[I][EHMTXv2:811]: don't show seconds
[I][app:029]: Running through setup()...
[I][i2c.arduino:183]: Performing I2C bus recovery
[I][EHMTXv2:847]: set_brightness 74 => 29.02 %
[I][wifi:446]: - 'IoT' [redacted]▂▄▆█
[I][wifi:446]: - 'IoT' [redacted]▂▄▆█
[I][wifi:446]: - 'IoT' [redacted]▂▄▆█
[I][wifi:446]: - 'IoT' [redacted]▂▄▆█
[I][wifi:258]: WiFi Connecting to 'IoT'...
[I][wifi:519]: WiFi Connected!
[W][ota:103]: Last Boot was an unhandled reset, will proceed to safe mode in 5 restarts
[I][app:062]: setup() finished successfully!
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400dc9e8  PS      : 0x00060f30  A0      : 0x800ebdbc  A1      : 0x3ffb2630  
A2      : 0x3ffb4d50  A3      : 0x3ffb5350  A4      : 0x3ffc3b08  A5      : 0x646d060d  
A6      : 0x00004d83  A7      : 0xb7391c01  A8      : 0x800dc9d6  A9      : 0x3ffb2610  
A10     : 0x000023f0  A11     : 0x00000001  A12     : 0x3f4ccccd  A13     : 0x3ffb264c  
A14     : 0x3ffb2650  A15     : 0x3ffb2654  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xb7391c17  LBEG    : 0x40084bb1  LEND    : 0x40084bb9  LCOUNT  : 0x00000027  

Backtrace:0x400dc9e5:0x3ffb26300x400ebdb9:0x3ffb26c0 0x400dabf7:0x3ffb26e0 0x400dac13:0x3ffb2700 0x400d3568:0x3ffb2720 0x400d357a:0x3ffb2740 0x40184bad:0x3ffb2760 0x400ebab3:0x3ffb2780 0x400e95f9:0x3ffb27d0 0x400ec5b6:0x3ffb2800 0x400fb961:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...
```

jblb, 2023-05-24:
Hi @lubeda,
witch develop branch does we need to use: you component or esphome one ?
Here nothing works (infinite reboot) whith 
```
external_components:
  - source:
      type: git
      url: https://github.com/lubeda/EspHoMaTriXv2
      ref: "develop"
    refresh: 60s
    components: [ ehmtxv2 ]
```

lubeda, 2023-05-25:
@jblb 
Hi, the infinite reboot problem is very tricky. I think it has something to do with th limited RAM. There is a way to get it working but only/sadly on ESP32.  I will have a deeper look at the weekend.

closed 2023-05-25 by lubeda