lubeda/EspHoMaTriXv2 #21: [BUG] can't compile

jblb, 2023-05-22:
# Bug report

## Describe the bug
can't compile the simple code exemple.
seem to be related to commit 1036509, at least part of code how generate error comme from this commit

here is the log
```
Compiling .pioenvs/esphomatrix/src/esphome/components/ehmtxv2/EHMTX.cpp.o
Compiling .pioenvs/esphomatrix/src/esphome/components/homeassistant/time/homeassistant_time.cpp.o
Compiling .pioenvs/esphomatrix/src/esphome/components/light/addressable_light.cpp.o
Compiling .pioenvs/esphomatrix/src/esphome/components/light/automation.cpp.o
Compiling .pioenvs/esphomatrix/src/esphome/components/light/esp_color_correction.cpp.o
Compiling .pioenvs/esphomatrix/src/esphome/components/light/esp_hsv_color.cpp.o
Compiling .pioenvs/esphomatrix/src/esphome/components/light/esp_range_view.cpp.o
src/esphome/components/ehmtxv2/EHMTX.cpp: In member function 'void esphome::EHMTX::bitmap_screen(std::string, int, int)':
src/esphome/components/ehmtxv2/EHMTX.cpp:109:29: error: 'JSON_ARRAY_SIZE' was not declared in this scope; did you mean 'LWIP_ARRAYSIZE'?
  109 |     const size_t CAPACITY = JSON_ARRAY_SIZE(256);
      |                             ^~~~~~~~~~~~~~~
      |                             LWIP_ARRAYSIZE
src/esphome/components/ehmtxv2/EHMTX.cpp:110:5: error: 'StaticJsonDocument' was not declared in this scope
  110 |     StaticJsonDocument<CAPACITY> doc;
      |     ^~~~~~~~~~~~~~~~~~
src/esphome/components/ehmtxv2/EHMTX.cpp:110:34: error: 'doc' was not declared in this scope
  110 |     StaticJsonDocument<CAPACITY> doc;
      |                                  ^~~
src/esphome/components/ehmtxv2/EHMTX.cpp:111:5: error: 'deserializeJson' was not declared in this scope
  111 |     deserializeJson(doc, text);
      |     ^~~~~~~~~~~~~~~
src/esphome/components/ehmtxv2/EHMTX.cpp:112:5: error: 'JsonArray' was not declared in this scope
  112 |     JsonArray array = doc.as<JsonArray>();
      |     ^~~~~~~~~
src/esphome/components/ehmtxv2/EHMTX.cpp:115:10: error: 'JsonVariant' was not declared in this scope
  115 |     for (JsonVariant v : array)
      |          ^~~~~~~~~~~
src/esphome/components/ehmtxv2/EHMTX.cpp:129:22: error: expected ')' before ';' token
  129 |     screen->text = "";
      |                      ^
      |                      )
src/esphome/components/ehmtxv2/EHMTX.cpp:115:9: note: to match this '('
  115 |     for (JsonVariant v : array)
      |         ^
src/esphome/components/ehmtxv2/EHMTX.cpp:130:5: error: 'screen' was not declared in this scope; did you mean 'strlen'?
  130 |     screen->endtime = this->clock->now().timestamp + lifetime * 60;
      |     ^~~~~~
      |     strlen
Compiling .pioenvs/esphomatrix/src/esphome/components/light/light_call.cpp.o
Compiling .pioenvs/esphomatrix/src/esphome/components/light/light_json_schema.cpp.o
Compiling .pioenvs/esphomatrix/src/esphome/components/light/light_output.cpp.o
*** [.pioenvs/esphomatrix/src/esphome/components/ehmtxv2/EHMTX.cpp.o] Error 1
```

jblb, 2023-05-22:
same issue as https://github.com/lubeda/EspHoMaTriXv2/issues/20
this one can be closed.

jblb, 2023-05-22:
c

closed 2023-05-22 by jblb

lubeda, 2023-05-23:
crazy! How are you using esphome? Is it installed as add-on in the home assistant os (like I use it) or do you use a separate esphome docker? An wich platform does it run (i.e. raspberry or x86...)