lubeda/EspHoMaTriXv2 #22: [BUG] infinite reboot

lubeda, 2023-05-25:
# Bug report

## Describe the bug

Add a description of the bug. Detail the expected behavior in contrast with the behavior you're observing.

## Additional information

- used Hardware:
  - ESP32 or ESP8266
  - EspHoMaTriXv2 version: [e.g. 2023.5.0]

## To Reproduce

Steps to reproduce the behavior:

## Expected behavior

A clear and concise description of what you expected to happen.

## Configuration

```text
(optional) The YAML you used in epshome without any password
```

## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Services calls

```text
(optional) The YAML of your service calls
```

lubeda, 2023-05-25:
I think this issue is related to https://github.com/esphome/issues/issues/1960

On ESP32 platforms there is a workaround by using `PROGMEM` but for ESP8266 with even fewer RAM I don't know yet

darkpoet78, 2023-05-25:
I have an old Awtrix that absolutely works using develop branch.  Just flashed now.  I had to remove a lot of the features (no buzzer, no TM chip) but it works.

lubeda, 2023-06-09:
I can't resolve this issue because the cause is not only my component. The ESP-platform is limited in resources, especially RAM. During runtime this can cause problems, depending on what configuration you use, memory intensive components, like web server and MQTT will increase the change of this infinite reboots. I did not find a way to probably debug this issue, but it is there. Since [EspHoMaTriXv2](https://github.com/lubeda/EspHoMaTriXv2) needs a lot of RAM too, you have to reduce the components and lambdas in your YAML.

closed 2023-06-09 by lubeda