lubeda/EspHoMaTriXv2 #26: [feature request] always scroll display

Firewolf1337, 2023-06-04:
Hi,
I would like to make a round display and for that ist would be great to have to the function to let screen scroll all the time.

darkpoet78, 2023-06-04:
Has anyone even attempted this on a display that wasn't 8x32?  I'd say give it a shot... but it would take some serious math to figure out how to "map" a square matrix to a circle shape.  The LED's genuinely think they're all in one straight line.  The very definition of a matrix is a square.  Perhaps some "dead" pixels that take up an address space but don't actually light up?

Firewolf1337, 2023-06-05:
I'm sorry, I did not express myself correctly. I want to bend the 8x32 to a tube, so that it build an infinite circle.
[Like this](https://www.printables.com/de/model/217568-led-matrix-lamp)

darkpoet78, 2023-06-05:
Ah-ha. Whew.  IF I understand how ESPHome's [https://esphome.io/components/display/addressable_light.html#addressable-light](addressable_light) component works (I may not), it's not necessarily dependent on lubeda's code.  But you would have to trade a lot of the numbers in the pixel_mapper for variables that are shifted every x milliseconds or something.  And it would basically be hardcoded so there wouldn't be any stopping the scrolling.

But you'd some pretty skillful lambda coding to be able to achieve that.

Unless lubeda can somehow work that into his code...  I mean he did just triumph over RTL text so... let's wait and see.

lubeda, 2023-06-05:
Nice, but why? In your plan, you will only see 8x16 because the rest of the pixels are on the opposite side. The readability will be quite low. Did you try it all ready? What's about awtrix-light and PixelIt, perhaps do they work better for your request?

Firewolf1337, 2023-06-06:
My idea was to put it on my bar counter and/or a sidebar in the living room. Somewhere where the viewing angle is not always just the front. The edge would face the wall or somewhere where not always someone stands.

I haven't printed a case for it right now so it looks a bit scrappy but you can see the idea
https://github.com/lubeda/EspHoMaTriXv2/assets/18141890/eba5cc3b-3a4b-467f-8775-d291bcb61179

lubeda, 2023-06-07:
Ich antworte mal auf Deutsch, ist einfacher. Das Lesen des Textes klappt besser, als ich dachte. Das Durchscrollen der Icons möchte ich nicht programmieren, weil die Anforderung zu speziell ist. Ich könnte mir aber `rainbow_text_screen` und `text_screen` als Dauerscrolltext vorstellen. Die Uhrzeit bekommst Du über Templates wie in Deinem Beispiel rein.

Ich prüfe das wohlwollend...

lubeda, 2023-06-08:
Ich werde das umsetzen und dann einen kurzen Wiki-Artikel dazu verfassen.

lubeda, 2023-06-09:
will be released with 2023.6.4

closed 2023-06-09 by lubeda