lubeda/EspHoMaTriXv2 #27: [BUG] Special font doesn't display.

darkpoet78, 2023-06-04:
# Bug report

My clock screen only uses the default_font instead of the special_font.

## Describe the bug

Worked as expected on develop branch right up to a few hours ago (just updated lol)... saw you push through an update so updated again... both on main and develop branches.   Only shows the time with default_font.

## Additional information

- used Hardware:
  - Ulanzi TC-001

## Configuration

```
font:
  - file: ehmtxv2/MatrixClock-Chunky8.bdf #
    id: special_font
    #size: 16
    glyphs:  |
      ! "#$%&'()*+,-./0123456789:APMTapmt
  - file: ehmtxv2/MatrixClock-Light8.bdf #
    id: default_font
    #size: 16
    glyphs:  |
      ! "#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz°{|}~¡¢£¤¥¦§¨©ª«¬®¯±²³´µ¶·¸¹º»¼½¾¿ÀÁÂÃÄÅÆÇÈÉÊËÌÍÎÏÐÑÒÓÔÕÖ×ØÙÚÛÜÝÞßàáâãäåæçèéêëìíîïðñòóôõö÷øùúûüýþÿ
```

lubeda, 2023-06-04:
Since 2023.6.2 the clock is displayed with the default font. From my point of view, this fits, and the previous behavior was wrong. Before this release, the date and clock were indeed displayed with the special font.

I will search for a solution.

lubeda, 2023-06-04:
Please check the develop branch and the new parameter `default_clock_font: true`

You should use: default_clock_font: false to have the old behavior

darkpoet78, 2023-06-04:
Yup.  That works on develop branch well.

When compiling again, I saw:
```
/config/pixelclock-red.yaml: In lambda function:
/config/pixelclock-red.yaml:79:16: error: 'class esphome::EHMTX' has no member named 'set_clock_interval'; did you mean 'set_update_interval'?
           id(rgb8x32)->set_clock_interval(2);
                ^~~~~~~~~~~~~~~~~~
                set_update_interval
```
So I changed it, assuming it's the same function.  I checked the develop code and it looks you're trying to cleanup the code and variables a bit.  But it seems like the update_interval is getting ignored somewhere as the clock scrolls through all the icon screens in the stack before showing the clock again.

darkpoet78, 2023-06-04:
Oh and lol, not a big deal but it looks like the ehmtx♥ screen is briefly on the stack for the first minute after boot.  That's pretty nice!

darkpoet78, 2023-06-05:
Oh I'm an idiot.  I used to have that variable set by an on_boot lambda.  It's been moved to:
```
ehmtxv2:
  clock_interval: 2
```