lubeda/EspHoMaTriXv2 #35: [FEATURE REQUEST] - Transition between screens

geekofweek, 2023-06-08:
# Feature Request

### Describe the solution / feature you'd like

As it currently stands, from what I can tell, there is no configuration option to do a scroll effect between the screens, it just jumps to the new screen.  It would be nice to have either a default or configurable transition.  Example when you add a screen it will scroll top down to the new screen from the old screen, or left to right, etc.  Similar effects exist for other solutions so those can be used as another example.

geekofweek, 2023-06-08:
It definitely doesn't provide any additional usefulness outside of the "cool" factor.  Even a basic fade in/out would add a nice touch.  Configurable option would at least pull it closer to parity with other options out there.  I prefer the ESPHome route over the others

KlausHans, 2023-06-08:
All pixel display thingys are more or less schnick schnack. They don't provide any more information than other displays, more often less information imho. The main purpose is to be cool so why not make them more cool with nice transitions. :)

lubeda, 2023-06-10:
There is a slight effect in the develop branch that slowly blends in the new screen when transitioning to another screen. 
Hope this is cool enough ;-)

Configure with `blend_steps: 16`

closed 2023-06-10 by lubeda

genehand, 2023-06-16:
Tried it out, I like it! However during the work day I like it keep it low key so I turn off the date screen and it stays on the clock.  With blend_steps enabled I noticed it still "transitions" to the same screen.

chertvl, 2023-10-29:
> With blend_steps enabled I noticed it still "transitions" to the same screen.

Ive created an issue
https://github.com/lubeda/EspHoMaTriXv2/issues/126