lubeda/EspHoMaTriXv2 #36: [QUESTION] - Documentation Discussion

geekofweek, 2023-06-08:
# Question

This is more of a discussion topic than a question.  I think the documentation so far is pretty good, I was able to work my way through going from v1 to v2 and figuring out the differences.  I would like to contribute to getting the documentation a little more extensive.  

The question part comes in to where should the documentation go?  Is the README file the best place or would it be better to start porting a lot of it over to the wiki section?

Suggested additions:

- Breaking changes going from v1 to v2
  - outline all of the changes required to do the upgrade
- Detail services exposed to Home Assistant in a little more depth
- Additional Home Assistant automation examples

lubeda, 2023-06-08:
All of your additions, are helpful!
For the priority, i think the automation examples are the highest. An explanation of the services, second. After that, the changes.

I think for starters, the examples are the most practical way to start with the display and describing the changes is a more "scientific and theoretical" way to learn [EspHoMaTriXv2](https://github.com/lubeda/EspHoMaTriXv2).

KlausHans, 2023-06-08:
Hello, i am not very deep in esphome and a new user of this project. I am using an Ulanzi TC001 and the default ulanzi-easy.yaml.
Currently it loops through a clock page and a date page. I just don't get where the two screens are defined in the yaml. How can i delete these? I hoped to have a declarative, straight forward way to define pages. Like in https://github.com/joBr99/nspanel-lovelace-ui . But i think in esphome it is just different, correct?

For the beginning i want just one icon_screen, that displays the state of an entity. How is it done? The documentation lacks some easy to understand examples and more general explanations imho.

Great project by the way!

geekofweek, 2023-06-08:
@KlausHans you'll need to integrate it into home assistant via the [ESPHome integration](https://www.home-assistant.io/integrations/esphome/). From there it exposes some services to Home Assistant, you can see [my examples here](https://github.com/geekofweek/homeassistant/blob/master/automation/desktop_clock.yaml#L78).  In short you call up the `esphome.DEVICE_NAME_icon_screen` passing the various variables like icon, text, duration, etc. to it

@lubeda has created some [blueprints to get started](https://github.com/lubeda/EspHoMaTriXv2/tree/main/copy2esphome), but it can also be easily done via the default automation editor using `call_service` as the action.  You will need to trigger an event to send a new screen to the display, this can be anything.  an input_boolean that you just toggle on and off for example.

lubeda, 2023-06-09:
@KlausHans "Currently it loops through a clock page and a date page." These are "build in", The main purpose is the time display so if there are no other screens it will loop through time/date. If you don't need the date-display you can disable it with the parameter `show_date: false` in the yaml. During runtime you can remove them of the queue with:

```yaml
service: esphome.m5mic_del_screen
data:
  icon_name: "*"
  mode: 3
```

```yaml
service: esphome.m5mic_del_screen
data:
  icon_name: "*"
  mode: 2
```

and add them with

```yaml
service: esphome.m5mic_clock_screen
data:
  default_font: true
  lifetime: 1440
  screen_time: 10
  r: 192
  g: 192
  b: 192
```

and

service: esphome.m5mic_date_screen
data:
  default_font: true
  lifetime: 1440
  screen_time: 5
  r: 192
  g: 192
  b: 192
```

closed 2023-06-09 by geekofweek