lubeda/EspHoMaTriXv2 #42: [BUG] del_screen with wildcard not working

jdohnstad, 2023-06-14:
# Bug report

## Describe the bug

When calling the `del_screen` service through home assistant with a `icon_name` wildcard (i.e. `weather_*`) it doesn't actually delete any screens. When calling the same service with a fully specified `icon_name`, screen (i.e. `weather_partlycloudy` is deleted as expected.

## Additional information

  - Ulanzi TC001
  - 2023.6.4
  - ESPHome 2023.5.5

## To Reproduce

Configure an icon such as `weather_partlycloudy`. Add an `icon_screen` to the queue with with `esphome.<DEVICENAME>_icon_screen` service. Attempt to delete it via the `esphome.<DEVICENAME>_del_screen` service using a wildcard that should match such as `weather_*`.

```yaml
  - service: esphome.ulanzi_icon_screen
    data:
      icon_name: weather_partlycloudy
      text: some text
      default_font: true
      lifetime: 180
      screen_time: 2
      r: 255
      g: 255
      b: 255
    enabled: true
  - service: esphome.ulanzi_del_screen
    data:
      icon_name: weather_*
      mode: 5
```

## Expected behavior

Screen should be removed from queue.

## Configuration

<details>
<summary>Full esphome.yaml</summary>

```yaml
---
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 10.0.0.227
    gateway: !secret gateway
    subnet: !secret subnet
  ap:
    ssid: "$devicename Fallback Hotspot"
    password: !secret ap_password

captive_portal:
logger:
ota:
  password: !secret ota_password

substitutions:
  devicename: ulanzi
  matrix_pin: GPIO32
  buzzer_pin: GPIO15
  battery_pin: GPIO34
  ldr_pin: GPIO35
  left_button_pin: GPIO26
  mid_button_pin: GPIO27
  right_button_pin: GPIO14
[... 500 lines cut ...]
    icon_name: weather_partlycloudy
    text: some text
    default_font: true
    lifetime: 180
    screen_time: 2
    r: 255
    g: 255
    b: 255

# observe "weather_partlycloudy" screen has been added

- service: esphome.ulanzi_del_screen
  data:
    icon_name: weather_*
    mode: 5 

# observe "weather_partlycloudy" screen has not been deleted

- service: esphome.ulanzi_del_screen
  data:
    icon_name: weather_partlycloudy
    mode: 5

# observe "weather_partlycloudy" screen has finally been deleted
```

lubeda, 2023-06-14:
Fixed in the "develop" branch

jdohnstad, 2023-06-15:
I can confirm. Nice job!

closed 2023-06-15 by jdohnstad