lubeda/EspHoMaTriXv2 #47: scroll_small_text: false does not work

andrew-codechimp, 2023-06-20:
# Bug report

## Describe the bug
The flag scroll_small_text does not work correctly.  Setting it to false small text still scrolls

## Additional information
I attempted to fix it as did @joncar, see links below but neither quite get's it right

https://github.com/lubeda/EspHoMaTriXv2/pull/46
https://github.com/joncar/EspHoMaTriXv2/tree/user/joncar/01-fix-scroll

- used Hardware:
  - ESP32
  - EspHoMaTriXv2 version: [e.g. 2023.6.5]

## To Reproduce

Steps to reproduce the behavior:
Set scroll_small_text: false and add a screen with small text

## Expected behavior

A clear and concise description of what you expected to happen.
Small text will not scroll when set to false

## Configuration

```text
(optional) The YAML you used in epshome without any password
```

## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Services calls

```text
(optional) The YAML of your service calls
```

andrew-codechimp, 2023-06-20:
I tried it but it's inverted, setting scroll_small_text: false still scrolled small text, you have to set it to true not to scroll.

joncar, 2023-06-20:
I don't see how. Strange. I just tested `scroll_small_text: true`, `scroll_small_text: false`, and with `scroll_small_text` unspecified and all behaved how I expected.

joncar, 2023-06-20:
Any chance your edit to `EHMTX_queue.cpp` was still in your esphome code when you tested? That would invert the flag.

andrew-codechimp, 2023-06-20:
Yes, I did leave my edit in EHMT_queue.cpp, which I didn't think I did, so I'm an idiot :)  Blame it on early morning tinkering.

Are you going to submit a PR with your fix?

joncar, 2023-06-20:
Yup, I'll submit the PR to fix this issue.

lubeda, 2023-06-21:
merged in main. Thanx for the support!

PR #48 (2023-06-21): scroll_small_text: false does not work
Fix bool defines so that ifdef responds to a yaml false correctly. Fixes #47.
closed 2023-06-21 by lubeda