lubeda/EspHoMaTriXv2 #51: [QUESTION] Configure display times for clock and date.

chrisb86, 2023-06-23:
# Question

I'm using EspHoMaTriXv2 on a Ulanzi TC001. I want to display the time for 55 seconds and the date for 5 seconds in a loop. I don't understand, how I can configure the display times. I tried _clock_interval: 55_ but it doesn't seem to help.

What's the corerct way?

## Additional information

- used Hardware:
  - Ulanzi TC001
  - EspHoMaTriXv2 version: latest

lubeda, 2023-06-23:
Hi, I can't test, since I am on holiday. But I think the easiest way is by a home assistant script like this:

```yaml
alias: ulanzi 55-5 clock-date
sequence:
  - service: esphome.ulanzi_del_screen
    data:
      icon_name: "*"
      mode: 2
  - service: esphome.ulanzi_clock_screen
    data:
      default_font: false
      lifetime: 1440
      screen_time: 55
      r: 192
      g: 192
      b: 192
  - service: esphome.ulanzi_del_screen
    data:
      icon_name: "*"
      mode: 3
  - service: esphome.ulanzi_date_screen
    data:
      default_font: false
      lifetime: 1440
      screen_time: 5
      r: 192
      g: 192
      b: 192
mode: single
icon: mdi:clock-digital
```
This deletes the default date and clock display and fills the loop with a 55 sec clock and a 5 sec date screen. The lifetime is one day, so the script must run once a day or on boot.

This is also possible in the esphome YAML, but I can't provide a sample.

closed 2023-06-23 by lubeda

chrisb86, 2023-06-23:
This works for now. Thank you.
I think it would be a nice idea to make this base functionality easily configurable in the firmware.

Enjoy your holidays and thank you for your great work!

dbuezas, 2023-07-12:
This did it for me:
```yaml
esphome:
 [...]
  on_boot:
    then:
      - ds1307.read_time:
      - lambda: |-
          id(rgb8x32)->set_clock_time(2);
  ```
  
  +1 on the feature request to set both clock and date time via yaml, and also removing the `Ehmtx❤️` and fw version messages at startup