lubeda/EspHoMaTriXv2 #52: [QUESTION] Compilation errors

fmarkf, 2023-06-23:
# Question

Trying to compile and getting the errors. Log is below. Any idea on how to fix it?

Home Assistant 2023.5.4
Supervisor 2023.06.2
Operating System 10.3
Frontend 20230503.3 - latest

## Additional information

- used Hardware:
  - ESP32 or ESP8266 (tried both)

  - EspHoMaTriXv2 version: tried Develop and Main branches

## Logs

```text
In file included from src/esphome.h:26,
                 from src/esphome/components/ehmtxv2/EHMTX.cpp:1:
src/esphome/components/ehmtxv2/EHMTX.h:109:5: error: 'time' does not name a type
     time::RealTimeClock *clock;
     ^~~~
src/esphome/components/ehmtxv2/EHMTX.h:148:20: error: 'time' is not a class, namespace, or enumeration
     void set_clock(time::RealTimeClock *clock);
                    ^~~~
src/esphome/components/ehmtxv2/EHMTX.h:148:40: error: expected ',' or '...' before '*' token
     void set_clock(time::RealTimeClock *clock);
                                        ^
In file included from src/esphome.h:26,
                 from src/esphome/components/ehmtxv2/EHMTX_queue.cpp:1:
src/esphome/components/ehmtxv2/EHMTX.h:109:5: error: 'time' does not name a type
     time::RealTimeClock *clock;
     ^~~~
src/esphome/components/ehmtxv2/EHMTX.h:148:20: error: 'time' is not a class, namespace, or enumeration
     void set_clock(time::RealTimeClock *clock);
                    ^~~~
src/esphome/components/ehmtxv2/EHMTX.h:148:40: error: expected ',' or '...' before '*' token
     void set_clock(time::RealTimeClock *clock);
                                        ^
src/esphome/components/ehmtxv2/EHMTX.cpp: In member function 'void esphome::EHMTX::bitmap_screen(std::__cxx11::string, int, int)':
src/esphome/components/ehmtxv2/EHMTX.cpp:142:29: error: 'class esphome::EHMTX' has no member named 'clock'
     screen->endtime = this->clock->now().timestamp + lifetime * 60;
                             ^~~~~
src/esphome/components/ehmtxv2/EHMTX.cpp: In member function 'void esphome::EHMTX::bitmap_small(std::__cxx11::string, std::__cxx11::string, int, int, bool, int, int, int)':
src/esphome/components/ehmtxv2/EHMTX.cpp:177:29: error: 'class esphome::EHMTX' has no member named 'clock'
     screen->endtime = this->clock->now().timestamp + lifetime * 60;
                             ^~~~~
src/esphome/components/ehmtxv2/EHMTX.cpp: In member function 'void esphome::EHMTX::blank_screen(int, int)':
src/esphome/components/ehmtxv2/EHMTX.cpp:390:26: error: 'class esphome::EHMTX' has no member named 'clock'
     scr->endtime = this->clock->now().timestamp + lifetime * 60;
                          ^~~~~
src/esphome/components/ehmtxv2/EHMTX.cpp: In member function 'virtual void esphome::EHMTX::update()':
src/esphome/components/ehmtxv2/EHMTX.cpp:397:17: error: 'class esphome::EHMTX' has no member named 'clock'
       if (this->clock->now().is_valid())
                 ^~~~~
src/esphome/components/ehmtxv2/EHMTX.cpp: In member function 'void esphome::EHMTX::force_screen(std::__cxx11::string, int)':
src/esphome/components/ehmtxv2/EHMTX.cpp:432:44: error: 'class esphome::EHMTX' has no member named 'clock'
             this->next_action_time = this->clock->now().timestamp;
                                            ^~~~~
src/esphome/components/ehmtxv2/EHMTX.cpp: In member function 'uint8_t esphome::EHMTX::find_oldest_queue_element()':
src/esphome/components/ehmtxv2/EHMTX.cpp:442:30: error: 'class esphome::EHMTX' has no member named 'clock'
     time_t last_time = this->clock->now().timestamp;
                              ^~~~~
src/esphome/components/ehmtxv2/EHMTX.cpp: In member function 'uint8_t esphome::EHMTX::find_last_clock()':
src/esphome/components/ehmtxv2/EHMTX.cpp:463:25: error: 'class esphome::EHMTX' has no member named 'clock'
       time_t ts = this->clock->now().timestamp;
                         ^~~~~
src/esphome/components/ehmtxv2/EHMTX.cpp: In member function 'void esphome::EHMTX::remove_expired_queue_element()':
[... 99 lines cut ...]
                                      ^~~~~
src/esphome/components/ehmtxv2/EHMTX_queue.cpp:212:59: error: 'class esphome::EHMTX' has no member named 'clock'
                                            this->config_->clock->now());
                                                           ^~~~~
src/esphome/components/ehmtxv2/EHMTX_queue.cpp:213:31: error: 'class esphome::EHMTX' has no member named 'clock'
           if ((this->config_->clock->now().second % 2 == 0) && this->config_->show_seconds)
                               ^~~~~
src/esphome/components/ehmtxv2/EHMTX_queue.cpp:229:28: error: 'class esphome::EHMTX' has no member named 'clock'
         if (this->config_->clock->now().is_valid())
                            ^~~~~
src/esphome/components/ehmtxv2/EHMTX_queue.cpp:232:38: error: 'class esphome::EHMTX' has no member named 'clock'
           time_t ts = this->config_->clock->now().timestamp;
                                      ^~~~~
src/esphome/components/ehmtxv2/EHMTX_queue.cpp:234:59: error: 'class esphome::EHMTX' has no member named 'clock'
                                            this->config_->clock->now());
                                                           ^~~~~
src/esphome/components/ehmtxv2/EHMTX_queue.cpp:235:31: error: 'class esphome::EHMTX' has no member named 'clock'
           if ((this->config_->clock->now().second % 2 == 0) && this->config_->show_seconds)
                               ^~~~~
*** [/data/espmatrix01/.pioenvs/espmatrix01/src/esphome/components/ehmtxv2/EHMTX_queue.cpp.o] Error 1
*** [/data/espmatrix01/.pioenvs/espmatrix01/src/esphome/components/ehmtxv2/EHMTX_icons.cpp.o] Error 1
*** [/data/espmatrix01/.pioenvs/espmatrix01/src/esphome/components/ehmtxv2/EHMTX.cpp.o] Error 1
```

## Additional context

lubeda, 2023-06-23:
Fixed with the PR from frittes many thanx for the support!!

closed 2023-06-23 by lubeda

frittes, 2023-06-23:
Please try again, should be fixed.

fmarkf, 2023-06-23:
Yes, it's fixed now!
Thank you very much!