lubeda/EspHoMaTriXv2 #55: [BUG]

vitalogy, 2023-07-01:
# Bug report

## Describe the bug

Wanted to use my Awtrix with EspHoMaTriXv2, but the esp crashes on boot with or without the connected awtrix.
I've tried different settings in the yml file, but it keeps crashing.
Additional components have already been deactivated.

The esp is working with other firmwares.

```
[01:14:32][I][EHMTXv2:983]: set_brightness 80 => 31.37 %
[01:14:32]Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
```

Add a description of the bug. Detail the expected behavior in contrast with the behavior you're observing.

## Additional information

- used Hardware:
  - ESP32 ([wemos_d1_mini32](https://docs.platformio.org/en/latest/boards/espressif32/wemos_d1_mini32.html)) 
  - EspHoMaTriXv2 version: git

## To Reproduce

Steps to reproduce the behavior:

## Expected behavior

A clear and concise description of what you expected to happen.

## Configuration

**awtrix.yml**
```text
substitutions:
  devicename: awtrix
  friendly_name: Awtrix LED Matrix
  board: wemos_d1_mini32
#  ldr_pin: GPIO36
  matrix_pin: GPIO21
#  left_button_pin: GPIO26
#  mid_button_pin: GPIO16
#  right_button_pin: GPIO5
#  scl_pin: GPIO22 
#  sda_pin: GPIO17

esphome:
  name: $devicename
  friendly_name: $friendly_name

esp32:
  board: $board

logger:

api:

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Awtrix Fallback Hotspot"
    password: !secret wifi_fallback_pass

time:
  - platform: homeassistant
    id: homeassistant_time

external_components:
  - source:
      type: git
      url: https://github.com/lubeda/EspHoMaTriXv2
      ref: main

font: 
  - file: fonts/MatrixClock-Chunky6.bdf
    id: special_font
    #size: 16
    glyphs:  |
      ! "#$%&'()*+,-./0123456789:APMTapmt
  - file: fonts/MatrixClock-Light8.bdf
    id: default_font
    #size: 16
    glyphs:  |
      ! "#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz°
#  - file: 'fonts/EHMTXv2.ttf'
#    size: 16
#    id: default_font
#    glyphs:  |
#      !?"%()+*=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnÖÄÜöäüopqrstuvwxyz@<>ß§€/

display:
  - platform: addressable_light
    id: ehmtx_display
    addressable_light_id: ehmtx_light
    width: 32
    height: 8
    pixel_mapper: |-
      if (x % 2 == 0) {
        return (x * 8) + y;
      }
      return (x * 8) + (7 - y);
    rotation: 0°
    update_interval: 16ms
    auto_clear_enabled: true
    lambda: |-
      id(rgb8x32)->tick();
      id(rgb8x32)->draw();

light:
  - platform: neopixelbus
    id: ehmtx_light
    type: GRB
    internal: true
    variant: WS2812
    pin: $matrix_pin
    num_leds: 256
#    color_correct: [30%, 30%, 30%]
#    gamma_correct: 2.0
    name: "Awtrix Light"
    restore_mode: ALWAYS_OFF
    on_turn_on:
      lambda: |-
         id(ehmtx_display)->set_enabled(false);
    on_turn_off:
       lambda: |-
         id(ehmtx_display)->set_enabled(true);

ehmtxv2:
  id: rgb8x32
  icons2html: true
  matrix_component: ehmtx_display
  time_component: homeassistant_time
  time_format: "%H:%M"
  date_format: "%d.%m.%Y"
  show_seconds: false
  default_font_id: default_font
  special_font_id: special_font
  icons: 
    - id: error
      lameid: 40530
#  on_next_screen:
#    - homeassistant.event:
#        event: esphome.new_screen
#        data_template:
#          iconname: !lambda "return icon.c_str();"
#          text: !lambda "return text.c_str();"
```

## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

```text
...
[01:14:32]Rebooting...
[01:14:32]ets Jul 29 2019 12:21:46
[01:14:32]
[01:14:32]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[01:14:32]configsip: 0, SPIWP:0xee
[01:14:32]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[01:14:32]mode:DIO, clock div:2
[01:14:32]load:0x3fff0018,len:4
[01:14:32]load:0x3fff001c,len:1044
[01:14:32]load:0x40078000,len:8896
[01:14:32]load:0x40080400,len:5828
[01:14:32]entry 0x400806ac
[01:14:32][I][logger:262]: Log initialized
[01:14:32][C][ota:469]: There have been 7 suspected unsuccessful boot attempts.
[01:14:32][D][esp32.preferences:114]: Saving 1 preferences to flash...
[01:14:32][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[01:14:32][D][EHMTXv2:007]: Constructor start
[01:14:32][D][EHMTXv2:032]: Constructor finish
[01:14:32][D][EHMTXv2:1063]: add_icon no.: 0 name: error frame_duration: 500 ms
[01:14:32][D][EHMTXv2:1002]: set_display
[01:14:32][D][EHMTXv2:1008]: set_clock
[01:14:32][I][EHMTXv2:983]: set_brightness 80 => 31.37 %
[01:14:32]Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
[01:14:32]
[01:14:32]Core  1 register dump:
[01:14:32]PC      : 0x400d8a6a  PS      : 0x00060c30  A0      : 0x800e8b16  A1      : 0x3ffb2700  
[01:14:32]A2      : 0xffffffff  A3      : 0x437f0000  A4      : 0x42c80000  A5      : 0x00000008  
[01:14:32]A6      : 0x00000002  A7      : 0x000001f4  A8      : 0x00000200  A9      : 0x3ffb26e0  
[01:14:32]A10     : 0x00000050  A11     : 0x3f400d48  A12     : 0x000003d7  A13     : 0x3f400f64  
[01:14:32]A14     : 0x00000050  A15     : 0x3ffb4efc  SAR     : 0x00000006  EXCCAUSE: 0x0000001d  
[01:14:32]EXCVADDR: 0x00000218  LBEG    : 0x4008ab2d  LEND    : 0x4008ab3d  LCOUNT  : 0xfffffff1  
[01:14:32]
[01:14:32]
[01:14:32]Backtrace:0x400d8a67:0x3ffb27000x400e8b13:0x3ffb2740 0x400ed46a:0x3ffb2820 
[01:14:32]
[01:14:32]
[01:14:32]
[01:14:32]
[01:14:32]ELF file SHA256: 0000000000000000
[01:14:32]
[01:14:32]Rebooting...
[01:14:32]ets Jul 29 2019 12:21:46
[01:14:32]
[01:14:32]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[01:14:32]configsip: 0, SPIWP:0xee
[01:14:32]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[01:14:32]mode:DIO, clock div:2
[01:14:32]load:0x3fff0018,len:4
[01:14:32]load:0x3fff001c,len:1044
[01:14:32]load:0x40078000,len:8896
[01:14:32]load:0x40080400,len:5828
[01:14:32]entry 0x400806ac
[01:14:33][I][logger:262]: Log initialized
[01:14:33][C][ota:469]: There have been 8 suspected unsuccessful boot attempts.
[01:14:33][D][esp32.preferences:114]: Saving 1 preferences to flash...
[01:14:33][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[01:14:33][D][EHMTXv2:007]: Constructor start
[01:14:33][D][EHMTXv2:032]: Constructor finish
[01:14:33][D][EHMTXv2:1063]: add_icon no.: 0 name: error frame_duration: 500 ms
[01:14:33][D][EHMTXv2:1002]: set_display
[01:14:33][D][EHMTXv2:1008]: set_clock
[01:14:33][I][EHMTXv2:983]: set_brightness 80 => 31.37 %
[01:14:33]Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
[01:14:33]
[01:14:33]Core  1 register dump:
[01:14:33]PC      : 0x400d8a6a  PS      : 0x00060c30  A0      : 0x800e8b16  A1      : 0x3ffb2700  
[01:14:33]A2      : 0xffffffff  A3      : 0x437f0000  A4      : 0x42c80000  A5      : 0x00000008  
[01:14:33]A6      : 0x00000002  A7      : 0x000001f4  A8      : 0x00000200  A9      : 0x3ffb26e0  
[01:14:33]A10     : 0x00000050  A11     : 0x3f400d48  A12     : 0x000003d7  A13     : 0x3f400f64  
[01:14:33]A14     : 0x00000050  A15     : 0x3ffb4efc  SAR     : 0x00000006  EXCCAUSE: 0x0000001d  
[01:14:33]EXCVADDR: 0x00000218  LBEG    : 0x4008ab2d  LEND    : 0x4008ab3d  LCOUNT  : 0xfffffff1  
[01:14:33]
[01:14:33]
[01:14:33]Backtrace:0x400d8a67:0x3ffb27000x400e8b13:0x3ffb2740 0x400ed46a:0x3ffb2820
...
```

## Services calls

```text
(optional) The YAML of your service calls
```

trip5, 2023-07-02:
You're sure your Awtrix uses an ESP32 D1 Mini?  My Awtrix uses an ESP8266 D1 Mini.

'''
esp8266:
  board: d1_mini
'''

BTW, to lubeda:  I have no problems compiling ESPHome for the ESP8266.  I do have to edit the YAML quite a bit but it works just fine.

vitalogy, 2023-07-02:
Yes, originally the Awtrix uses an EPS8266 (D1 Mini). The old blueforcer firmware is still running on this.
But now I wanted to switch to ESPHome and use the full potential of EspHoMaTriXv2, so I got an ESP32 (D1 Mini ESP32) to replace the ESP8266. This should fit, the pin assignment is identical in the two inner pin rows.
Unfortunately, the above error always occurs, regardless of whether the ESP32 is built into the Awtrix or is only powered via USB.
I'm a little confused...

vitalogy, 2023-07-02:
BTW, how can I be sure that the current/latest code from main is used?

lubeda, 2023-07-02:
Hi,
with this/your  configuration you always use the latest version:

```yaml
external_components:
  - source:
      type: git
      url: https://github.com/lubeda/EspHoMaTriXv2
      ref: main
```

For the "bug" I experienced this error also sometimes, but I don't think it is a problem of my module. I think it is a problem of the compiler or esphome. If you change your yaml and add another component, like `webserver:` it may work. 

Sometimes it worked when I pressed "clean build files" in esphome.

lubeda, 2023-07-02:
I'm about 1000km away from my display on holiday. I will take a look after my vacation.

vitalogy, 2023-07-02:
I added `web_server:` to the yml, and cleaned build-files (I had already done this several times before compiling), but the esp32 keeps crashing/rebooting.
Here is the data that the esphome-flasher delivers:
```
Using 'COM8' as serial port.
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.........
Detecting chip type... ESP32
Connecting...........

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-D0WD-V3 (revision 3)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: XX:XX:XX:XX:XX:XX
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
 - Flash Mode: dio
 - Flash Frequency: 40MHz
Erasing flash (this may take a while)...
....
```

trip5, 2023-07-03:
The only thing I can recommend is to try downloading a bin and flashing using other flashers.

In particular for the ESP32 in Windows, I like: https://github.com/esphome/esphome-flasher/releases

I also sometimes use the Python esptool as well.

Eightquake, 2023-07-11:
I also had an issue like this, where I got error message `Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.`

I was able to fix it by moving the light configuration to be above the display configuration in the yaml:
```
esphome:
  name: matrix
  friendly_name: Matrix

esp32:
  board: esp32dev

external_components:
  - source:
      type: git
      url: https://github.com/lubeda/EspHoMaTriXv2
      ref: main

light:
  - platform: neopixelbus
    id: ehmtx_light
    type: GRB
    num_leds: 256
    variant: WS2812
    pin: 13
    on_turn_on:
      lambda: |-
         id(ehmtx_display)->set_enabled(false);
    on_turn_off:
       lambda: |-
         id(ehmtx_display)->set_enabled(true);
#    color_correct: [30%, 30%, 30%]
#    gamma_correct: 2.0

display:
  - platform: addressable_light
    id: ehmtx_display
    addressable_light_id: ehmtx_light
    width: 32
    height: 8
    pixel_mapper: |-
        return ((int) (x/8 * 64)) + x % 8 + y * 8;
    rotation: 0
    update_interval: 16ms
    auto_clear_enabled: true
    lambda: |-
      id(rgb8x32)->tick();
      id(rgb8x32)->draw();

ehmtxv2:
  id: rgb8x32
  time_component: ntp
  matrix_component: ehmtx_display
  default_font_id: ehmtx_font
  special_font_id: ehmtx_font
  show_seconds: true
  brightness: 20 # percent
  time_format: "%H:%M"
  date_format: "%d.%m."
  scroll_count: 2 # scroll long text at least two times
  scroll_interval: 80 # milliseconds
  icons:
    - id: boot 
      lameid: 49151
    - id: temp 
      lameid: 2369
```

vitalogy, 2023-07-16:
> I also had an issue like this, where I got error message `Guru Meditation Error: Core 1 panic'ed (StoreProhibited). Exception was unhandled.`
> 
> I was able to fix it by moving the light configuration to be above the display configuration in the yaml:
> 
> ```
> esphome:
> ....
> light:
> ...
> display:
> ...
> ehmtxv2:
> ...
> ```

YES ... that did the trick 👍

vitalogy, 2023-07-16:
Should that be mentioned somewhere?

lubeda, 2023-07-20:
Thanks for your investigation. I noted this tip in a wiki article.

closed 2023-07-20 by lubeda

trip5, 2024-08-18:
Just wanted to add a note here for people searching in the future.  I recently updated my clocks (for the first time in a year - whoops, been busy).  And I got a similar kernel panic error (but with a divide by zero error)... and I ended up following the ulanzi-easy.yaml file. I copied and pasted my chunks of code back in the order they appear in the easy file.  Compiled, uploaded, and ran fine after that.

I still can't figure out the exact ordering error - and I've even tested out a few re-orders of components... but sometimes starting from scratch clears up any problems.