lubeda/EspHoMaTriXv2 #56: [BUG] YAML-Files not loaded in Blueprint

clfberlin, 2023-07-12:
The EHMTX_easy*.yaml are not loaded as Blueprints. All generate the same error.

`Failed to load blueprint: mapping values are not allowed here in "/config/blueprints/automation/ehmtxv2/EHMTX_easy_color_gauge.yaml", line 198, column 74`

In Home Assistant, the _File Editor_ states:
`"end of the stream or a document separator is expected (198:74)"`

YAML files uploaded in /config/blueprints/automation/ehmtxv2/

Home Assistant 2023.7.1
Supervisor 2023.07.1
Operating System 10.3

## To Reproduce

Upload YAML-files in folder and restart Home Assistant

Thanks!

lubeda, 2023-07-15:
Is there anyone that has the same issue?

clfberlin, 2023-07-15:
Sorry, I have no idea what went wrong in the transfer process. I just recreated each YAML and copied the content from here to the blueprint and it now loads and accepts everything. The problem must have occurred with me somewhere between down- and uploading. Sorry for the hassle.

closed 2023-07-15 by lubeda