lubeda/EspHoMaTriXv2 #57: Allow removing Ehmtx logo and version screens

dbuezas, 2023-07-12:
# Feature Request

Allow for removing those screens via yaml. Together they take ~1kb of flash and they are a bit intrusive.

e.g:
```yaml
disable_logo_at_boot: true
disable_version_screen_at_boot: true
```

# Workaround

For now, this seems to be working:

```yaml
esphome:
  [...]
  on_boot:
    priority: -100
    then:
      - ds1307.read_time:
      - delay: 1ms
      - lambda: |-
          // delete all pre-defined screens
          id(rgb8x32)->del_screen("*", MODE_CLOCK);
          id(rgb8x32)->del_screen("*", MODE_DATE);
          id(rgb8x32)->del_screen("*", MODE_FULL_SCREEN);
          id(rgb8x32)->del_screen("*", MODE_ICON_SCREEN);
          id(rgb8x32)->del_screen("*", MODE_TEXT_SCREEN);
          id(rgb8x32)->del_screen("*", MODE_RAINBOW_ICON);
          id(rgb8x32)->del_screen("*", MODE_RAINBOW_TEXT);
          id(rgb8x32)->del_screen("*", MODE_RAINBOW_CLOCK);
          id(rgb8x32)->del_screen("*", MODE_RAINBOW_DATE);
          id(rgb8x32)->del_screen("*", MODE_BITMAP_SCREEN);
          id(rgb8x32)->del_screen("*", MODE_BITMAP_SMALL);
          
          // recreate clock screens with shorter screen_time
          int lifetime = 24*60*365*10;
          int screen_time = 1;
          bool default_font = true;
          id(rgb8x32)->clock_screen(lifetime, screen_time, default_font, 240, 240, 240); // also create 
          id(rgb8x32)->date_screen(lifetime, screen_time, default_font, 240, 240, 240);
```

lubeda, 2023-07-13:
See the develop branch. You can now define your own non-intrusive logo if you like, or leave it.
With your YAML sample, you would see the logo only once every ten years but changing date to clock each second isn't intrusive!?!?!

That's how taste differs.

closed 2023-07-13 by lubeda

dbuezas, 2023-07-13:
Oh, cool, I didn't check the develop branch.
With intrusive I meant that I didn't have the freedom to remove it. Glad to hear this is already covered in develop :)

I have a bunch of data showing in the display (co2, temperature inside/outside, trash schedules, etc) so i want them to cycle quickly, hence 1 second per screen