lubeda/EspHoMaTriXv2 #58: Hold screen doesn't respect parameter

dbuezas, 2023-07-12:
# Bug report

## Describe the bug
hold_screen adds the default hold_time to the next action and then changes that default. Instead, I'd expect it to set the next_action_time to `this->clock->now().timestamp + time`.

### Current implementation:
```cpp
void EHMTX::hold_screen(int time)
  {
    this->next_action_time += this->hold_time;
    this->hold_time = time;
  }
```

### Suggested change:
```cpp
void EHMTX::hold_screen(int time)
  {
    this->next_action_time = this->clock->now().timestamp + time;
  }
```

lubeda, 2023-07-13:
Why don't you provide a pull request?

dbuezas, 2023-07-13:
Ok, I'll do that. I'm new to this codebase so I wasn't sure I had understood it correctly

lubeda, 2023-07-20:
I will take a further look tomorroy

lubeda, 2023-07-21:
fixed in develop