lubeda/EspHoMaTriXv2 #59: Calling skip_screen repeatedly brings back deleted screens

dbuezas, 2023-07-12:
# Bug report
Calling skip_screen repeatedly brings back deleted screens

## Describe the bug

By adding `id(rgb8x32)->skip_screen();` to the on_press of the right button, I can call it in rapid succession.
It looks like doing it more than once before there is an automatic screen change makes it pick up broken/deleted screens.

dbuezas, 2023-07-12:
Looking at the logs, I see "tick: nothing to do. Restarting clock display", and I am deleting those so this seems to be the problem. I would expect it to just jump to the front of the circular queue.
Interestingly, the unexpectedly added clock screens have the wrong y offset (I'm using 10 since I have a full height font)

`allow_empty_screen: true` avoids the issue by just showing a blank screen

lubeda, 2023-07-13:
Why don't you provide a pull request based on the develop-branch?

dbuezas, 2023-07-13:
I'll give it a try when I have some time.
Thanks for this cool project!