lubeda/EspHoMaTriXv2 #60: [BUG] no longer building

nickrout, 2023-07-14:
# Bug report

## Describe the bug

esphome no longer building easy

## Additional information

- used Hardware:
  - ulanzi

## To Reproduce

Steps to reproduce the behavior: compile on esphome 2023.7

## Expected behavior

A clear and concise description of what you expected to happen.

## Configuration

```text
(optional) The YAML you used in epshome without any password
```

## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

```text
INFO ESPHome 2023.7.0-dev
INFO Reading configuration /config/esphome/ulanzi-easy.yaml...
WARNING 
WARNING Please check the documentation and wiki https://github.com/lubeda/EspHoMaTriXv2
WARNING 
WARNING GPIO15 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'Pacific/Auckland'
INFO Detected timezone 'Pacific/Auckland'
INFO Generating C++ source...
INFO Preparing icons, this may take some seconds.
INFO EsphoMaTrix: wrote html-file with icon preview: /config/esphome/ulanzi-easy.html
INFO List of icons for e.g. blueprint:

["error","home_assistant","temperature","lightbulb","music","phone","car","sleep8x32",]

INFO Compiling app...
Processing ulanzi (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.3.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/AsyncTCP-esphome @ 1.2.2
INFO Installing esphome/AsyncTCP-esphome @ 1.2.2
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@1.2.2 has been installed!
INFO AsyncTCP-esphome@1.2.2 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@2.1.0 has been installed!
INFO ESPAsyncWebServer-esphome@2.1.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing bblanchon/ArduinoJson @ 6.18.5
INFO Installing bblanchon/ArduinoJson @ 6.18.5
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
[... 243 lines cut ...]
     if (this->get_animation_frame_count() > 1)
               ^~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/ehmtxv2/EHMTX_icons.cpp:21:37: error: 'class esphome::EHMTX_Icon' has no member named 'get_current_frame'
         if (this->reverse && (this->get_current_frame() == this->get_animation_frame_count() - 2))
                                     ^~~~~~~~~~~~~~~~~
src/esphome/components/ehmtxv2/EHMTX_icons.cpp:21:66: error: 'class esphome::EHMTX_Icon' has no member named 'get_animation_frame_count'
         if (this->reverse && (this->get_current_frame() == this->get_animation_frame_count() - 2))
                                                                  ^~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/ehmtxv2/EHMTX_icons.cpp:25:9: error: 'Animation' has not been declared
         Animation::next_frame();
         ^~~~~~~~~
src/esphome/components/ehmtxv2/EHMTX_icons.cpp:29:19: error: 'class esphome::EHMTX_Icon' has no member named 'get_current_frame'
         if (this->get_current_frame() == 1) // this->get_animation_frame_count())
                   ^~~~~~~~~~~~~~~~~
src/esphome/components/ehmtxv2/EHMTX_icons.cpp:33:9: error: 'Animation' has not been declared
         Animation::prev_frame();
         ^~~~~~~~~
Compiling /data/ulanzi/.pioenvs/ulanzi/src/esphome/components/font/font.cpp.o
*** [/data/ulanzi/.pioenvs/ulanzi/src/esphome/components/ehmtxv2/EHMTX_icons.cpp.o] Error 1
*** [/data/ulanzi/.pioenvs/ulanzi/src/esphome/components/ehmtxv2/EHMTX_queue.cpp.o] Error 1
*** [/data/ulanzi/.pioenvs/ulanzi/src/esphome/components/ehmtxv2/EHMTX.cpp.o] Error 1
========================= [FAILED] Took 47.84 seconds =========================
```

## Services calls

nickrout, 2023-07-14:
Suspecting what looks like a major display re-factor https://github.com/esphome/esphome/pull/4996

lubeda, 2023-07-15:
I will wait until the release of esphome before a take a closer look. The PR https://github.com/esphome/esphome/pull/4996 would be helpfull for ehmtx

closed 2023-07-15 by lubeda

trip5, 2023-07-15:
I think there's something wrong with your YAML.  Or are you using a dev/edge branch of ESPHome?

I had no problem on ESPHome 2023.6.5 with either EHTMXV2-2023.6.5 main branch or the 7.0 dev branch (as of just now) with an Ulanzi TC-001.

BTW, it's not bad practice to download the zip and use local component files. I tend to keep a few copies of stable EHTMXV2 files in my ESPHome just in case.

nickrout, 2023-07-15:
I do not believe I have changed the yaml from what I downloaded here. But maybe the code on github has updated.

Yes I am using the dev addon, hence 

```
INFO ESPHome 2023.7.0-dev
```
It is a pity the dev addon for esphome does not reveal a hash for the exact development state it is using.

nickrout, 2023-07-15:
> I will wait until the release of esphome before a take a closer look. The PR [esphome/esphome#4996](https://github.com/esphome/esphome/pull/4996) would be helpfull for ehmtx

Fair enough, thanks.