lubeda/EspHoMaTriXv2 #62: Issues with ESPHome 2023.07 update

clfberlin, 2023-07-19:
# Bug report

## Describe the bug

Cannot update after ESPHome 2023.07. It seems that the new version brought changes that prevents updates (to Ulanzi). This seems to evolve around the font-handling.

## Logs

```text
In file included from src/esphome.h:25,
                 from src/esphome/components/ehmtxv2/EHMTX_icons.cpp:1:
src/esphome/components/ehmtxv2/EHMTX.h:95:14: error: 'Font' in namespace 'esphome::display' does not name a type
     display::Font *default_font;
              ^~~~
src/esphome/components/ehmtxv2/EHMTX.h:96:14: error: 'Font' in namespace 'esphome::display' does not name a type
     display::Font *special_font;
              ^~~~
src/esphome/components/ehmtxv2/EHMTX.h:150:36: error: 'esphome::display::Font' has not been declared
     void set_default_font(display::Font *font);
                                    ^~~~
src/esphome/components/ehmtxv2/EHMTX.h:151:36: error: 'esphome::display::Font' has not been declared
     void set_special_font(display::Font *font);
                                    ^~~~
src/esphome/components/ehmtxv2/EHMTX.h:267:3: error: expected class-name before '{' token
   {
   ^
src/esphome/components/ehmtxv2/EHMTX.h:272:107: error: 'esphome::display::ImageType' has not been declared
     EHMTX_Icon(const uint8_t *data_start, int width, int height, uint32_t animation_frame_count, display::ImageType type, std::string icon_name, bool revers, uint16_t frame_duration);
                                                                                                           ^~~~~~~~~
In file included from src/esphome.h:25,
                 from src/esphome/components/ehmtxv2/EHMTX.cpp:1:
src/esphome/components/ehmtxv2/EHMTX.h:95:14: error: 'Font' in namespace 'esphome::display' does not name a type
     display::Font *default_font;
              ^~~~
src/esphome/components/ehmtxv2/EHMTX.h:96:14: error: 'Font' in namespace 'esphome::display' does not name a type
     display::Font *special_font;
              ^~~~
src/esphome/components/ehmtxv2/EHMTX.h:150:36: error: 'esphome::display::Font' has not been declared
     void set_default_font(display::Font *font);
                                    ^~~~
src/esphome/components/ehmtxv2/EHMTX.h:151:36: error: 'esphome::display::Font' has not been declared
     void set_special_font(display::Font *font);
                                    ^~~~
src/esphome/components/ehmtxv2/EHMTX.h:267:3: error: expected class-name before '{' token
   {
   ^
src/esphome/components/ehmtxv2/EHMTX.h:272:107: error: 'esphome::display::ImageType' has not been declared
     EHMTX_Icon(const uint8_t *data_start, int width, int height, uint32_t animation_frame_count, display::ImageType type, std::string icon_name, bool revers, uint16_t frame_duration);
                                                                                                           ^~~~~~~~~
src/esphome/components/ehmtxv2/EHMTX_icons.cpp:6:117: error: 'esphome::display::ImageType' has not been declared
   EHMTX_Icon::EHMTX_Icon(const uint8_t *data_start, int width, int height, uint32_t animation_frame_count, display::ImageType type, std::string icon_name, bool revers, uint16_t frame_duration)
                                                                                                                     ^~~~~~~~~
src/esphome/components/ehmtxv2/EHMTX_icons.cpp: In constructor 'esphome::EHMTX_Icon::EHMTX_Icon(const uint8_t*, int, int, uint32_t, int, std::__cxx11::string, bool, uint16_t)':
src/esphome/components/ehmtxv2/EHMTX_icons.cpp:7:9: error: class 'esphome::EHMTX_Icon' does not have any field named 'Animation'
       : Animation(data_start, width, height, animation_frame_count, type)
         ^~~~~~~~~
src/esphome/components/ehmtxv2/EHMTX_icons.cpp: In member function 'void esphome::EHMTX_Icon::next_frame()':
src/esphome/components/ehmtxv2/EHMTX_icons.cpp:17:15: error: 'class esphome::EHMTX_Icon' has no member named 'get_animation_frame_count'
     if (this->get_animation_frame_count() > 1)
               ^~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/ehmtxv2/EHMTX_icons.cpp:21:37: error: 'class esphome::EHMTX_Icon' has no member named 'get_current_frame'
         if (this->reverse && (this->get_current_frame() == this->get_animation_frame_count() - 2))
                                     ^~~~~~~~~~~~~~~~~
src/esphome/components/ehmtxv2/EHMTX_icons.cpp:21:66: error: 'class esphome::EHMTX_Icon' has no member named 'get_animation_frame_count'
         if (this->reverse && (this->get_current_frame() == this->get_animation_frame_count() - 2))
                                                                  ^~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/ehmtxv2/EHMTX_icons.cpp:25:9: error: 'Animation' has not been declared
         Animation::next_frame();
         ^~~~~~~~~
src/esphome/components/ehmtxv2/EHMTX_icons.cpp:29:19: error: 'class esphome::EHMTX_Icon' has no member named 'get_current_frame'
         if (this->get_current_frame() == 1) // this->get_animation_frame_count())
                   ^~~~~~~~~~~~~~~~~
src/esphome/components/ehmtxv2/EHMTX_icons.cpp:33:9: error: 'Animation' has not been declared
         Animation::prev_frame();
         ^~~~~~~~~
src/esphome/components/ehmtxv2/EHMTX.cpp: In member function 'void esphome::EHMTX::tick()':
src/esphome/components/ehmtxv2/EHMTX.cpp:581:67: error: 'class esphome::EHMTX_Icon' has no member named 'set_frame'; did you mean 'next_frame'?
             this->icons[this->queue[this->screen_pointer]->icon]->set_frame(0);
                                                                   ^~~~~~~~~
                                                                   next_frame
src/esphome/components/ehmtxv2/EHMTX.cpp: At global scope:
src/esphome/components/ehmtxv2/EHMTX.cpp:677:41: error: variable or field 'set_default_font' declared void
   void EHMTX::set_default_font(display::Font *font)
                                         ^~~~
src/esphome/components/ehmtxv2/EHMTX.cpp:677:41: error: 'Font' is not a member of 'esphome::display'
src/esphome/components/ehmtxv2/EHMTX.cpp:677:41: note: suggested alternative:
In file included from src/esphome.h:28,
                 from src/esphome/components/ehmtxv2/EHMTX.cpp:1:
src/esphome/components/font/font.h:41:7: note:   'esphome::font::Font'
 class Font : public display::BaseFont {
       ^~~~
src/esphome/components/ehmtxv2/EHMTX.cpp:677:51: error: expected primary-expression before ')' token
   void EHMTX::set_default_font(display::Font *font)
                                                   ^
src/esphome/components/ehmtxv2/EHMTX.cpp:682:41: error: variable or field 'set_special_font' declared void
   void EHMTX::set_special_font(display::Font *font)
                                         ^~~~
src/esphome/components/ehmtxv2/EHMTX.cpp:682:41: error: 'Font' is not a member of 'esphome::display'
src/esphome/components/ehmtxv2/EHMTX.cpp:682:41: note: suggested alternative:
In file included from src/esphome.h:28,
                 from src/esphome/components/ehmtxv2/EHMTX.cpp:1:
src/esphome/components/font/font.h:41:7: note:   'esphome::font::Font'
 class Font : public display::BaseFont {
       ^~~~
src/esphome/components/ehmtxv2/EHMTX.cpp:682:51: error: expected primary-expression before ')' token
   void EHMTX::set_special_font(display::Font *font)
                                                   ^
*** [/data/ulanzi/.pioenvs/ulanzi/src/esphome/components/ehmtxv2/EHMTX_icons.cpp.o] Error 1
*** [/data/ulanzi/.pioenvs/ulanzi/src/esphome/components/ehmtxv2/EHMTX.cpp.o] Error 1
```

Thanks!

paolo-rendano, 2023-07-19:
same issue here. In the [2023.7.0 esphome changelog](https://esphome.io/changelog/2023.7.0.html) there are breaking changes regarding Fonts

lubeda, 2023-07-19:
Yes, there are even more breaking changes in esphome 2023.7.0 that are not mentioned. I have to investigate

lubeda, 2023-07-19:
Looking for helpl here:

https://discord.com/channels/429907082951524364/524177279270780928/1131268742815436920

idodov, 2023-07-19:
Also having compilation problem

```dart
Compiling /data/ulanzi/.pioenvs/ulanzi/src/esphome/components/homeassistant/time/homeassistant_time.cpp.o
In file included from src/esphome.h:25,
                 from src/esphome/components/ehmtxv2/EHMTX_icons.cpp:1:
src/esphome/components/ehmtxv2/EHMTX.h:5:10: fatal error: esphome/components/animation/animation.h: No such file or directory
 #include "esphome/components/animation/animation.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [/data/ulanzi/.pioenvs/ulanzi/src/esphome/components/ehmtxv2/EHMTX_icons.cpp.o] Error 1
In file included from src/esphome.h:25,
                 from src/esphome/components/ehmtxv2/EHMTX.cpp:1:
src/esphome/components/ehmtxv2/EHMTX.h:5:10: fatal error: esphome/components/animation/animation.h: No such file or directory
 #include "esphome/components/animation/animation.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [/data/ulanzi/.pioenvs/ulanzi/src/esphome/components/ehmtxv2/EHMTX.cpp.o] Error 1
In file included from src/esphome.h:25,
                 from src/esphome/components/ehmtxv2/EHMTX_queue.cpp:1:
src/esphome/components/ehmtxv2/EHMTX.h:5:10: fatal error: esphome/components/animation/animation.h: No such file or directory
 #include "esphome/components/animation/animation.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [/data/ulanzi/.pioenvs/ulanzi/src/esphome/components/ehmtxv2/EHMTX_queue.cpp.o] Error 1
========================== [FAILED] 
```

jfurtner, 2023-07-20:
> Also having compilation problem
> 

@idodov You can work around this by bringing in an animation (which itself requires image apparently):

```
animation:
  - id: animation_black
    file: "black.gif"

image:
  - id: image_black
    file: "black.gif"
```
black.gif is just an 1x1 black gif located in the esphome config folder (along side the esphome device yaml file). The graphic only consumes 1 byte of data + the object so I'd guess it's <250 bytes of flash that's consumed for this part (in addition to whatever size the animation & image code is).

This is a work around - it should be a dynamic dependency of the component instead but I don't see how to do that yet.

idodov, 2023-07-20:
> @idodov You can work around this by bringing in an animation (which itself requires image apparently):
>
> ```
> animation:
>   - id: animation_black
>     file: "black.gif"
>
> image:
>   - id: image_black [...]
@jfurtner , Where in the yaml file I need to add those lines?

candyman2903, 2023-07-20:
I had no black.gif in esphome folder. That's why I quickly made a 1x1 gif in black via the site: https://onlinegiftools.com/create-1x1-pixel-gif

I put the code part under the font block.

lubeda, 2023-07-20:
@jfurtner Thanks, that is what I just found out.  Only components that are referenced in the yaml are included in the compile part. Since i use animation only internally, it isn't included and the compiler error is "correct".

I am searching for a way to tweak the code generation from the yaml.

idodov, 2023-07-20:
@jfurtner @candyman2903 Thank you very much! Problem sovled.

lubeda, 2023-07-20:
> @jfurtner @candyman2903 Thank you very much! Problem sovled.

This is no solution, it's only a workaround!!!!

Issues in esphome around "my" problem:
https://github.com/esphome/issues/issues/4726 and https://github.com/esphome/issues/issues/4719

lubeda, 2023-07-21:
There are some other projects that suffer from this breaking change, e.g. https://github.com/esphome/issues/issues/4720

For now there is no solution but a workaround.

TheGabeMan, 2023-08-03:
Fixed it for me!

closed 2023-08-16 by lubeda

Empty2k12, 2023-08-24:
```
wget https://upload.wikimedia.org/wikipedia/commons/e/ed/Pix.gif && mv Pix.gif black.gif
```
in your config folder

RikMuilwijk, 2023-10-31:
@clfberlin ever found a solution to the font issue? Having the exact same issue, skipping ESPhome updates since 2023.6.5

clfberlin, 2023-10-31:
Hi @RikMuilwijk - The workround mentioned above worked for me.
Generate a 1x1px black pixel, place it in /config/esphome and add to the YAML file

```
animation:
  - id: animation_black
    file: "black.gif"
display:
  ...
ehmtxv2:
  ...
font:
  ...
image:
  - id: image_black
    file: "black.gif"
```

RikMuilwijk, 2023-11-02:
Thanks, added the lines but sadly no luck :( My log and yaml;

```ruby
INFO ESPHome 2023.10.5
INFO Reading configuration /config/esphome/ledclock.yaml...
INFO Updating https://github.com/RikMuilwijk/EsphoMaTrix@None
WARNING 
WARNING If you are upgrading EsphoMaTrix from a version before 2023.4.0,
WARNING you should read the section https://github.com/lubeda/EsphoMaTrix/#how-to-update for tipps.
WARNING 
WARNING GPIO15 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'Europe/Amsterdam'
INFO Detected timezone 'Europe/Amsterdam'
INFO Generating C++ source...
INFO Compiling app...
Processing ulanzi (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.0.1
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- ESPmDNS @ 2.0.0
|-- Wire @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- NeoPixelBus @ 2.7.3
Compiling .pioenvs/ulanzi/src/esphome/components/ehmtx/EHMTX.cpp.o
Compiling .pioenvs/ulanzi/src/esphome/components/ehmtx/EHMTX_icons.cpp.o
Compiling .pioenvs/ulanzi/src/esphome/components/ehmtx/EHMTX_screen.cpp.o
Compiling .pioenvs/ulanzi/src/esphome/components/ehmtx/EHMTX_store.cpp.o
In file included from src/esphome.h:26,
                 from src/esphome/components/ehmtx/EHMTX.cpp:1:
src/esphome/components/ehmtx/EHMTX.h:62:14: error: 'Font' in namespace 'esphome::display' does not name a type
     display::Font *font;
              ^~~~
src/esphome/components/ehmtx/EHMTX.h:102:28: error: 'esphome::display::Font' has not been declared
     void set_font(display::Font *font);
                            ^~~~
src/esphome/components/ehmtx/EHMTX.h:552:3: error: expected class-name before '{' token
   {
   ^
src/esphome/components/ehmtx/EHMTX.h:557:107: error: 'esphome::display::ImageType' has not been declared
     EHMTX_Icon(const uint8_t *data_start, int width, int height, uint32_t animation_frame_count, display::ImageType type, std::string icon_name, bool revers, uint16_t frame_duration);
                                                                                                           ^~~~~~~~~
In file included from src/esphome.h:26,
                 from src/esphome/components/ehmtx/EHMTX_icons.cpp:1:
src/esphome/components/ehmtx/EHMTX.h:62:14: error: 'Font' in namespace 'esphome::display' does not name a type
     display::Font *font;
              ^~~~
src/esphome/components/ehmtx/EHMTX.h:102:28: error: 'esphome::display::Font' has not been declared
     void set_font(display::Font *font);
                            ^~~~
src/esphome/components/ehmtx/EHMTX.h:552:3: error: expected class-name before '{' token
   {
   ^
src/esphome/components/ehmtx/EHMTX.h:557:107: error: 'esphome::display::ImageType' has not been declared
     EHMTX_Icon(const uint8_t *data_start, int width, int height, uint32_t animation_frame_count, display::ImageType type, std::string icon_name, bool revers, uint16_t frame_duration);
                                                                                                           ^~~~~~~~~
In file included from src/esphome.h:26,
                 from src/esphome/components/ehmtx/EHMTX_store.cpp:1:
src/esphome/components/ehmtx/EHMTX.h:62:14: error: 'Font' in namespace 'esphome::display' does not name a type
     display::Font *font;
              ^~~~
src/esphome/components/ehmtx/EHMTX.h:102:28: error: 'esphome::display::Font' has not been declared
     void set_font(display::Font *font);
[... 373 lines cut ...]
    on_time_sync:
      then:
        ds1307.write_time:
  - platform: ds1307
    update_interval: never
    id: ehmtx_time
  
display:
  - platform: addressable_light
    id: ehmtx_display
    addressable_light_id: ehmtx_light
    width: 32
    height: 8
    pixel_mapper: |-
      if (y % 2 == 0) {
        return (y * 32) + x;
      }
      return (y * 32) + (31 - x);
    rotation: 0°
    update_interval: 16ms
    auto_clear_enabled: true
    lambda: |-
      id(rgb8x32)->tick();
      id(rgb8x32)->draw();
```

lubeda, 2023-11-02:
You are using the "old" repo https://github.com/lubeda/EsphoMaTrix which isn't maintained anymore. The newer https://github.com/lubeda/EsphoMaTrixv2 repo especially the branch 2023.9.1 is the way to go. **But** there are breaking changes for your automations in home assistant.