lubeda/EspHoMaTriXv2 #63: My two custom fonts

dbuezas, 2023-07-20:
I wanted to use the full height of the display and maximize the number of chars in the screen so I made these two fonts
[image]

[CalciumSuperThinTall.ttf.zip](https://github.com/lubeda/EspHoMaTriXv2/files/12104023/CalciumSuperThinTall.ttf.zip)
[CalciumThinTall.ttf.zip](https://github.com/lubeda/EspHoMaTriXv2/files/12104024/CalciumThinTall.ttf.zip)

```yaml
ehmtxv2:
   ...
  default_font_yoffset: 8
  default_font_id: default_font
  special_font_yoffset: 8
  special_font_id: special_font
font:
  - file: rgb-clock/CalciumThinTall.ttf
    size: 16
    id: default_font
    glyphs: |
      !?"%‰()+*=,-_.:°µ²³0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnÖÄÜöäüopqrstuvwxyz€$@<>/
  - file: rgb-clock/CalciumSuperThinTall.ttf
    size: 16
    id: special_font
    glyphs: |
      !?"%‰()+*=,-_.:°µ²³0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnÖÄÜöäüopqrstuvwxyz€$@<>/
```

lubeda, 2023-07-20:
Die Schriften sind klasse. Darf ich die mit in dieses Repo einbinden? (sobald sich das Modul wieder compilieren lässt)

dbuezas, 2023-07-20:
Sehr gerne! :)

dbuezas, 2023-07-20:
Can we call them differently? I originally started with the calcium font as a base, but then completely changed all characters.

I'd like `mateine.ttf` and `mateineThin.ttf`

lubeda, 2023-07-21:
OK, sind umbenannt und in develop gemerged

closed 2023-07-21 by lubeda

jcristovao, 2023-12-01:
Hey @dbuezas , this font looks really good!
Any chance that you could add more accented characters, like ã, Ã, à, À?
Thank you!

trip5, 2023-12-01:
I don't want to step on dbuezas' toes (it is a very nice font!) but if you need accents, mine have them already: https://github.com/trip5/Matrix-Fonts

dbuezas, 2023-12-01:
Those look very nice @trip5 ! 
@jcristovao you can edit them online, it's fairly straightforward:)