lubeda/EspHoMaTriXv2 #67: [BUG] Cann´t compile

softservicegit, 2023-07-24:
# Bug report

## Describe the bug

Add a description of the bug. Detail the expected behavior in contrast with the behavior you're observing.

## Additional information

- used Hardware:
  - ESP32 or ESP8266
  - EspHoMaTriXv2 version: [e.g. 2023.7.0]
  - Ulanzi TC001

## To Reproduce

Just try to compile the ulanzi-easy.yaml

## Expected behavior

no error message :)
## Configuration

```plain unchanged ulanzi-easy.yaml```

## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

```NFO ESPHome 2023.7.0
INFO Reading configuration /config/esphome/ulanzi-easy.yaml...
WARNING 
WARNING Please check the documentation and wiki https://github.com/lubeda/EspHoMaTriXv2
WARNING This will only work with esphome >= 2023.7.0
WARNING 
WARNING GPIO15 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'Europe/Berlin'
INFO Detected timezone 'Europe/Berlin'
INFO Generating C++ source...
INFO Preparing icons, this may take some seconds.
INFO EsphoMaTrix: wrote html-file with icon preview: /config/esphome/ulanzi-easy.html
INFO List of icons for e.g. blueprint:

["error","home_assistant","temperature","lightbulb","music","phone","car","sleep8x32",]

INFO Compiling app...
Processing ulanzi (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.3.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|-- ESPmDNS @ 2.0.0
|-- Wire @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- NeoPixelBus @ 2.7.3
Compiling /data/ulanzi/.pioenvs/ulanzi/src/esphome/components/ehmtxv2/EHMTX.cpp.o
Compiling /data/ulanzi/.pioenvs/ulanzi/src/esphome/components/ehmtxv2/EHMTX_icons.cpp.o
In file included from src/esphome.h:25,
                 from src/esphome/components/ehmtxv2/EHMTX.cpp:1:
src/esphome/components/ehmtxv2/EHMTX.h:6:10: fatal error: esphome/components/animation/animation.h: No such file or directory
 #include "esphome/components/animation/animation.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [/data/ulanzi/.pioenvs/ulanzi/src/esphome/components/ehmtxv2/EHMTX.cpp.o] Error 1
In file included from src/esphome.h:25,
                 from src/esphome/components/ehmtxv2/EHMTX_icons.cpp:1:
src/esphome/components/ehmtxv2/EHMTX.h:6:10: fatal error: esphome/components/animation/animation.h: No such file or directory
 #include "esphome/components/animation/animation.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [/data/ulanzi/.pioenvs/ulanzi/src/esphome/components/ehmtxv2/EHMTX_icons.cpp.o] Error 1
========================== [FAILED] Took 1.49 seconds ==========================```

## Services calls

```text
(optional) The YAML of your service calls
```

lubeda, 2023-07-24:
Hi, in your issue, you forgot to mention your update to esphome to 2023.7.0.

There is a workaround, see here: https://github.com/lubeda/EsphoMaTrixv2

softservicegit, 2023-07-24:
Yes, i was on 2023.7.0 before the clock arrived today. 
and, yes, I have not checked if your mentioned additions where already included in the template yaml because the last update was only three days ago. My fault. 

Yes. it compiles with the additional lines. Thanks

closed 2023-07-24 by softservicegit