lubeda/EspHoMaTriXv2 #71: [BUG] Bootloop

TheGroundZero, 2023-08-16:
# Bug report

## Describe the bug

I flashed a Ulanzi Pixel clock that used to run the v1 code with the latest YAML version of this v2 and now it enters a bootloop.

## Additional information

- used Hardware:
  - ESP32 (Ulanzi TC001 Pixel Clock)
  - EspHoMaTriXv2 version: 2023.7.0
  - ESPHome version: 2023.7.1

## To Reproduce

Steps to reproduce the behavior:

Copy YAML example and files according to documentation and flash device.

## Expected behavior

Clockface to display time and date like it used to in older versions

## Configuration

Based on https://github.com/lubeda/EspHoMaTriXv2/blob/main/copy2esphome/ulanzi-easy.yaml (with fixed font name)

```yaml
substitutions:
  devicename: ulanzi-tc001-d67068
  comment: "Ulanzi TC001 d6:70:68"
  board: esp32dev
  # Pin definition from https://github.com/aptonline/PixelIt_Ulanzi
  battery_pin: GPIO34
  buzzer_pin: GPIO15
  ldr_pin: GPIO35
  matrix_pin: GPIO32
  left_button_pin: GPIO26
  mid_button_pin: GPIO27
  right_button_pin: GPIO14
  scl_pin: GPIO22
  sda_pin: GPIO21

esp32:
  board: esp32dev

external_components:
  - source:
      type: git
      url: https://github.com/lubeda/EspHoMaTriXv2
      ref: main
    refresh: 60s 
    components: [ ehmtxv2 ]   

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: !secret not_domain
  use_address: 192.168.70.67

esphome:
  comment: $comment
  name: $devicename 
  project:
    name: "Ulanzi.EHMTXv2"
    version: "2.0.0"
  on_boot:
    then:
      - ds1307.read_time:

[... 430 lines cut ...]
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400da44c  PS      : 0x00060a30  A0      : 0x800efa13  A1      : 0x3ffb26d0  
A2      : 0xffffffff  A3      : 0x437f0000  A4      : 0x00000020  A5      : 0x00000008  
A6      : 0x00000010  A7      : 0x00000082  A8      : 0x00000200  A9      : 0x3ffb26b0  
A10     : 0x00000050  A11     : 0x437f0000  A12     : 0x3ffb44d9  A13     : 0x3ffb44d0  
A14     : 0x00000010  A15     : 0x00000003  SAR     : 0x00000006  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000218  LBEG    : 0x40089e98  LEND    : 0x40089eae  LCOUNT  : 0xffffffff  

Backtrace:0x400da449:0x3ffb26d00x400efa10:0x3ffb26f0 0x400fdfce:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...
[E][ota:476]: Boot loop detected. Proceeding to safe mode.
[W][ota:103]: Last Boot was an unhandled reset, will proceed to safe mode in 0 restarts
[W][ota:103]: Last Boot was an unhandled reset, will proceed to safe mode in 0 restarts
```

## Services calls

```text
(optional) The YAML of your service calls
```

TheGroundZero, 2023-08-16:
First tried OTA (device was already flashed with v1), but that already got it stuck (constant beeping, no display).
Then I tried usin the web flasher over USB.

Basic ESPHome Config (from ESPHome web flasher - Prepare for first use) did work

lubeda, 2023-08-16:
Hi, move the light component before the display component and it should work. The definition order in the YAML determines the order of initialization of the components. In your case, ehmtx will start its thing before the underlying components are working. 

Hope it helps, please close the issue if it is working

TheGroundZero, 2023-08-17:
Moving the light component between the font and display components fixed the issue :)

@lubeda
Another issue has popped up however, the new font it too big and the top of the time and date is chopped off...
So either mateine is bigger than Calcium, of there's been a change to the font size compared to v1.

closed 2023-08-17 by TheGroundZero