lubeda/EspHoMaTriXv2 #78: Installation with ESPHome 2023.8.2 and default fonts from ulanzi-easy.xaml not possible

CM000n, 2023-09-03:
# Bug report

## Describe the bug

When I try to install the config from the current ulanzi-easy.yaml file with ESPHome 2023.8.2, the installations aborts with the following error message when trying to convert the `MatrixClock-Light8.bdf` and `MatrixClock-Chunky6.bdf` fonts to Pillow format
```bash
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/esphome/esphome/__main__.py", line 1023, in main
    return run_esphome(sys.argv)
  File "/esphome/esphome/__main__.py", line 1010, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/esphome/esphome/__main__.py", line 441, in command_run
    exit_code = write_cpp(config)
  File "/esphome/esphome/__main__.py", line 185, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 197, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 626, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 177, in wrapped
    await coro(conf)
  File "/esphome/esphome/components/font/__init__.py", line 360, in to_code
    font = load_bitmap_font(CORE.relative_config_path(conf[CONF_PATH]))
  File "/esphome/esphome/components/font/__init__.py", line 334, in load_bitmap_font
    pil_font_path = convert_bitmap_to_pillow_font(filepath)
  File "/esphome/esphome/components/font/__init__.py", line 327, in convert_bitmap_to_pillow_font
    return cv.file_(local_pil_font_file)
  File "/esphome/esphome/config_validation.py", line 1397, in file_
    raise Invalid(
voluptuous.error.Invalid: Could not find file 'config/config/.esphome/font/31010cf8/MatrixClock-Light8.pil'. Please make sure it exists (full path: /config/config/config/.esphome/font/31010cf8/MatrixClock-Light8.pil).
```

## Additional information

- used Hardware:
  - Ulanz LED Clock

## To Reproduce

Steps to reproduce the behavior:
Try to install current ulanzi-easy.yaml Config with ESPHome 2023.8.2

## Expected behavior

ESPHome sould install without any error

## Configuration

```yaml
# https://github.com/lubeda/EspHoMaTriXv2
substitutions:
  devicename: "ulanzi-led-matrix"
  friendly_name: "Ulanzi LED Matrix"
  # Pin definition from https://github.com/aptonline/PixelIt_Ulanzi 
  battery_pin: GPIO34 
  buzzer_pin: GPIO15 
  ldr_pin: GPIO35 
  matrix_pin: GPIO32 
  left_button_pin: GPIO26 
  mid_button_pin: GPIO27 
  right_button_pin: GPIO14 
  scl_pin: GPIO22 
  sda_pin: GPIO21 

esphome:
[... 293 lines cut ...]
    - id: weather_fog
      lameid: 52167
    - id: weather_sunny
      lameid: 52156
    - id: sleep8x32
      url: https://user-images.githubusercontent.com/16407309/224850723-634c9b2d-55d9-44f2-9f93-765c0485b090.gif 
  on_next_screen:
    - homeassistant.event:
        event: esphome.new_screen
        data_template:
          iconname: !lambda "return icon.c_str();"
          text: !lambda "return text.c_str();"

text_sensor:
  - platform: wifi_info
    ip_address:
      name: "${devicename} IP Address"
    ssid:
      name: "${devicename} SSID"
    bssid:
      name: "${devicename} BSSID"
    mac_address:
      name: "${devicename} Mac Address"

```

trip5, 2023-09-03:
There's a clue in your error message.  Did you put the BDF font files on the same machine that is running ESPHome?  They should be in a folder called 'fonts'.

lubeda, 2023-09-03:
Normaly a missing font or wrong font path raises an other error. can you try to "clean build files" before recompiling?

CM000n, 2023-09-03:
> There's a clue in your error message. Did you put the BDF font files on the same machine that is running ESPHome? They should be in a folder called 'fonts'.

ESPHome runs as a Docker container and I have created a `fonts/` directory with the fonts inside the ESPHome
[image]
 config directory that is linked to the Docker image.

trip5, 2023-09-03:
If your ESPHome yaml files are all in esphome/config then that's a genuine error... but if your yaml files are in the esphome folder, you need to change the path under the font: section.

lubeda, 2023-09-03:
Versuch mal `chmod 0777 *.bdf `in dem Verzeichnis. Bei der Konstellation kann ich nicht sagen wie die rechte sein müssen.

CM000n, 2023-09-03:
> If your ESPHome yaml files are all in esphome/config then that's a genuine error... but if your yaml files are in the esphome folder, you need to change the path under the font: section.

Sorry, I don't understand @trip5. ESPHome knows the path and it exists. If I tried to enter a path that does not exist, the error message would be the following:

```bash
Failed config

font: [source config/ulanzi-led-matrix.yaml:134]
  - 
    Could not find file 'config/nonexistentpath/MatrixClock-Light8.bdf'. Please make sure it exists (full path: /config/config/nonexistentpath/MatrixClock-Light8.bdf).
    file: nonexistentpath/MatrixClock-Light8.bdf
    id: default_font
    glyphs: |-
      ! "#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz°
  - 
    Could not find file 'config/nonexistentpath/MatrixClock-Chunky6.bdf'. Please make sure it exists (full path: /config/config/nonexistentpath/MatrixClock-Chunky6.bdf).
    file: nonexistentpath/MatrixClock-Chunky6.bdf
    id: special_font
    glyphs: ! "#$%&'()*+,-./0123456789:APMTapmt
```

CM000n, 2023-09-03:
> Versuch mal `chmod 0777 *.bdf `in dem Verzeichnis. Bei der Konstellation kann ich nicht sagen wie die rechte sein müssen.

No, extended file permissions do not change anything. Shall we continue the conversation in German? Maybe English would be better, in case someone with the same problem comes across this issue thread?

trip5, 2023-09-03:
I think lubeda is on the right track with permissions.  Have you got Portainer?  Sometimes I use it to get a shell inside the container and run tests like copying and deleting files.  If the container shell has issues, ESPHome probably will too.

Sometimes, I just throw my hands up and run the container with --privileged and hope that works and try to fix permissions later.

lubeda, 2023-09-03:
OK, so your error is either a local problem or a bug in esphome. I assume it's a local problem.

CM000n, 2023-09-03:
> I think lubeda is on the right track with permissions. Have you got Portainer? Sometimes I use it to get a shell inside the container and run tests like copying and deleting files. If the container shell has issues, ESPHome probably will too.
> 
> Sometimes, I just throw my hands up and run the container with --privileged and hope that works and try to fix permissions later.
> 
> Edit: looks like I actually am running the ESPHome container like that. Probably not the best solution but... sigh, linux...

No need for portainer ;-)
```bash
docker exec -it esphome bash
```

I can't find a permission problem within the container either. The owners and group rights of the folders and files are set correctly, even recursively. So it must be something else.

CM000n, 2023-09-03:
Interesting. Esphome seems to have created the right files. But then it uses the wrong path when compiling.

```bash
root@esphome:/config/config/.esphome/font/31010cf8# ls -a -l
total 48
drwxrwxrwx+ 1 root users   132 Sep  3 21:04 .
drwxrwxrwx+ 1 root users    16 Sep  3 21:04 ..
-rwxrwxrwx+ 1 root users 35741 Sep  3 21:04 MatrixClock-Light8.bdf
-rwxrwxrwx+ 1 root users   608 Sep  3 22:16 MatrixClock-Light8.pbm
-rwxrwxrwx+ 1 root users  5142 Sep  3 22:16 MatrixClock-Light8.pil
```

trip5, 2023-09-03:
Well I can confirm it's not the latest version of ESPHome... I just updated to official 2023.8.2 and it compiled just fine.  I'm honestly lost to help you other than to paste from my own yaml which I can confirm compiles with the latest ehmtx too.  Of course, this doesn't match the official yaml and I keep all my icons local so perhaps remove that before working with it.

I can say I keep the font: section much higher.  Would that affect anything you think?

Sorry but I'm past my bedtime.  Good luck.  I'll check back later!

````
substitutions:
  devicename: pixelclock-bunny
  friendly_name: PixelClock-bunny
# Pin definition from https://github.com/aptonline/PixelIt_Ulanzi 
  ldr_pin: GPIO35 
  ledpin: GPIO32 
  left_button_pin: GPIO26 
  mid_button_pin: GPIO27 
  right_button_pin: GPIO14 
  battery_pin: GPIO34 
  buzzer_pin: GPIO15 
  scl_pin: GPIO22 
  sda_pin: GPIO21

esp32:
  board: esp32dev # for Ulanzi TC-001

switch:
  - platform: template
    name: "$friendly_name Auto-Adjust Brightness"
    id: switch_autobrightness
    icon: mdi:brightness-auto
    restore_mode: RESTORE_DEFAULT_ON
    lambda: |-
      if (id(aab_enable)) {
        return true;
      } else {
        return false;
      }
    turn_on_action:
      lambda: |-
        id(aab_enable) = true;
    turn_off_action:
      lambda: |-
        id(aab_enable) = false;

globals:
  # aab = auto-adjustable brightness
  - id: aab_enable
    type: "bool"
    restore_value: true # Pixel clock change both to false (unless you fix the ldr)
    initial_value: "true"
  - id: aab_add
    type: int
    initial_value: '20'
  - id: aab_max
    type: int
    initial_value: '200'
  - id: aab_min
    type: int
    initial_value: '20'

external_components:
  - source:
      type: git
      url: https://github.com/lubeda/EspHoMaTriXv2
      ref: main
    refresh: 60s
    components: [ ehmtxv2 ]
#  - source:
#      type: local
#      path: my-components/ehmtxv2-2023.7.0 # e.g. /config/esphome/components
#    components: [ ehmtxv2 ]
[... 240 lines cut ...]
      file: ehmtxv2/weather_sunny.gif
    - id: weather_windy
      file: ehmtxv2/weather_windy.gif
    - id: weather_windy_variant
      file: ehmtxv2/weather_windy_variant.gif
    - id: weather_cloudy_night
      file: ehmtxv2/weather_cloudy_night.gif
    - id: bitcoin
      file: ehmtxv2/bitcoin.gif
  on_next_screen:
    - homeassistant.event:
        event: esphome.new_screen
        data_template:
          iconname: !lambda "return icon.c_str();"
          text: !lambda "return text.c_str();"

image: # here because ESPHome 2023.7.0 has broken something
    - file: ehmtxv2/1pixel.gif
      id: breaking20237

animation: # here because ESPHome 2023.7.0 has broken something
    - file: ehmtxv2/1pixel.gif
      id: breaking202371

````

CM000n, 2023-09-03:
Thank you for your yaml config @trip5. But the problem really seems to be with ESPHome or the ESPHome Docker container and the "convert_bitmap_to_pillow_font" function. It is now also time for me to go to bed. I'll take a closer look tomorrow.

trip5, 2023-09-04:
I just wanted to swing by and say I did not have a problem with the ulanzi-easy.yaml file, provided I tacked this on at the end:

````
image: # here because ESPHome 2023.7.0 has broken something
    - file: ehmtxv2/1pixel.gif
      id: breaking20237

animation: # here because ESPHome 2023.7.0 has broken something
    - file: ehmtxv2/1pixel.gif
      id: breaking202371
````
All I can think is your ESPHome container is bad.

A year or so ago ESPHome gave me some hiccups and there were a lot of leftover garbage files so I copied out my yamls, external files, and secrets file... then deleted the entire config folder, removed the container, and started fresh.

lubeda, 2023-09-04:
Can you check if the pdf file is ok. You can use a text editor, the first line should be "STARTFONT 2.1" not "<HTML>. perhaps your pdf file needs to be downloaded again.

CM000n, 2023-09-05:
The .bdf files seems to be valid.
Can you please share your compose file, if you use docker-compose @trip5 ?
I started all over again. Büro my container behaves the same.

CM000n, 2023-09-05:
very wierd. My docker container always creates a config folder inside the config folder that is linked as docker volume. 
I think I'll close this issue here and open a new one in the ESPHome repo, as it seems to be a more general problem.
I'll copy the new issue here later as a reference.

CM000n, 2023-09-05:
https://github.com/esphome/issues/issues/4853

closed 2023-09-05 by CM000n

trip5, 2023-09-05:
Sure.  Actually I have a bad (?) habit of avoiding docker files if I can do it via command line:

````
sudo docker run -d --name esphome --privileged --restart=unless-stopped -e TZ=Asia/Seoul -v '/home/trip5/docker/esphome/:/config/' -v /etc/localtime:/etc/localtime:ro --network=host esphome/esphome:latest
````

I also have a bad habit of using the host network (almost everything unless I need to re-map ports) and running privileged containers (actually just ESPHome, Home Assistant).  I don't need a password because it's behind Authentik.

I'm pretty sure ping is on by default so you don't need to add it.  Also your vars for PATH, LANG, LC_ALL, and PLATFORMIO_GLOBALLIB_DIR all appear to conform to the default (I checked mine with Portainer) so why bother specifying them?  Same question for the COMMAND.  You also don't specify an ENTRYPOINT... the default is /entrypoint.sh by the way.  I don't know if there is some side-effect of specifying one but not the other...

Another difference I see is you're using an environment variable to pass the location of the config folder... I've always explicitly specified mine.  Not sure if any of this will help you.  In general, though, in my experience with docker, I've sometimes had some issues when specifying things that didn't need to be specified... Here's a pretty bare-bones compose file that I probably used before working out the command-line:

````
version: '3'

services:
  esphome:
    container_name: esphome
    image: esphome/esphome:latest
    volumes:
      - '/home/trip5/docker/esphome/:/config/'
      - /etc/localtime:/etc/localtime:ro
    restart: unless-stopped
    privileged: true
    network_mode: host
````

CM000n, 2023-09-05:
> Sure. Actually I have a bad (?) habit of avoiding docker files if I can do it via command line...

Thank you. I like doing things explicitly rather than trusting that they will work implicitly. ;-)