lubeda/EspHoMaTriXv2 #79: Error after Home Assistant update 2023.09.0

clfberlin, 2023-09-07:
Unfortunately Home Assistant version 2023.09.0 came with changes relevant for EspHoMaTriXv2.
esphome.[device]_icon_screen is considered an _unkown service_.

clfberlin, 2023-09-07:
Strange. If triggered under Developer Tools > Services, it works. It just complains about being used in the automations.
It seems that the message doesn't appear when "default_font" is set to "true".

lubeda, 2023-09-07:
can you provide the original error message?

clfberlin, 2023-09-07:
I was able to trigger it again by reactivating an older automation. It's German, but you should be able to cope with that. :-)
[image]

The full message is
```
Ulanzi-EG Backup verwendet einen unbekannten Dienst
Die Automatisierung "Ulanzi-EG Backup" (automation.ulanzi_eg_backup) hat eine Aktion, die einen unbekannten Dienst aufruft: esphome.ulanzi_eg_add_screen.

Dieser Fehler verhindert, dass die Automatisierung ordnungsgemäß ausgeführt wird. Vielleicht ist dieser Dienst nicht mehr verfügbar oder vielleicht hat ein Tippfehler ihn verursacht.

Um diesen Fehler zu beheben, [bearbeite die Automatisierung](http://blabla) und entferne die Aktion, die diesen Dienst aufruft.

Drücke unten auf SENDEN, um zu bestätigen, dass du diese Automatisierung korrigiert hast.

```
[image]

clfberlin, 2023-09-07:
Sorry, just realized after posting that this automation is indeed old "add_screen". 
But I got the same messages for my Ulanzis with their current automations and _icon_screen
Unfortunately (or maybe fortunately) I can't reproduce those anymore since I edited the YAML for those automations mentioned.

lubeda, 2023-09-07:
kann passieren!

closed 2023-09-07 by lubeda

trip5, 2023-09-07:
Just checked myself after updating Home Assistant.  No issues.  I was about to suggest removing the clocks from ESPHome integration and re-adding them... it helps clear out the old services that way, too.

clfberlin, 2023-09-07:
Ok, I apologize. If I can't reproduce it at the moment and no one else seems to have issues, it was probably just a local hickup after the update. Sorry...

clfberlin, 2023-09-13:
Oddly the error reappears with every update (and when I restore to a previous version). In this case Home Assistant 2023.9.2.
I works, though - despite the error message.

[image]

trip5, 2023-09-13:
Sorry I can't read German and my auto-translator isn't great... Did you try removing the Ulanzi devices completely from Home Assistant?  That really helps to clear out old entities.

Try also looking at the Entities screen and checking for certain that there isn't something like esphome.ulanzi_eg_2_icon_screen or something like that.  Anytime I see that _2_ I go and remove the devices and re-add them...

But this sometimes means I also have to go check the automations and be sure they're referring to the right entity.

clfberlin, 2023-09-13:
I apologize. That wasn't very thoughtful of me. Let me just translate the critical passage... The automation xyz has an action that calls up an unknown service device_icon_screen.
I am supposed to edit the automation.
However, the automation actually works. And the service _icon_screen also behaves as it should. So there isn't actually an error - just an error message. 
Thanks for the tip. But since everything does work, I have not yet removed the Ulanzis. And as far as I see, it is not a wide spred problem, which is why I just entered this here for others to find if they happen to get the same error message. All is good. And once i touch the automations and acknowledge the error message, it goes away (until the next update).
Thanks and cheers!