lubeda/EspHoMaTriXv2 #80: [BUG] [Wiki] night-mode.md

andrewjswan, 2023-09-14:
# Bug report
```
  [on_empty_queue] is an invalid option for [ehmtxv2]. Did you mean [allow_empty_screen]?
  on_empty_queue: 
    then: 
      lambda: |-
        if (id(night))
        {
          id(rgb8x32)->clock_screen(10,10,true,140,120,120);
          id(rgb8x32)->date_screen(10,5,true,120,200,150);
        }
        else 
        {
          id(rgb8x32)->blank_screen(1,15);
        }
```
## Describe the bug

[`on_empty_queue`] is an invalid option for [ehmtxv2]. Did you mean [allow_empty_screen]?

## Additional information

- used Hardware:
  - ESP32 on Ulanzi TC001

## To Reproduce

Steps to reproduce the behavior:
Add to config:
```
  on_empty_queue: 
    then: 
      lambda: |-
```

andrewjswan, 2023-09-15:
Same thing with the weather article https://github.com/lubeda/EspHoMaTriXv2/wiki/weather-display
The `esphome.ulanzi_del_screen` service requires mandatory `mode` specification.
The `esphome.ulanzi_add_screen` service does not exist
All existing services `require mandatory filling of all fields`, including color.

PS: What do rainbow modes do?

lubeda, 2023-09-16:
Hi, the `on_empty_queue` action is part of the not official released version 2023.8.0 which has plenty of breaking changes...

i updated the wiki article to indicate the needed version.

lubeda, 2023-09-16:
> PS: What do rainbow modes do?

it changes the colors of the text like a rainbow

andrewjswan, 2023-09-16:
> it changes the colors of the text like a rainbow

How can I display Hours in one color, Minutes in another color, Seconds in a third color and dividers in a fourth color?

andrewjswan, 2023-09-19:
> the `on_empty_queue` action is part of the not official released version 2023.8.0 which has plenty of breaking changes

When is the 2023.8.0 release expected?

closed 2023-10-05 by andrewjswan