lubeda/EspHoMaTriXv2 #81: [BUG] Time Jump in %H:%M:%S format

andrewjswan, 2023-09-14:
# Bug report

## Describe the bug

If I output the time in `%H:%M:%S` format, when 1 occurs in seconds, it jumps constantly, this was not the case on the original firmware Ulanzi TC001.
How can this be solved? Changing the font? Or an additional parameter in the settings?

## Additional information

- used Hardware:
  - ESP32
  - EspHoMaTriXv2 version: 2.0.0
  - Ulanzi TC001

## To Reproduce

Steps to reproduce the behavior:

Output the time in `%H:%M:%S` format with the font `MatrixClock-Chunky6.bdf`.

## Expected behavior

So that when outputting in `%H:%M:%S` format, the starting position is the same and does not change during output. The font is monospaced, so stable output is expected.

## Configuration

```text
font:
  - file: fonts/MatrixClock-Chunky8.bdf
    id: default_font
    glyphs:  |
      ! "#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz°
  - file: fonts/MatrixClock-Chunky6.bdf
    id: special_font
    glyphs:  |
      ! "#$%&'()*+,-./0123456789:APMTapmt

ehmtxv2:
  id: rgb8x32
  icons2html: true
  matrix_component: ehmtx_display
  time_component: ehmtx_time
  time_format: "%H:%M:%S"
  date_format: "%d/%m"
  default_clock_font: false
  show_seconds: false
  default_font_id: default_font
  default_font_yoffset: 8
  special_font_id: special_font
  special_font_yoffset: 7
```

trip5, 2023-09-14:
Hey there.  I'm the creator of these fonts you're referring to. Actually, the one of Chunky6 is variable width.  If you look closely, you'll notice the colon also only uses 1 column and 1 space column...  I didn't realize that would pretty inconvenient, especially if you decide to display seconds.

Can you close this issue?  It's not an EHMTX issue... it's an issue with my font.  I've already changed the Chunky8 fonts so the numbers all have the same width... I'll change the Chunky6 font, too.  Check my repo and re-download what you need.  I'll have it updated in less than an hour.  https://github.com/trip5/MatrixClockFonts

trip5, 2023-09-14:
Sidenote that I'm thinking I should create a slightly taller version of the Chunky fonts similar as I did with the X fonts and stretch everything to the full 6 pixel height.  That might make the numbers look better on our screens.  I don't really use the 6 pixel fonts, actually so it hasn't crossed my mind until now.  What do you think?

andrewjswan, 2023-09-15:
> Check my repo and re-download what you need. I'll have it updated in less than an hour.

Thank you so much. Everything is great with the new font. I'll close the issue.

andrewjswan, 2023-09-15:
> Sidenote that I'm thinking I should create a slightly taller version of the Chunky fonts similar as I did with the X fonts and stretch everything to the full 6 pixel height. That might make the [...]

On the contrary, I liked the fonts with a height of 6 the most, they look harmonious on the screen, especially with icons and when displaying the time and date. If only text is displayed, a font height of 8 is good. I'm even thinking of making the font height 6 as the main font. 
Can you tell me in what program you edit fonts, I lack the Ukrainian language, I want to add.

andrewjswan, 2023-09-15:
Solution - https://github.com/lubeda/EspHoMaTriXv2/issues/81#issuecomment-1720016043

closed 2023-09-15 by andrewjswan

trip5, 2023-09-15:
Can you open another issue over there?  And tell me specifically what you're looking for.  By the way, ESPHome and BDF still appear to have issues with Unicode/UTF encoding but if the characters you want function in the 8-series fonts, we can probably work something out!