lubeda/EspHoMaTriXv2 #82: [FEATURE REQUEST] Add alert_screen to display instant messages

andrewjswan, 2023-09-19:
# Feature Request

### Describe the solution / feature you'd like

Add alert_screen to display instant messages:
`alert_screen` or `info_screen` with parameters ("icon_name", "text", "screen_time", "default_font", "r", "g", "b")

`screen_time` - how long is this screen display in the loop (seconds), also `lifetime` is set automatically to `screen_time`. And immediately `force_screen` is called for this screen. 
Each next call to `alert_screen` or `info_screen` replaces the previous one (while the previous one lives on), and sets the new passed values for `screen_time` and `lifetime`.
This will be convenient for displaying instant messages, especially if they need to be displayed once, say within 15sec, and will replace the inconvenient co in the form:
- del_screen
- icon_screen (You can't set the `lifetime` less than a minute, so you'll have to wait and delete it, if you want to show a message, like 15 seconds)
- force_screen
- delay 
- del_screen

### Describe alternatives you've considered

- del_screen
- icon_screen (You can't set the `lifetime` less than a minute, so you'll have to wait and delete it, if you want to show a message, like 15 seconds)
- force_screen
- delay 
- del_screen

lubeda, 2023-09-19:
For me force screen works, especially with all modes, so I need an example in detail why your suggestion is helpful.

andrewjswan, 2023-09-19:
I need to show an instant message with an icon, within 15 sec. Now it happens like this:
- `del_screen`
- `icon_screen` with `screen_time: 15` and `lifetime: 1` (Minimum possible value)
- `force_screen`
- `delay 15s`
- `del_screen`

If make the changes I suggest, it will all be called in one command, and the screen will always display the actual instant message.
- `alert_screen` with `screen_time: 15`

Seems logical, convenient, and simple to me.

andrewjswan, 2023-09-20:
Like this:
```
  EHMTX_queue *EHMTX::find_mode_queue_element(int mode)
  {
    for (size_t i = 0; i < MAXQUEUE; i++)
    {
      if ((this->queue[i]->mode == mode))
      {
        ESP_LOGD(TAG, "free_screen: found by mode %d", mode);
        return this->queue[i];
      }
    }
    return this->find_free_queue_element();
  }

  void EHMTX::alert_screen(std::string iconname, std::string text, int screen_time, bool default_font, int r, int g, int b)
  {
    uint8_t icon = this->find_icon(iconname.c_str());

    if (icon >= this->icon_count)
    {
      ESP_LOGW(TAG, "icon %d not found => default: 0", icon);
      icon = 0;
      for (auto *t : on_icon_error_triggers_)
      {
        t->process(iconname);
      }
    }
    EHMTX_queue *screen = this->find_mode_queue_element(MODE_ALERT_SCREEN);

    screen->text = text;
    screen->endtime = this->clock->now().timestamp + screen_time;
    screen->text_color = Color(r, g, b);
    screen->default_font = default_font;
    screen->mode = MODE_ALERT_SCREEN;
    screen->icon_name = iconname;
    screen->icon = icon;
    screen->calc_scroll_time(text, screen_time);
    for (auto *t : on_add_screen_triggers_)
    {
      t->process(screen->icon_name, (uint8_t)screen->mode);
    }
    ESP_LOGD(TAG, "alert screen icon: %d iconname: %s text: %s screen_time: %d", icon, iconname.c_str(), text.c_str(), screen_time);
    screen->status();

    force_screen(iconname, MODE_ALERT_SCREEN);
  }

  void EHMTX::force_screen(std::string icon_name, int mode)
  {
    for (uint8_t i = 0; i < MAXQUEUE; i++)
    {
      if (this->queue[i]->mode == mode)
      {
        if ((mode == MODE_ICON_SCREEN) || (mode == MODE_FULL_SCREEN) || (mode == MODE_RAINBOW_ICON))
        {
          if (strcmp(this->queue[i]->icon_name.c_str(), icon_name.c_str()) == 0)
          {
            ESP_LOGD(TAG, "force_screen: found position: %d", i);
            this->queue[i]->last_time = 0;
            this->queue[i]->endtime += this->queue[i]->screen_time_;
            this->next_action_time = this->clock->now().timestamp;
            ESP_LOGW(TAG, "force_screen: icon %s in mode %d", icon_name.c_str(), mode);
          }
        }

        if ((mode == MODE_ALERT_SCREEN))
        {
          if ((this->queue[i]->mode == MODE_ALERT_SCREEN))
          {
            ESP_LOGD(TAG, "force_screen: found position: %d", i);
            this->queue[i]->last_time = 0;
            this->queue[i]->endtime += this->queue[i]->screen_time_;
            this->next_action_time = this->clock->now().timestamp;
            ESP_LOGW(TAG, "force_screen: icon %s in mode %d", icon_name.c_str(), mode);
          }
        }
      }
    }
  }
```

lubeda, 2023-09-23:
Now I understand! I don't like the concept of showing the message only once, but it doesn't "hurt" in the code. So P will include it within the next days.

andrewjswan, 2023-09-23:
> I don't like the concept of showing the message only once, but it doesn't "hurt" in the code

In my opinion, this is the most convenient way to show instant messages, such as incoming call, air alert, etc. They are shown once for 15/30 seconds, and that's it, they attracted attention, if you need to show constantly, then for this there are already all the necessary mechanisms.

closed 2023-09-24 by lubeda

lubeda, 2023-10-01:
Parameter name was wrong, corrected to `icon_name` instead of `iconname`

andrewjswan, 2023-10-03:
@lubeda For uniformity, it's worth changing https://github.com/andrewjswan/EspHoMaTriXv2/commit/3075fd8da7de53fe058a775434d1abe61eaeba57