lubeda/EspHoMaTriXv2 #84: [FEATURE REQUEST] Increase the number of icons

andrewjswan, 2023-09-19:
# Feature Request

### Is your feature request related to a problem?

There is no way to pass an icon to EspHoMaTriXv2 from HA, but want to have flexibility, variety of information displayed. 90 icons are not always enough.

### Describe the solution / feature you'd like

Is it possible to increase the number of icons to at least 100?
What are these limitations associated with?

### Describe alternatives you've considered

Or maybe could add transferring icons from HA to EspHoMaTriXv2 when adding a new screen.

trip5, 2023-09-19:
Since I'm not lubeda, I can't give a definitive answer but I'd say it's a rough guess of flash size limitation.  Have you tried more than that?  I'd say go for it and see what happens.  At worst, your compile fails and now you have an error log to share and we can all take a gander at it.

By the way, sending icons from Home Assistant would be... weird at best.  I'm guessing the ESPHome integration probably communicates with the device via some MQTT-similar protocol so it you couldn't send the file directly, it have to be re-encoded as text, something like uuencoding... which... would be slow as heck.  The only reason Awtrix could do this was because it ran a special server that controlled the display directly...  With ESPHome, a sent icon would have to be stored in the ESP's ram, then discarded... so you could only send one, use it, then send another... And uuencode would double (or more) the data being sent...  With ESPHome, there's no (easy) way to save the file for re-use later.

andrewjswan, 2023-09-19:
Well I tried adding larger than 90 and got an error, I can post it here but it's trivial, size larger than 90 and that's it.

andrewjswan, 2023-09-19:
About the transfer of icons from HA, I understand the mechanism, you can organize a cache for 2-3 icons and other methods, it will not be a flexible solution, but it will work

lubeda, 2023-09-19:
Yes there is a limitation in flash size, so you are somewhat limited. The 90 icons limit can be increased but depending on your code e.g. used sensors there will be compile errors.

in the unreleased version [2023.8.0](https://github.com/lubeda/EspHoMaTriXv2/tree/2023.8.0) are some work arounds. you may define your icons in the service call. so unlimited icons.

```
bitmap_screen("[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,63519,63519,63519,63519,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,63519,0,0,0,0,2016,0,0,0,0,0,0,0,0,0,0,31,0,0,0,0,0,0,0,0,0,63488,0,63488,0,0,0,63519,0,0,0,0,2016,2016,0,0,0,65514,0,65514,0,0,0,31,0,0,0,64512,0,0,64512,0,63488,63488,0,63488,63488,0,0,63519,63519,63519,0,0,2016,0,2016,0,65514,0,65514,0,65514,0,31,31,31,0,0,0,64512,64512,0,0,63488,63488,63488,63488,63488,0,0,63519,0,0,0,0,2016,0,2016,0,65514,0,65514,0,65514,0,0,31,0,0,0,0,64512,64512,0,0,0,63488,63488,63488,0,0,0,63519,63519,63519,63519,0,2016,0,2016,0,65514,0,65514,0,65514,0,0,0,31,31,0,64512,0,0,64512,0,0,0,63488,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]",1,10);
bitmap_small("[2016,0,0,0,2016,0,0,0,2016,0,0,0,2016,0,0,0,2016,0,0,0,2016,0,0,0,0,2016,0,2016,0,31,31,0,0,0,2016,0,31,0,0,31,0,0,0,0,0,0,31,0,0,0,0,0,0,31,0,0,0,0,0,0,31,31,31,31]", "2023.x.y", 1, 10);

```

andrewjswan, 2023-09-19:
> in the unreleased version [2023.8.0](https://github.com/lubeda/EspHoMaTriXv2/tree/2023.8.0) are some work arounds. you may define your icons in the service call. so unlimited icons.

It's more like a full screen picture rather than an icon next to which text will be displayed.
But in general, we'll have to see how it works. 

> Yes there is a limitation in flash size, so you are somewhat limited. The 90 icons limit can be increased but depending on your code e.g. used sensors there will be compile errors.

I'm looking at the size that ESPHome displays when flashing, there's still a margin there (on esp32), but I haven't tried 100 icons, has there been a problem with that?

andrewjswan, 2023-09-20:
Decided not to open another Issue, to ask here, how about changing the starting rectangle to a rectangle extending from the center.
https://github.com/lubeda/EspHoMaTriXv2/blob/main/components/ehmtxv2/EHMTX.cpp#L650-L651
```
      uint8_t w = 2 + ((uint8_t)(32 / 16) * (i / 16)) % 32;
      uint8_t l = 32 / 2 - w / 2 ;
      this->display->rectangle(l, 2, w, 4, this->rainbow_color); // Color(120, 190, 40));
```

Or just show `BOOT LOGO` with rainbow color at startup if it is specified.

lubeda, 2023-09-22:
> Or just show `BOOT LOGO` with rainbow color at startup if it is specified.

At this early stage in the boot process, the device doesn't know anything about the `BOOT_LOGO` etc. so only native capabilities do work, like this rectangle. But your code is in the 2023.8.0 branch. Thanks

andrewjswan, 2023-09-22:
> At this early stage in the boot process, the device doesn't know anything about the `BOOT_LOGO` etc. so only native capabilities do work, like this rectangle

We can use the logo boot as a drawing primitive, i.e. it will be a bit pattern on which we just output points in the right places, as we do with a rectangle, and change the color. I can try to do that.

andrewjswan, 2023-09-25:
> At this early stage in the boot process, the device doesn't know anything about the `BOOT_LOGO` etc. so only native capabilities do work, like this rectangle.

Like this: https://wokwi.com/projects/376831384418125825