lubeda/EspHoMaTriXv2 #85: [FEATURE REQUEST] Add a screen with an icon and a time with the day of the week.

malinovsku, 2023-09-20:
# Feature Request

### Describe the solution / feature you'd like

It would be great if you added an additional display option similar to icon_screen, but instead of text that would display the time and day of the week. The name of the new function could be selected icon_time,
parameters to run: "icon_name", "lifetime", "screen_time", "default_font", "r", "g", "b"

### Additional context

As an example of a screens, displaying the weather and time icons with the day of the week.
[image]

andrewjswan, 2023-09-21:
Interesting idea as an additional time display screen, or as an alternate screen with getting the weather from the HA.

lubeda, 2023-09-23:
It's implemented in 2023.8.0. Can you please give me an example for the wiki?

lubeda, 2023-09-23:
https://github.com/lubeda/EspHoMaTriXv2/wiki/icon-clock

malinovsku, 2023-09-25:
@lubeda thank you, the display is the right one, but the icon moves two columns to the right, while everything is as it should be in icon_screen, as I understood it in the [code](https://github.com/lubeda/EspHoMaTriXv2/blob/2023.8.0/components/ehmtxv2/EHMTX_queue.cpp#L343), but it would be better if the icon starts from the very edge without indentation and the clock is shifted to the middle of the remaining one, since when changing displays with icons, they are like they would jump because of the difference in boundaries.
[image]

lubeda, 2023-09-25:
Upps, i will fix this.

malinovsku, 2023-09-27:
@lubeda after a couple of days of using this display, I realized that there are not enough variations of icons with hours and days of the week, so that they could be created in the same amount as, for example, icon_screen and deleted by the del_screen service. as an example of usage, display both a calendar icon with a clock and a weather icon with a clock.
There may be a conflict when deleting, and https://github.com/lubeda/EspHoMaTriXv2/issues/83 it would help in solving this issue, to delete by screen id

andrewjswan, 2023-09-28:
Interesting idea, it can be well developed, for example to output date with hours, or date with day of week and year, etc.
For example, for the date, we take the backing icon `calendar`
[image]
And on top of that, display the current date:
[image]
And next to it we can display either time or date, also we can display the current day of the week:
[image] [image]

And we can rooluchat let's say such a variant of the screen (here you can even safely use the crawl line).
[image]
Or this one:
[image]
[image]

andrewjswan, 2023-09-28:
Since we come to an increase of screens with the same information, e.g. Clock, Icon with Clock, Date, Icon with Date, it would be nice to give the possibility to define a starting set of screens in EspHoMaTriXv2 configuration  (if not specified, the default is time, date), for example:
```
emhtx:
  screens:
    - icon_time
    - icon_date
```

Also immediately add localized abbreviations to display the current day of the week (if not specified, the default is English):
```
emhtx:
  week_days:
    - ПН
    - ВТ
    - СР
    - ЧТ
    - ПТ
    - СБ
    - ВС
```
That would solve the problem  https://github.com/lubeda/EspHoMaTriXv2/issues/89

andrewjswan, 2023-09-28:
On the concept of outputting an icon and text on top of it, it is necessary to think through a convenient mechanism for specifying what to output, let's say type, value, for example:
- weather, *
- calendar, day
- calendar, week_day

malinovsku, 2023-09-28:
> Upps, i will fix this.

I updated the firmware, and the icon is now in the right place, thank you, but the days of the week are not in the middle at the bottom, 3 indent on the left, and 1 indent on the right.
[image]

lubeda, 2023-09-29:
Hopefully finally fixed. @malinovsku Please check again

malinovsku, 2023-10-01:
@lubeda everything is great now, thank you!

closed 2023-10-01 by lubeda