lubeda/EspHoMaTriXv2 #89: [FEATURE REQUEST] [Wiki] Localized abbreviated day of the week

andrewjswan, 2023-09-26:
# Feature Request

### Describe the solution / feature you'd like

This is rather a question for Wikipedia, whether it is possible to make the output of the date with localized abbreviated day of the week (ПО, ВТ or По, Вт), or at least in English but not 3 letters but 2, like MO, TU.... Or Mo, Tu... 3 letters take up a lot of space.

andrewjswan, 2023-09-26:
Or give the possibility to specify a pseudo value like %l (for Mo, ...) or %L for (MO, ...) and before displaying 
https://github.com/lubeda/EspHoMaTriXv2/blob/main/components/ehmtxv2/EHMTX_queue.cpp#L233
check if there is %l or %L in the string then replace this part of the string with the current day of the week of 2 letters, if %L then bring to upper case.

trip5, 2023-09-26:
When playing around how to properly get local time without the leading zero, I was annoyed ESPHome's documentation seemed incomplete... Try this: https://man7.org/linux/man-pages/man3/strftime.3.html

Note that ESPHome doesn't recognize everything from this list.  And the catch appears to be that your installation of ESPHome has to be able to get your locale settings anyway.  Can you use Portainer, get a console and type "date"?  If it's showing English, that's probably issue #1.  And it might be a common issue: https://www.reddit.com/r/Esphome/comments/shcbw5/question_can_strftime_be_localized/

Also, check this: https://github.com/lubeda/EspHoMaTriXv2/wiki/time-format - if we can be successful showing your local language, we can add more info to this Wiki entry.

But I fear you may be correct, localized time may need some "replacement value" - perhaps a 7 word string in the ehmtxv2 section that replaces %a with other strings?  That way users could even substitute their own desired strings...

andrewjswan, 2023-09-27:
[image]

trip5, 2023-09-27:
That seems like expected behavior... Ugly but it would squeeze if you use a decimal or pipe or bar?

Still... String substitution would be pretty cool...

Mon -> M (or whatever characters)

andrewjswan, 2023-09-27:
> Try this: https://man7.org/linux/man-pages/man3/strftime.3.html

I looked there, but couldn't find how to output the current day of the week in 2 letters.

> But I fear you may be correct, localized time may need some "replacement value" - perhaps a 7 word string in the ehmtxv2 section that replaces %a with other strings? That way users could even substitute their own desired strings...

Yes, that would be great, then everyone would be able to deduce the localized day of the week, even if it's one letter, two letters, or three.

> Ugly but it would squeeze if you use a decimal or pipe or bar?

Two letters would be great, localized 2 letters would be even better.

> Still... String substitution would be pretty cool...

I agree, that would be just fine.

andrewjswan, 2023-09-28:
> String substitution would be pretty cool...

Like this:
```
emhtx:
  week_days:
    - ПН
    - ВТ
    - СР
    - ЧТ
    - ПТ
    - СБ
    - ВС
```
or
```
emhtx:
  week_days:
    - 일
    - 월 
    - 화 
    - 수 
    - 목 
    - 금 
    - 토 
```
Hmm, I like it, I'd put it on my screen in Korean.
[image]

andrewjswan, 2023-09-28:
PS: @trip5 Do you happen to know if we have such a symbol in our fonts, a dot in the middle?
[image]

trip5, 2023-09-28:
I sure do. Exactly like that, 2 pixels wide. It's in all fonts as Unicode character 2022 •  <- This one (copy and paste).

And I'm not sure Korean days of the week would work so well.  Lol, Google tells me that Korean has 11,172 glyphs... way beyond the 255-character limit of ESPHome... but I suppose I could encode just a few like the days of week, the word for month (Koreans just use 1월 ~ 12월 and the 3 characters used by AM (오전) and PM (오후)... hmmm...

How hard would it be to allow character-substitution across the date AND time strings?  Would it be possible to run a regex on an internal string like that? Then, it should be simple to add to the code AND although a bit hard to read in the yaml, provide unlimited possibilities.

````
emhtx:
  replace_time_date: "/SUN/ПН|MON/ВТ|TUE/СР|WED/ЧТ|THU/ПТ|FRI/СБ|SAT/ВС/g"
````
https://stackoverflow.com/questions/16576983/replace-multiple-characters-in-one-replace-call

The only thing I'm not sure about is replacing multiple characters in one go...  It might be easier to run an array through a loop:
````
emhtx:
  replace_time_date: "SUN,ПН;MON,ВТ;TUE,СР;WED,ЧТ;THU,ПТ;FRI,СБ;SAT,ВС"
````

andrewjswan, 2023-09-28:
> How hard would it be to allow character-substitution across the date AND time strings?

If done as I suggested above, i.e. just an array with designations, then no substitution is needed, just get the current day of the week as a number, and take the desired value from the array. This is simple and efficient. In the Output Template we can add any substitution variant, for example `%#`, and change it to what we need.
The same can be done for Month. Just a set of required designations, default English. And that's it.

lubeda, 2023-09-28:
This does not belong to ehmtx, this belongs more to the language setting (if there is any) of esphome and the esphome time-component. But you can code a lambda triggered every minute with your wanted format and display it with "mode full screen". Be careful, string operations are RAM intensive!

andrewjswan, 2023-09-28:
Well we can always package them in PROGMEM, and it has little to do with Language settings, as you can see in the picture above, I cannot adequately output the current day name and date. The screen is small and three letters is a lot. 
And 14 letters, that's not that many.

trip5, 2023-09-28:
Actually I just did manage to localize a container of ESPHome into Ukrainian but... *sigh* can we petition ESPHome to recognize a language other than English?  There's already an issue open on "features request" here: https://github.com/esphome/feature-requests/issues/1520 It doesn't seem to have much traction.

Digging into EHMTX_queue.cpp I see the time is displayed directly using strftime which is hooked into the display functions of ESPHome...  The time itself is drawn using this function and it's *** annoying to manipulate that.

The alternative is very messy... I did actually play with extracting the time from strftime and manipulating the values with lambdas for an LED Clock and what used to be a 10 lines ended up being 50 lines.  Strftime is not a normal string... and running that every second still doesn't quite feel right.

Perhaps the easiest solution is... lubeda says the 2023.8 version has the ability to simply not show the default clock... so perhaps just get Home Assistant to send the time/date?

andrewjswan, 2023-09-28:
> The alternative is very messy...

Well I don't see a problem in replacing the pattern with a string value once per second, for Esp it's just nonsense.
Conceptually:
```
auto time = id(sntp_time).now();
auto day = week_days[time.day_of_week];
String timetext = date_mask;
timetext .replace("%#", day);
it.strftime(0, 0, id(font), timetext, id(time).now());
```

trip5, 2023-09-29:
Correction to previous post.  The bullet is Unicode character 00B7 • - oops.

andrewjswan, 2023-10-01:
It is possible not to implement it in the code, but to give the possibility to set this array in their function or service, for example, like this:
`set_weekdays_names({ 'MO', 'TU', ... }`

andrewjswan, 2023-10-06:
**2023.9.1**
Strange, tried to add russian letters to the configuration, to output to the information icon, but only rectangles are displayed, and in the log error:
```
[18:44:49][W][font:129]: Encountered character without representation in font: '\xd0'
[18:44:49][W][font:129]: Encountered character without representation in font: '\xa0'
```
Config:
```
  - file: fonts/MatrixChunky6.ttf
    size: 6
    id: default_font
    glyphs: |
      ! "#$%&'()*+,-./0123456789:;°C|¬¾¼•SUMOTWEHFRAПОВТСРЧБ

ehmtxv2:
  id: rgb8x32
  weekdays: "ПОВТСРЧТПТСБВС"
```
Same thing with MatrixChunky8.ttf. 

Today is ПТ, Saturday will be СБ, Sunday will be ВС.

@trip5 @lubeda Any ideas?

https://github.com/lubeda/EspHoMaTriXv2/blob/fbdb8feb0d8f1a5d93076d10e959141ca658b441/components/ehmtxv2/EHMTX_queue.cpp#L389-L390

trip5, 2023-10-06:
I'm working on this issue too but from a different angle...  I've already managed to stick strftime into a useable string that can have a string substitution applied to it on any element (AM, day, month, even the numbers).

I'm trying to make substitution arrays but I'm having trouble figuring out the correct conversions between the array and the string... C++ types are hard

I really think there should be a simple format that allows multi-character substitutions in whatever the ... that way it can be applied to all 21 (ish) strings (AM, PM, Weekdays, Months)... or whatever the user needs, since it's not always going to be 2 characters (Korean short days of the week are 1)...

My test array is simple (space is a delimiter):
````
  replace_time_date_from: "AM PM"
  replace_time_date_to: "am pm"
````

Later, it can be more complex:
````
  replace_time_date_from: "AM PM Jan Feb Mar Apr May Jun Jul Aug Sep Oct Nov Dec Sun Mon Tue Wed Thu Fri Sat"
  replace_time_date_to: "오후 오전 1월 2월 3월 4월 5월 6월 7월 8월 9월 10월 11월 12월 일 월 화 수 목 금 토"
````

But to answer your question, none of the 6-series fonts have Cyrillic yet.  I'm surprised it compiled despite you specifying glyphs not actually in the font file.

andrewjswan, 2023-10-06:
I think the problem is in Unicode, we take a character from the string, but this character does not seem to support Unicode, it must be converted somehow, but I do not know C++, so I have no ideas. Maybe @igor will tell me something....

trip5, 2023-10-06:
OK.  So I have code that compiles... not sure it works because I'm not sure that I'm building the array correctly on startup... it's either not doing that at all or it's being called constantly...  I can't tell because my clock won't display anything or even connect to Wifi.

At this point, I think I got the classes worked out.  I need @lubeda's advice where to put a void function that only needs to run once on startup to define up the array.  Right now it's in the middle of EHMTX.cpp but I think that puts it in the PollingComponent...

trip5, 2023-10-07:
Or I've got that part right... Serial monitor reveals `Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.`  So, likely I'm not building the array correctly.  I was trying to avoid external dependencies but I see iostream is part of the new branch so I'll try something else.  I'm 99% confident I can do this.

andrewjswan, 2023-10-07:
I'm going to try it today. I guess i could just use **substr**, I'll have to try it.

lubeda, 2023-10-07:
Hi,
the automatic translation of the weekday doesn't belong into [EspHoMaTriXv2](https://github.com/lubeda/EspHoMaTriXv2), I think the new YAML parameter `weekdays` must be enough! So i can set it to german "SOMODIMIDOFRSA" or any other language.

andrewjswan, 2023-10-07:
> So i can set it to german "SOMODIMIDOFRSA" or any other language.

There are Latin letters, no problems, Cyrillic already works as Unicode. they seem to have problems. I'll try different options.

andrewjswan, 2023-10-07:
> the automatic translation of the weekday doesn't belong into [EspHoMaTriXv2](https://github.com/lubeda/EspHoMaTriXv2),

I'm on 2023.9.1 right now.

andrewjswan, 2023-10-07:
Yeah, it's a Unicode problem. I put it in the log file:
```
ESP_LOGW("TEXT", ":: %c - %c - %d - %d", EHMTXv2_WEEKDAYTEXT[10], EHMTXv2_WEEKDAYTEXT[11], sizeof(EHMTXv2_WEEKDAYTEXT),strlen(EHMTXv2_WEEKDAYTEXT));
ESP_LOGW("TEXT", ":: %s", EHMTXv2_WEEKDAYTEXT);
```
And I got:
- Cyrillic:
```
[15:43:24][W][TEXT:440]: :: \xd0 - \xa0 - 29 - 28
[15:43:24][W][TEXT:441]: :: ПНВТСРЧТПТСБВС
[15:43:24][W][font:129]: Encountered character without representation in font: '\xd0'
[15:43:24][W][font:129]: Encountered character without representation in font: '\xa7'
```
- English:
```
[15:48:35][W][TEXT:440]: :: F - R - 15 - 14
[15:48:35][W][TEXT:441]: :: SOMODIMIDOFRSA
```
The size of the array is larger when using Cyrillic, **28** vs. **14** ... 

That's the problem, we get part of the Unicode letter, not all of it. I don't quite understand how to solve this yet. :(

andrewjswan, 2023-10-07:
I think I've solved this problem, I just need to check.
https://onlinegdb.com/111HCjbqC

andrewjswan, 2023-10-07:
Checked it out, it's working perfectly!

andrewjswan, 2023-10-07:
> replace_time_date_to: "오후 오전 1월 2월 3월 4월 5월 6월 7월 8월 9월 10월 11월 12월 일 월 화 수 목 금 토"

If only to display the day of the week, we can use unused characters, like the old Cyrillic and render each character from the two.

trip5, 2023-10-07:
> I think the new YAML parameter `weekdays` must be enough! So i can set it to german "SOMODIMIDOFRSA" or any other language.

The problem with this is that not all languages will use 2 characters to represent the day of the week.  English can use 1 or 2 (ie. Su M Tu W Th F Sa) while Korean only uses 1.

Also, if I can code this correctly, it will give the ability to change AM, PM, and anything else that strftime outputs into anything.  Of course, if it affects processing/display speed it will be pointless.  I'll give it another go but... if you're not interested in adding it... I promise the solution is actually looking fairly minimal (about 55 extra lines of code: 45 to set up the array and <10 to do the replacement).

Also, how are you working with this `weekdays:` in the yaml?  I'll test Korean with a TTF font but I'm not sure what else I'm missing from the yaml...?  I've also got `  date_format: "%l %a %d"` - what am I missing?  Or was it added to 9.1 in the last 48 hours and I need to update my local files?

andrewjswan, 2023-10-07:
> The problem with this is that not all languages will use 2 characters to represent the day of the week. English can use 1 or 2 (ie. Su M Tu W Th F Sa) while Korean only uses 1.

But if you draw one Korean character in any two regular characters, and set the display of them in the middle (mode 1), then we get one character on the icon, and it will be in the middle and it will look great in my opinion.

> Also, if I can code this correctly, it will give the ability to change AM, PM, and anything else that strftime outputs into anything.

We don't use AM/PM, so I don't even know. I don't really have anything to say about it, since I won't be using it.

> Also, how are you working with this `weekdays:` in the yaml? I'll test Korean with a TTF font but I'm not sure what else I'm missing from the yaml...? I've also got ` date_format: "%l %a %d"` - what am I missing? Or was it added to 9.1 in the last 48 hours and I need to update my local files?

This was added to the PR, i.e. it's not in the repository yet, but the PR has it :) I checked with myself, works fine, if I have a font with Korean days of the week I can promptly check.

trip5, 2023-10-07:
Sure sure.  Actually I did add them to 2 TTF files with my wife laughing at how some of them look... 8 pixels is pretty tight for Korean.  Korean actually uses 4 bytes compared to Cyrillic 2... so... very curious here... and it'll give you a chance to see if your code is right.

[8X Fonts (K Weekdays).zip](https://github.com/lubeda/EspHoMaTriXv2/files/12838327/8X.Fonts.K.Weekdays.zip)

It might look okay if you use a thin space (1 pixel in the font) ` ` to pad for 2 characters.

andrewjswan, 2023-10-07:
> Add this to your glyphs, too: `일 월화수목금토` (that's a thin space there)

[image]
[image]

andrewjswan, 2023-10-07:
@trip5 I will finalize that if there are only 7 letters in the array of days of the week, then output in the center, if there are 14 letters, then output in pairs as before.

PS: I use:
```
  weekdays:  "일 월 화 수 목 금 토 "
```

trip5, 2023-10-07:
Oh!  It's related to this!  I was half-wondering why 2 letters was so necessary...  That looks pretty good!

> only 7 letters in the array of days of the week

Actually that's a good idea to check for either 14 or 7 letters.

BTW thanks for referring to onlinegdb... I should have been trying an online C++ compiler to test my code and I didn't even think that.  I was just using ESPHome's output as a guide.

I still think my idea can be useful.  Although it's more for localizing time more than date anyways... It really bugs me that ESPHome forces people to use AM or PM... and capitalized too!  And it made my wife smile at the idea she could have a Korean clock.

andrewjswan, 2023-10-07:
> And it made my wife smile at the idea she could have a Korean clock.

Nothing like a smile from your kids, wife and parents :)
I hope we make it!

andrewjswan, 2023-10-07:
> Actually that's a good idea to check for either 14 or 7 letters.

[image]

trip5, 2023-10-07:
Nice!

andrewjswan, 2023-10-08:
Request down in 2023.9.1

trip5, 2023-10-10:
This might be closed but @andrewjswan can you please add page to the wiki how to do this?

What format is this?  `weekdays:  "일 월 화 수 목 금 토 "` - must this string be 14 characters?

Is this supposed to be called from my Home Assistant or is it set by the YAML?  Also, looks like then it's an icon and would be nice to have a PNG that can be downloaded from the wiki as well.  I know you're loving the coding side of this but features need to be documented, too.

andrewjswan, 2023-10-11:
> What format is this? `weekdays: "일 월 화 수 목 금 토 "` - must this string be 14 characters?

No? 7 characters. `weekdays:  "일월화수목금토"` :)

trip5, 2023-10-11:
I'm still not sure how to make this display on my clock. I don't see it in the Home Assistant services... Does it replace the time or date Screen? Maybe I need to turn on show days of week in yaml?

andrewjswan, 2023-10-11:
> I'm still not sure how to make this display on my clock. I don't see it in the Home Assistant services... Does it replace the time or date Screen? Maybe I need to turn on show days of week in yaml?
Add to config:
```
weekdays:  "일월화수목금토"
```

Add to config Icon:
```
    - id: calendar
      lameid: 9337
```
And from HA call service:

```
      - service: esphome.esp_pixel_clock_set_infotext_color
        data:
          left_r: 0
          left_g: 250
          left_b: 250
          right_r: 0
          right_g: 250
          right_b: 250
          default_font: false
          y_offset: 1
```
and
```
      - service: esphome.esp_pixel_clock_icon_date
        data:
          icon_name: calendar|weekday
          lifetime: 5
          screen_time: 5
          default_font: true
          r: 240
          g: 0
          b: 240
```

andrewjswan, 2023-10-11:
Or from ESPHome config:
```
        id(rgb8x32)->icon_date("calendar|weekday", 5, 5, true);
```

andrewjswan, 2023-10-11:
Reopened due - https://github.com/lubeda/EspHoMaTriXv2/pull/104

trip5, 2023-10-11:
Oh.  Why re-opened?  I don't think it's possible to mix the two together, although I have been testing some code to assist in making your code a bit more efficient... and basing it from libraries I was forced to use (vector and algorithm).

Namely, we can put some checks in the __init__.py if it is 7 or 14 letters... which can help in determining whether or not it's Unicode because then string length will then be either 7, 14, 21, 28 or 14, 28, 42, or 56...

andrewjswan, 2023-10-11:
> Namely, we can put some checks in the **init**.py if it is 7 or 14 letters... which can help in determining whether or not it's Unicode because then string length will then be either 7, 14, 21, 28 or 14, 28, 42, or 56...

You can't do it the easy way, because 7 characters in Cyrillic is 14 bytes, 7 bytes in Latin, and 28 bytes in Korean. But I use the get character function, it is safe at any length.

trip5, 2023-10-11:
Yup.  Quick way can't be done with reliability.  I tried to re-do your function without checking for each and every character and it broke when mixing UTF bytes inside each day (ie Latin + Korean like 1일 = 1st Day)...

But I do think that `GetWeekdayCharCount` can be handled by `__init__.py` on compile... and since both iterate through the entire string, character by character, permanently setting a variable could save some cycles... Python doesn't have a problems counting characters in Unicode.  And at the same time it checks that the yaml has either 7 or 14 characters and then gives a warning if not...

I checked Latin and Korean.  The calendar icon looks good but the clock is no longer centered, which is annoying.  Is there some way to fix that?

Anyway that's PR https://github.com/lubeda/EspHoMaTriXv2/pull/107

Also, fixed some of your typos with Nightmode.  You may want to sync up your own code and double-check it still functions (it should, ESPHome's compiler is extremely picky).

trip5, 2023-10-11:
PS.  The wife has given the thumbs-up to my interpretation of Korean characters.  It looks odd to me but just like the Ukrainian Em, natives are always going to know what looks "right" more than me.

andrewjswan, 2023-10-12:
I'm wondering if I should add an abriaviaiur on the icon for the month? Display for example Roman numerals, or two-letter abbreviation.

andrewjswan, 2023-10-12:
> Python doesn't have a problems counting characters in Unicode. And at the same time it checks that the yaml has either 7 or 14 characters and then gives a warning if not...

Without the character count function, we will lose the ability to change the list at run time.

andrewjswan, 2023-10-12:
> The calendar icon looks good but the clock is no longer centered, which is annoying. Is there some way to fix that?

What mode? I think that's been fixed before

trip5, 2023-10-12:
I mean left-right centering per your instructions above.  It's part of Mode 2 isn't it?

andrewjswan, 2023-10-12:
> I mean left-right centering per your instructions above. It's part of Mode 2 isn't it?

I checked for English and Ukrainian, it worked fine, mode 2 is right letter pressed to the right edge, left letter pressed to the center, if my memory does not fail me, pressing on the sides is the default mode, or 0.

trip5, 2023-10-12:
The date is fine. It's the clock itself that isn't centered on the remaining part of the display.

andrewjswan, 2023-10-12:
> The date is fine. It's the clock itself that isn't centered on the remaining part of the display.

It seems to be a problem in the coordinates, it was 19, and when replaced it became 15, although the center should be 19 rather than 15 ...
https://github.com/lubeda/EspHoMaTriXv2/blob/2023.9.1/components/ehmtxv2/EHMTX_queue.cpp#L388-L390

trip5, 2023-10-12:
Oh Lord, I see it.  I introduced that mistake with the replace... I copied and pasted but wrong.

https://github.com/lubeda/EspHoMaTriXv2/pull/111