lubeda/EspHoMaTriXv2 #92: [FEATURE REQUEST] A screen with an icon and data displayed on it.

andrewjswan, 2023-10-01:
# Feature Request

### Describe the solution / feature you'd like

I've already displayed the concept in another Issue, but I'll move it here.

Add a family of screens that could display some limited data like day of the week, day of the month, on an icon that works as a carpet pad. 

For example:
- icon_time - display: icon and time, like weather and time
- icon_time_data - display: icon and time, like calendar icon, day or weekday on icon and time
- icon_date - display: icon and date, like weather and time
- icon_date_data - display: icon and date, like calendar icon, day or weekday on icon and time

Generate a window identifier for each screen, then you can have identical sets displaying different information, like:
- weather - time
- day - time
- weekday - data
- weather - temperature

Also add the ability to specify a starting set of screens), e.g. `icon_time with day` icon_date_with_weekday`

### Additional context

Сoncept:
https://github.com/lubeda/EspHoMaTriXv2/issues/85#issuecomment-1738627157

malinovsku, 2023-10-04:
@andrewjswan Thanks! I tested the display with this icon, and it seems that it would be better to omit a number or a word, since it is strongly pressed to the top. There are also doubts about the gaps in both days and days of the week. maybe in the config it is possible to specify which format to show? Since opinions may be divided. Options on the screenshots
[image]
[image]
[image]
[image]
[image]

andrewjswan, 2023-10-05:
> since it is strongly pressed to the top

The baseline specified in the config for the selected font is used for output, i.e. the information on the icon is on the same bottom line as the font displayed on the right. The easiest solution is to specify 7(8) as the baseline, then the information on the icon will be displayed starting from the second bottom line, the way is more complicated is to get the font metric, get its width and height, and according to it output the text. 
I will try both options, see how it will be better and easier.

> There are also doubts about the gaps in both days and days of the week. maybe in the config it is possible to specify which format to show?

Well, the usual, simple, `crutchy way` I can do it. The smart way is to pass the display format and store it, I don't know what is better, but let's start with `crutches`. :) Let's use a pipe `|` :)

andrewjswan, 2023-10-05:
Mode added ...
- icon|day#mode
- icon|weekday#mode
mode:
- 0 - default - left and right - baseline from font
- 1 - center - baseline from font
- 2 - center - right - baseline from font
- 3 - left and right - baseline 7
- 4 - center - baseline 7
- 5 - center - right - baseline 7
https://github.com/andrewjswan/EspHoMaTriXv2/commit/4949ceeb58535b004ef2fe4272191ed1c2d9e5b6

andrewjswan, 2023-10-05:
@malinovsku @lubeda 
I think this is the best option. 
I just need to add the ability to set 2 colors, or separately for days and days of the week, i.e. 4.
[image]

andrewjswan, 2023-10-05:
`set_infotext_color` added ...

andrewjswan, 2023-10-06:
Tested it yesterday, not all bad but not all good either. 
Different fonts bring a lot of variety, especially letters, each font has its own base line, width, height, and it is difficult to calculate and correctly fit into a **matrix 8x8**, I added `info_text_yoffset` to specify the base line for displaying information on the icon. I also redid the display modes. 
I'll check how it works and make a PR.

andrewjswan, 2023-10-06:
For now, the current display modes:
Default:
[image]
Mode 1:
[image]
Mode 2:
[image]
Mode 3:
[image]
Mode 4:
[image]

andrewjswan, 2023-10-06:
Although, I see Igor did it through functions, maybe that's more correct, I'll follow that way. I will make changes in my new code and check it.
https://github.com/lubeda/EspHoMaTriXv2/commit/fbdb8feb0d8f1a5d93076d10e959141ca658b441

andrewjswan, 2023-10-06:
Look like all work fine...
[image]
[image]
[image]
[image]
[image]
[image]

andrewjswan, 2023-10-06:
Test scrypt:
```yaml
script:
  pixel_clock_test:
    alias: "Pixel Clock - Test"
    sequence:
      - service: esphome.pixel_clock_set_infotext_color
        data:
          left_r: 50
          left_g: 50
          left_b: 50
          right_r: 50
          right_g: 50
          right_b: 240
          default_font: true
          y_offset: 1

      - service: esphome.pixel_clock_icon_date
        data:
          icon_name: calendar|day
          lifetime: 1
          screen_time: 5
          default_font: true
          r: 240
          g: 0
          b: 0
      - service: esphome.pixel_clock_icon_date
        data:
          icon_name: calendar|day#1
          lifetime: 1
          screen_time: 5
          default_font: true
          r: 0
          g: 240
          b: 0
      - service: esphome.pixel_clock_icon_date
        data:
          icon_name: calendar|day#2
          lifetime: 1
          screen_time: 5
          default_font: true
          r: 0
          g: 0
          b: 240
      - service: esphome.pixel_clock_icon_date
        data:
          icon_name: calendar|day#3
          lifetime: 1
          screen_time: 5
          default_font: true
          r: 240
          g: 240
          b: 240
      - service: esphome.pixel_clock_icon_date
        data:
          icon_name: calendar|day#4
          lifetime: 1
          screen_time: 5
          default_font: true
          r: 240
          g: 240
          b: 240

      - service: esphome.pixel_clock_icon_date
        data:
          icon_name: calendar|weekday
          lifetime: 1
          screen_time: 5
          default_font: true
          r: 240
          g: 0
          b: 240
```

andrewjswan, 2023-10-06:
Look like All worked ...

lubeda, 2023-10-07:
There was an error in `set_infotext_color` which is fixed now.

andrewjswan, 2023-10-08:
Request down in 2023.9.1

closed 2023-10-08 by andrewjswan

malinovsku, 2023-10-08:
thank you, you do cool things!

andrewjswan, 2023-10-13:
[image]