lubeda/EspHoMaTriXv2 #93: [BUG] The on_empty_queue event did not work

andrewjswan, 2023-10-02:
# Bug report

## Describe the bug

Yesterday started preparatory work to modify the source code to create Pull Request, flashed for the correct config `2023.9`, specified as in the documentation:
```
  on_empty_queue:
    then:
      lambda: |-
        id(rgb8x32)->clock_screen(10,10);
        id(rgb8x32)->date_screen(10,5);

  on_start_running:
     then:
       lambda: |-
          id(rgb8x32)->clock_screen(10,10);
          id(rgb8x32)->date_screen(10,5);
```

Also from HA, the weather is constantly displayed on the screen. I.e. constant 3 screens, `clock`, `date`, `weather` (updated often because it is tied not only to the weather, but also to the temperature).
But this morning, there was `only weather` on the screen, no clock or date. It seems that the `on_empty_queue` event did not happen.

## Additional information

- used Hardware:
  - ESP32
  - EspHoMaTriXv2 version: [e.g. 2023.9.0]

lubeda, 2023-10-02:
This night a had a similar effect :-(

omia, 2023-10-03:
The on_empty_queue event should not happen. The queue was not empty. There was a screen in the queue.

andrewjswan, 2023-10-03:
That's understandable, what's not so clear is where the date and time disappeared to.

omia, 2023-10-03:
They don't disappear. They are created with 10minutes of live.

omia, 2023-10-03:
I see two possible ways for this.
1. A second queue for permanent screens which don't have a live time.
2. Permanent screens should only be created within the clock(pulling data)

andrewjswan, 2023-10-03:
Well, we basically have no screens that are permanent, since we already have at least two types of watches. 
Here it is necessary either to constantly extend the life of the screen,
```
    this->endtime = this->clock->now().timestamp + 300;
```
 or to set the minimum starting set, which is immortal.

omia, 2023-10-03:
What is the differenz between a permanent screen and a immortal one?

andrewjswan, 2023-10-03:
Basically nothing, but we need to understand where the temporary screen is and where it is not, and when temporary can become permanent or vice versa.

andrewjswan, 2023-10-09:
We could try replacing the `on_empty_queue` event with `on_expired_screen`, then as soon as the clock becomes obsolete it can be returned :)

omia, 2023-10-09:
How do we know which screen expired?

andrewjswan, 2023-10-09:
> How do we know which screen expired?

Judging by the source code, the information about the screen is transmitted there, but we can not analyze it, just increase the life of the necessary screens, even by an hour, even by a day.

omia, 2023-10-09:
But if we have a weather screen running all the time even a life of a day will run out and the on_empty_screen should never be called.
What a about a on_screen_death event. So we can put a function as a parameter. This would even be a possibility to make it more generic. So a timer for example would make a sound on screen death.

lubeda, 2023-10-09:
Besides the clock I don't understand the concept of permanent screens. For me, e.g. the wind speed is pushed with a lifetime of two minutes to the display by an automation that triggers on every state change. So the wind speed is "permanent" on the screen but totally controlled by home assistant. If you need something like a "motto of the day" set the lifetime to 24*60 minutes triggers by an automation on home assistant at 12p.m..

andrewjswan, 2023-10-09:
I have:
```
  on_empty_queue:
    then:
      lambda: |-
        id(rgb8x32)->clock_screen(1440,10);
        id(rgb8x32)->date_screen(1440,5);
```
And weather and temperature from HA, trigger state changes, lifetime 30 min

And a day, two days later, all I have is the weather on my screen. No clock or time

omia, 2023-10-09:
The problem can also happen without any "permanent" screens. Ad some point in time the clock screen dies. An if then any screen is in the queue the clock is gone.
And only if all screens are removed we get a trigger to renew the clock. So if we have homeassistant add 4 10min screens for fun things in the meantime we don't have the time on the clock.