lubeda/EspHoMaTriXv2 #95: [BUG] set_clock_color changes the displayed font

andrewjswan, 2023-10-03:
# Bug report

## Describe the bug

I call the `set_clock_color` function from HA, it changes the `time` and `date` font from `default_font` to `special_font`

## Additional information

- used Hardware:
  - ESP32
  - EspHoMaTriXv2 version: 2023.9.0

## To Reproduce

Make call:
```
service: esphome.esp_pixel_clock_set_clock_color
data:
  r: 0
  g: 250
  b: 250
```
## Configuration

```text
font:
  # Matrix Clock Fonts
  # Grab these from hhttps://github.com/trip5/Matrix-Fonts
  # Store in a "fonts" subfolder of your ESPHome config folder or change paths below
  - file: fonts/MatrixChunky6.ttf
    size: 6
    id: default_font
    glyphs: |
      ! "#$%&'()*+,-./0123456789:;°C|¬¾¼•MTWFSonuedhriatПНВТСРЧБ

  - file: fonts/MatrixChunky8.ttf
    size: 8
    id: special_font
    glyphs: |
      ! "#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz°|АаБбВвГгҐґДдЕеЄєЖжЗзИиІіЇїЙйКкЛлМмНнОоПпРрСсТтУуФфХхЦцЧчШшЩщЬьЮюЯяЁёЪъЫыЭэ℃₴€

ehmtxv2:
  id: rgb8x32
  icons2html: true
  display_graph: ehmtx_graph
  matrix_component: ehmtx_display
  time_component: ehmtx_time
  time_format: "%H:%M:%S"
#  date_format: "%a•%d•%m"
  date_format: "%d•%m"
  show_seconds: false
  default_font_id: default_font
  default_font_yoffset: 7
  special_font_id: special_font
  special_font_yoffset: 8
#  blend_steps: 16
  scroll_count: 2 # scroll long text at least two times
  scroll_interval: 80 # milliseconds
  rainbow_interval: 32 # milliseconds
```

## Services calls

```text
service: esphome.esp_pixel_clock_set_clock_color
data:
  r: 0
  g: 250
  b: 250
```

trip5, 2023-10-04:
Are you showing the days of the week? Does the `default_font_yoffset: 7` not affect the text bumping into that line?  Probably should be `default_font_yoffset: 6` for 6-point fonts.

andrewjswan, 2023-10-04:
> Are you showing the days of the week?

Yes

> Does the `default_font_yoffset: 7` not affect the text bumping into that line? Probably should be `default_font_yoffset: 6` for 6-point fonts.

6 points font height, at `default_font_yoffset 6`, it turns out very high, at `7` just in the middle, one point above and one below. Both on numbers and text.

andrewjswan, 2023-10-04:
https://github.com/lubeda/EspHoMaTriXv2/pull/94/commits/80bfa4a3ca23a0fc44c737e4b8d508b8e1cceee9

andrewjswan, 2023-10-05:
Fixed in [2023.9.1-95: set_clock_color fix] https://github.com/andrewjswan/EspHoMaTriXv2/commit/80bfa4a3ca23a0fc44c737e4b8d508b8e1cceee9

closed 2023-10-05 by andrewjswan