lubeda/EspHoMaTriXv2 #103: [FEATURE REQUEST] Night mode

andrewjswan, 2023-10-09:
# Feature Request

### Describe the solution / feature you'd like

Add night mode:
- Add the ability to enable or disable the night mode of operation.
  - `night_mode_on`
  - `night_mode_off`
- Add the ability to set a list of screens to display in night mode:
  - Or directly in the config.
    - `night_modes: 2,3,16`
  - Or by an additional function through the screen arrays:
    - `set_night_modes([2,3,16]);`
- When night mode is enabled, **only the specified screens will be displayed**, all other screens will be skipped, but **will live in the queue until their lifetime expires**.

andrewjswan, 2023-10-09:
I made the preparatory changes https://github.com/andrewjswan/EspHoMaTriXv2/commit/d1c0e6c2228f6f5417986695495def0c9a7f56b0
But I don't fully understand how the queue works, @lubeda , could you please advise, or help? I can do PR.
Thanks!

lubeda, 2023-10-10:
The decision which data is important at night should be done by home assistant not by the ehmtx! 
Especially specifying modes  is not the way I would do this. If you don't want a clock at night use `del_screen` in the evening and `add_screen` it in the morning.

andrewjswan, 2023-10-10:
Well, this is the beauty of the mode that I propose, it is autonomous, and we can display at night only the necessary information, and at this time fill the queue with the same weather, events to inform and they will be in the queue but not shown, as soon as we go to the end of the night, everything in the queue will be shown. This is convenient for events that rarely occur, such as birthday.

andrewjswan, 2023-10-10:
Checked it out, in my opinion it works just fine. 

It would not be bad to add events when switching on and off this mode, as well as events when switching off the display.
But I do not know how to add them.

andrewjswan, 2023-10-11:
on_show_display, on_night_mode trigger added ...

andrewjswan, 2023-10-11:
Request down - https://github.com/lubeda/EspHoMaTriXv2/pull/105