lubeda/EspHoMaTriXv2 #106: [FEATURE REQUEST] Get Current showing Icon and text from device

DannAberHardt, 2023-10-11:
# Feature Request

### Is your feature request related to a problem?
no

### Describe the solution / feature you'd like
I use the ulanzi display to display multiple notifications (icon+text) and quitting the actial showing notification by pressing the middle button.
I use home assistant for the logic.
Currently i have to save the current icon in Home Assistant every time "next_screen" is called to
do a del_screen for the saved icon.
The problem is, the db is growing much couse of the massive events from next_screen.

I wish there were a api to get the current icon and text to get rid off the next next_screen event

### Describe alternatives you've considered
-
### Additional context
-

lubeda, 2023-10-11:
Hi, for now, the event method is the only way to get data back to homeassistant (i know). For now, there is nothing like return values on esphome service calls.

andrewjswan, 2023-10-11:
> The problem is, the db is growing much couse of the massive events from next_screen.

You can exclude this event from the recorder

andrewjswan, 2023-10-11:
> Hi, for now, the event method is the only way to get data back to homeassistant (i know). For now, there is nothing like return values on esphome service calls.

You can add sensors, and put the values you want in them. But I don't know how much it is necessary and how it is worse than a sensor trigger on the HA side.

lubeda, 2023-10-11:
Text sensors will also fill the database. so no gain

DannAberHardt, 2023-10-11:
> > The problem is, the db is growing much couse of the massive events from next_screen.
> 
> You can exclude this event from the recorder

I have already tried this, but unfortunately dependent actions no longer work reliably

andrewjswan, 2023-10-11:
> I have already tried this, but unfortunately dependent actions no longer work reliably

Actions don't depend he history of states, only history and stats.

andrewjswan, 2023-10-12:
> I use home assistant for the logic.

Wouldn't it be easier to do the logic in ESPHome?

DannAberHardt, 2023-10-12:
> Actions don't depend he history of states, only history and stats.
Okay, good to know.
So i double checked and retestet the HA logic and i figured out that the filtered recorder is not the root of the issue.

> Wouldn't it be easier to do the logic in ESPHome?
I use HA cause so i can quit these notifications also with other devices or automations.

Thanks all for your replies 👍
I close this topic since there is no way for my initial request for getting states from display on the fly.

closed 2023-10-12 by DannAberHardt