lubeda/EspHoMaTriXv2 #112: Show screen instantly

s00500, 2023-10-13:
# Question

Hey, cool project,
Is there a way to show something immediatly ? I tried creating a screen then force screen but this will wait until the current screens time is out before switching... anyway to stop that ?

Also having only minute steps for lifetime will mean when I want to show a message for 10 seconds and then remove the screen it will come up once more...

Is there anything I am doing wrong ? How would I best approach such a problem ?

Greetings
Lukas

## Additional information
Using Ulanzi Display

andrewjswan, 2023-10-13:
2023.9.1: `alert_screen` - show the specified icon with text, screen forced and lifetime = screen_time

lubeda, 2023-10-13:
and force_screen may be used

closed 2023-10-13 by lubeda

s00500, 2023-10-13:
Awesome, works as expected