lubeda/EspHoMaTriXv2 #123: [FEATURE REQUEST] Scroll icon along with long text

chertvl, 2023-10-28:
# Feature Request

### Is your feature request related to a problem?

Hello ;)
Our matrix has low resolution and size,  they can not contains too much text to display, that why we use long-text scrolling.
For now, I can to call many variants of services to display icon+long text, and long-text will scroll.
But there is no way to scroll icon along with text, icon always displays and uses 25% of screen (8x8 px).

### Describe the solution / feature you'd like

It would be great to have possibility to scroll icon along with text (from left to right, off the screen),  like it does Awtrix.
Thats save 25% of matrix space, to display more letters at the same moment.

### Additional context

Here is some video of Awtrix text scroll, what i mean.
https://drive.google.com/file/d/1Vj6Mb0sN__ide0T__dQGNk2aqG63QSFF/view?usp=sharing

Thank you!

andrewjswan, 2023-10-28:
Good idea for an info window... Especially if there is a lot of text being displayed...

andrewjswan, 2023-10-29:
https://youtu.be/Bwhiz0SDhfE

andrewjswan, 2023-10-29:
I'll add a new screen type.
`info_text`
`rainbow_info_text`
and make PR

chertvl, 2023-10-29:
> https://youtu.be/Bwhiz0SDhfE

It looks amazing!
We must implement it! 
Thank you

andrewjswan, 2023-11-01:
Done in 2023.9.1

closed 2023-11-01 by lubeda