lubeda/EspHoMaTriXv2 #126: [BUG] blend_steps with only 1 screen

chertvl, 2023-10-29:
# Bug report

## Describe the bug

Unfortunately, I noticed that the blend_steps function does not work correctly when there is only 1 screen in the queue.

In my concept, I use the device only as a watch; other screens appear and disappear depending on the automation in HA. That is, the device only displays the clock for long periods of time and does not change screens.

If blend_steps is enabled, the device will appear to **blink every 10 seconds** when the transition to the next screen should occur. But there is no next screen and we get a very unpleasant effect.

Is it possible not to use blend_steps when there are no other screens?
The transition feature is very nice and I wouldn't want to give it up.

Thank you very much for your time!

Video:

## Additional information

- used Hardware:
  - Ulanzi TC001

## To Reproduce

Set blend_steps: 16  and leave one screen only.

## Screenshots

Video:
https://drive.google.com/file/d/1VnpeHHaE8OvAVetf6i3eVzIksZKr8qkT/view?usp=sharing

lubeda, 2023-10-30:
this is solvable, I will take a look in the next days.

lubeda, 2023-11-01:
fixed

closed 2023-11-01 by lubeda