lubeda/EspHoMaTriXv2 #129: [FEATURE REQUEST] Turn off blend_step at low brightness

andrewjswan, 2023-10-31:
# Feature Request

### Is your feature request related to a problem?

At low brightness blend_step can be said to not work. The image just disappears from the matrix and a new one appears, in the evenings when the light is off it is very annoying.

### Describe the solution / feature you'd like

I suggest adding a check of brightness and if it is below a certain threshold (can be added to the configuration, or count half between the minimum and maximum brightness or just pick a number) and if the brightness is less than the threshold then do not perform blend_step.

lubeda, 2023-10-31:
nice idea, I think this is easily possible to do!

andrewjswan, 2023-11-01:
We can even make that if auto brightness is enabled, then reduce blend step, ie specified 32, at maximum brightness 32, at medium 16, if less then disabled.

andrewjswan, 2023-11-05:
Fixed in 2023.9.1

closed 2023-11-05 by andrewjswan