lubeda/EspHoMaTriXv2 #132: [QUESTION] Migrate from v1 to v2

popy2k14, 2023-11-03:
# Question

@lubeda is there an easy way to migrate over from v1 to v2?
Running two ulanzis now with v1 with no issues but as you archived v1 i am planning to migrate to v2.

## Additional information

- used Hardware:
  - ESP32

  - EspHoMaTriXv1 version: 2023.4.0

lubeda, 2023-11-03:
Not really! Since the possibilities changed between the two versions, you have to rethink your automations. 
Using a text editor could help to find old `add_screen` commands but the new `icon_screen` has additional parameters ("r", "g", "b").  So it is a manual process. Sorry

Schampanja, 2024-01-08:
Hi, 
I have a homemade DIY Display by using the Tutorial from the SmartHome Yourself Youtube Chanel.
I used the fullfeature.yaml from V1 as en example to cnfigure the project.
Now I realised, that there is a V2 version of EspHoMatrix available.
Can i use my old configuration by just changing the external_component?
Or do I have to create a new configuration file? And if Yes, is there a "fullfeature" example like in V1?
I just found the Ulanzi yaml.

Thx a lot

David

Schampanja, 2024-01-11:
OK, I got it. Now I am on V2! 
I had to change a view lines of YAML code but in fact it wasn't that much work.
@lubeda Thank you for the nice project! 
David

closed 2024-01-11 by lubeda

popy2k14, 2024-01-27:
Now i found time to also migrate to v2.
Took a little bit longer to merge my config (because of lot's of customizations) but now the most part is working.
I have two issues and will open an github issue for them.

thx