lubeda/EspHoMaTriXv2 #133: [BUG] Screens re-starts and showing twice with "scroll_count: 1" set

chertvl, 2023-11-03:
# Bug report
## BUG

I have set ```scroll_count: 1``` in config and calling services with ```screen_time: 1``` parameter (for minimal time of displaying)  for long texts wich scrolls.
Sometimes (but not always) after text scrolls from start to end, current screen re-lauched and scroll starts again.
I cant get the reason for this behavior, but the length of the text certainly does not matter here, nor do the parameters of the called service.

As always, ive recorded a video. Ive called same service with same parameters, in the first 3 attempts the screen repeats from the beginning, but the 4th time everything is fine.

We can see re-start of the screen in logs at 19:08:46 and 19:08:53.

Add a description of the bug. Detail the expected behavior in contrast with the behavior you're observing.

## Additional information

- used Hardware:
  - Ulanzi TC001

## To Reproduce

Steps to reproduce the behavior:  Set ```scroll_count: 1``` and call the service below

## Configuration

```
ehmtxv2:
  id: rgb8x32
  icons2html: true
  matrix_component: ehmtx_display
  time_component: ehmtx_time
  time_format: "%H:%M"
  date_format: "%d•%m"
  weekdays: "ВСПНВТСРЧТПТСБ"
  show_seconds: false
  default_font_id: default_font
  default_font_yoffset: 7
  special_font_id: special_font
  special_font_yoffset: 8
  blend_steps: 16
  scroll_count: 1  #scroll long text one time
  scroll_interval: 32 
  rainbow_interval: 32 
```

## Screenshots

Video: https://drive.google.com/file/d/1WxI0mMEYrB80Wu9x_nyBdtHB_BT9QX-f/

## Logs

```
[19:08:43][D][EHMTXv2:687]: oldest queue element is: 0/1
[19:08:43][D][EHMTXv2:119]: queue: icon clock: "day#1" for: 10.0 sec
[19:08:43][D][EHMTXv2:894]: on_next_screen trigger
[19:08:45][D][sensor:094]: 'Ulanzi Illuminance': Sending state 208.79446 lx with 0 decimals of accuracy
[19:08:46][D][EHMTXv2:365]: icon: home_assistant found id: 0
[19:08:46][D][EHMTXv2:1540]: free_screen: found by endtime 1
[19:08:46][D][EHMTXv2:833]: calc_scroll_time: mode: 16 text: "Testing strange second scroll with long text" pixels 178 calculated: 6.5 defined: 1 max_steps: 202
[19:08:46][D][EHMTXv2:1113]: alert screen icon: 0 iconname: home_assistant text: Testing strange second scroll with long text screen_time: 1
[19:08:46][D][EHMTXv2:125]: queue: icon: "home_assistant" for: 6.5 sec
[19:08:46][D][EHMTXv2:647]: force_screen: found position: 1
[19:08:46][W][EHMTXv2:651]: force_screen: icon home_assistant in mode 16
[19:08:46][D][EHMTXv2:687]: oldest queue element is: 1/2
[19:08:46][D][EHMTXv2:125]: queue: icon: "home_assistant" for: 6.5 sec
[19:08:46][D][EHMTXv2:894]: on_next_screen trigger
[19:08:53][D][EHMTXv2:687]: oldest queue element is: 1/2
[19:08:53][D][EHMTXv2:125]: queue: icon: "home_assistant" for: 6.5 sec
[19:08:53][D][EHMTXv2:894]: on_next_screen trigger
[19:08:55][D][sensor:094]: 'Ulanzi Illuminance': Sending state 208.54295 lx with 0 decimals of accuracy
[19:08:59][D][EHMTXv2:748]: remove expired queue element: slot 1: mode: 16 icon_name: home_assistant text: Testing strange second scroll with long text
[19:08:59][D][EHMTXv2:687]: oldest queue element is: 0/1
[19:08:59][D][EHMTXv2:119]: queue: icon clock: "day#1" for: 10.0 sec
[19:08:59][D][EHMTXv2:894]: on_next_screen trigger
[19:09:05][D][sensor:094]: 'Ulanzi Illuminance': Sending state 208.73730 lx with 0 decimals of accuracy
[19:09:09][D][EHMTXv2:687]: oldest queue element is: 0/1
[19:09:09][D][EHMTXv2:119]: queue: icon clock: "day#1" for: 10.0 sec
```

## Services calls

```
service: esphome.ulanzi_alert_screen
data:
  default_font: true
  icon_name: home_assistant
  screen_time: 1
  r: 255
  g: 255
  b: 255
  text: Testing strange second scroll with long text
```

chertvl, 2023-11-03:
Logs for attemp wich not re-started (all ok). Maybe there is some difference:

```
[19:53:14][D][EHMTXv2:687]: oldest queue element is: 0/1
[19:53:14][D][EHMTXv2:119]: queue: icon clock: "day#1" for: 10.0 sec
[19:53:14][D][EHMTXv2:894]: on_next_screen trigger
[19:53:17][D][EHMTXv2:365]: icon: home_assistant found id: 0
[19:53:17][D][EHMTXv2:1540]: free_screen: found by endtime 1
[19:53:17][D][EHMTXv2:833]: calc_scroll_time: mode: 16 text: "Testing strange second scroll with long text" pixels 178 calculated: 7 defined: 1 max_steps: 202
[19:53:17][D][EHMTXv2:1113]: alert screen icon: 0 iconname: home_assistant text: Testing strange second scroll with long text screen_time: 1
[19:53:17][D][EHMTXv2:125]: queue: icon: "home_assistant" for: 7 sec
[19:53:17][D][EHMTXv2:647]: force_screen: found position: 1
[19:53:17][W][EHMTXv2:651]: force_screen: icon home_assistant in mode 16
[19:53:17][D][EHMTXv2:687]: oldest queue element is: 1/2
[19:53:17][D][EHMTXv2:125]: queue: icon: "home_assistant" for: 7 sec
[19:53:17][D][EHMTXv2:894]: on_next_screen trigger
[19:53:19][D][sensor:094]: 'Ulanzi Illuminance': Sending state 168.37253 lx with 0 decimals of accuracy
[19:53:24][D][EHMTXv2:687]: oldest queue element is: 0/2
[19:53:24][D][EHMTXv2:119]: queue: icon clock: "day#1" for: 10.0 sec
[19:53:24][D][EHMTXv2:894]: on_next_screen trigger
```

chertvl, 2023-11-04:
In couple of days of playing with many combinations of screens and services, 
I discovered that the problem is in the force_screen service.

I dont see any screen restarts if I don't send a ```force``` service after any screen service.

And I cant call Alert screen without ```force``` , because its hardcoded, and bug presents here at random percent of attemps.

lubeda, 2023-11-05:
When you have only one screen in the queue this is normal behavior. The single element is repeated until lifetime ends or  another screen is added, for showing a screen only once there is `alarm_screen`

`[19:56:42][D][EHMTXv2:687]: oldest queue element is: 0/1`

chertvl, 2023-11-05:
> When you have only one screen in the queue this is normal behavior. The single element is repeated until lifetime ends or another screen is added, for showing a screen only once there is `alarm_screen`
> 
> `[19:56:42][D][EHMTXv2:687]: oldest queue element is: 0/1`

I myself am already confused in my description of this bug.

In short:
If my queue is empty, and I call an Alert, which is automatically calculated for 7 seconds, then the Alert in half the cases will be displayed not for 7 seconds but for 14. The text will scroll not twice, but four times.
The alert will reach the end and restart. Because there is a hardcoded force on alert.

Similar behavior with other types of screens if I send a force service immediately after creating the screen.

chertvl, 2023-11-12:
Tested more times with Alarm screen (wich has hardcored "force" func)

The difference between normal visualisation of screen and of double-time showed screen somewhere here:
If I see in logs ```oldest queue element is: 0/2``` - all will be ok,  forced screen will be deleted by lifetime, showing only once.
But if there in logs ```oldest queue element is: 1/2``` - screen will restart after him screentime ends, and showing twice (x2 scrolls, x2 screentime).

I think, problem somewhere in ```oldest quere element``` calculation, which leads to the screen starting over again, despite the fact that it was just displayed (and for some reason not always, in 50% of cases, there is no pattern, sometimes 0/2, sometimes 1/2 and a bug where screen again is next).

lubeda, 2023-11-12:
Good analyzed, I think in the latest version of 2023.9.1 it's fixed!

chertvl, 2023-11-12:
> Good analyzed, I think in the latest version of 2023.9.1 it's fixed!

Thank you for your attention.
Unfortunately, this adjustment has not helped yet.
The alert screen is still called twice half the time. 

```
[13:37:32][D][EHMTXv2:119]: queue: icon clock: "day#1" for: 10.0 sec
[13:37:32][D][EHMTXv2:1015]: on_next_screen trigger
[13:37:32][D][text_sensor:064]: 'Ulanzi Current Screen': Sending state 'day#1'
[13:37:35][D][EHMTXv2:465]: icon: spotify found id: 29
[13:37:35][D][EHMTXv2:1674]: free_screen: found by endtime 1
[13:37:35][D][EHMTXv2:996]: calc_scroll_time: mode: 16 text: "Song name - Song artist" pixels 94 calculated: 3.8 defined: 0 max_steps: 118
[13:37:35][D][EHMTXv2:1247]: alert screen icon: 29 iconname: spotify text: Song name - Song artist screen_time: 0
[13:37:35][D][EHMTXv2:125]: queue: icon: "spotify" for: 3.8 sec
[13:37:35][D][EHMTXv2:748]: force_screen: found position: 1
[13:37:35][W][EHMTXv2:752]: force_screen: icon spotify in mode 16

[13:37:35][D][EHMTXv2:788]: oldest queue element is: 1/2
[13:37:35][D][EHMTXv2:125]: queue: icon: "spotify" for: 3.8 sec
[13:37:35][D][EHMTXv2:1015]: on_next_screen trigger
[13:37:35][D][text_sensor:064]: 'Ulanzi Current Screen': Sending state 'spotify'

[13:37:39][D][EHMTXv2:788]: oldest queue element is: 1/2
[13:37:39][D][EHMTXv2:125]: queue: icon: "spotify" for: 3.8 sec
[13:37:39][D][EHMTXv2:1015]: on_next_screen trigger
[13:37:39][D][text_sensor:064]: 'Ulanzi Current Screen': Sending state 'spotify'

[13:37:42][D][sensor:094]: 'Ulanzi Illuminance': Sending state 136.59384 lx with 0 decimals of accuracy
[13:37:42][I][EHMTXv2:1725]: set_brightness 39 => 15.29 %
[13:37:43][D][EHMTXv2:849]: remove expired queue element: slot 1: mode: 16 icon_name: spotify text: Song name - Song artist
[13:37:43][D][EHMTXv2:788]: oldest queue element is: 0/1
[13:37:43][D][EHMTXv2:119]: queue: icon clock: "day#1" for: 10.0 sec
```

andrewjswan, 2023-11-12:
Yesterday saw a screen restart at my place, the restart was on the icon_date screen in the queue at that moment there were at least 5 screens, force was not used, the screen was visible, then almost at the end flashed with a blend and was shown again.

andrewjswan, 2023-11-14:
Look like fixed ...

chertvl, 2023-11-17:
https://github.com/lubeda/EspHoMaTriXv2/pull/148  fixed it!
Many thanks!

closed 2023-11-17 by chertvl