lubeda/EspHoMaTriXv2 #134: [FEATURE REQUEST] Add transition to brightness.

chertvl, 2023-11-03:
# Feature Request

### Describe the solution / feature you'd like

I think in would be great if we can add a transition for brightness.
For now if I change brightness (from 30/255 to 100/255), the matrix will change it instantly.
Like a smartbulbs, LED-panel should change self brightness smoothly for some time,
and it will be great if we could set time in seconds/millis in config.

For example, current brightness is 30/255,  and I set 100/255, then LED matrix does:  31/255, 32/255, 33/255... until it get ...100/255, for ex. for 3 seconds.

I think this will improve the visual aspect of the auto-brightness function also. No more instantly brightness-down when I turn off light in room, and vice versa.

andrewjswan, 2023-11-04:
@lubeda , I looked at the source code, we don't have brightness control as such, brightness is changed via `set_correction`, so the native light function as `transition` doesn't work.
Is this done on purpose?

lubeda, 2023-11-04:
There is no API for brightness, so I had to use set_correction.

chertvl, 2023-11-05:
Done by https://github.com/lubeda/EspHoMaTriXv2/pull/136

closed 2023-11-05 by chertvl