lubeda/EspHoMaTriXv2 #139: [WIP] Bitmap Stack screen

andrewjswan, 2023-11-09:
# Work In Progress

### Describe the solution / feature you'd like

A screen that allows you to display from 1 to 64 icons described in the configuration.
I don’t know what kind of case it can be used in, but I was interested in trying out what it would look like, and when I saw it live, I realized that it was funny, and even possibly useful.

### Additional context

I don’t have a lot of videos coming out, but maybe one of those who have already tried it will publish what it looks like.

andrewjswan, 2023-11-09:
Usage example:
```
service: esphome.ulanzi_bitmap_stack
data:
  icons: icon1,icon2
  lifetime: 1
  screen_time: 20
```

```
service: esphome.ulanzi_bitmap_stack
data:
  icons: icon1,icon2,icon3|two
  lifetime: 1
  screen_time: 20
```

chertvl, 2023-11-09:
Looks amazing, already tried from your fork, here is a video

https://drive.google.com/file/d/1XvQt1VZ5APJuRZQ1SJEl_YT9_Yh0KfZT

https://drive.google.com/file/d/1XtQ7D08-ovWCPMzsgSgp79jMJtL6oR1Y

riker09, 2023-11-10:
Looks great!

Off topic question: Can you share your code for your clock screen? Personally I'd like to have both, time and date, on one screen.

chertvl, 2023-11-10:
> Can you share your code for your clock screen? Personally I'd like to have both, time and date, on one screen.

In my configuration, its:
```
  on_start_running:
    then:
      lambda: |-
        id(rgb8x32)->icon_clock("blank|day#1", 1440, 1, true);
        id(rgb8x32)->set_infotext_color(150, 150, 150, 150, 150, 150, true, 0); 

  on_empty_queue:
    then:
      lambda: |-
        id(rgb8x32)->icon_clock("blank|day#1", 1440, 1, true);
        id(rgb8x32)->set_infotext_color(150, 150, 150, 150, 150, 150, true, 0); 
```

andrewjswan, 2023-11-10:
Added to 2023.9.1

closed 2023-11-10 by andrewjswan